raspberrypi/linux #2510: system/dmesg logs full of: "Transfer to device N endpoint 0x2 frame nnnn failed"

E3V3A, 2018-04-16:
I'm using a USB C-media sound card, and every now and then my `dmesg` is filled with:
```
Transfer to device 6 endpoint 0x2 frame 1924 failed - FIQ reported NYET. Data may have been lost.
```
Apparently this happens when there have been updates and changes to the kernel and it's boot flags.
My OS is: 
```bash
 # uname -a ; sudo lsb_release -a

Linux xxx 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux

No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.3 (stretch)
Release:        9.3
Codename:       stretch

 # cat /boot/config.txt

disable_overscan=1
hdmi_drive=2
dtparam=audio=on
display_rotate=3
disable_splash=1

 # cat /boot/cmdline.txt

dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=0xxxxxxxx-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait splash plymouth.ignore-serial-consoles
```
---
However, this is not a new issue. Already back in 2014 people reported this.  
But the issues were closed but never really resolved!  
For example, some related issues are: #608 and #624 and possibly many more.

popcornmix, 2018-04-16:
Raspbian is now on 4.14 kernel (you can get it with apt upgrade).
I'm not saying the issue is likely to be fixed there, but it's worth trying the update before investigating further.

E3V3A, 2018-04-17:
Yep, that's how I found out the spam issues, because now I can't see anything from my kernel logs anymore, since any rare and relevant error is now rotated out of existence due to #2512.

ThomasKaiser, 2018-04-18:
@E3V3A in your other issue you're happily reporting your system being that underpowered that the voltage drops below 4.63V even without **no load at all**. What about fixing this basic hardware issue first and then see whether this affects the other 'issues' you run into? 

BTW: Is this 'sound card' powered by the Pi or separately?

E3V3A, 2018-04-24:
It's interesting to note that the last logged *dmesg* item in #2139 is the same as the message I got here.

ThomasKaiser, 2018-04-24:
> It's interesting to note...

LOL! It's interesting to note: 

* you're using a host powered USB device (USB specs tell you: this thing needs at least 4.75V)
* [you are permanently and severly undervolting your Pi](https://github.com/raspberrypi/linux/issues/2512) (treshold is lower than 4.63V so it's even less at the USB ports)
* in the meantime you even ignore this warning just to ensure you run in all sorts of weird troubles that look like software issues but are most probably just the result of underpowering your board

The web is full of `FIQ reported NYET. Data may have been lost` hits. All originating from Raspberry Pis. And a lot of reports mention also general instabilities. Now with the undervoltage logging in place fortunately a lot of these users can find out whether their 'USB problems' are in reality just underpowering problems they were not aware of before (if they're not plagued with ignorance of course :) )

JamesH65, 2018-06-26:
@E3V3A Is this still an issue? Did sorting out the power supply fix the problem or is it still there even with a stable and reliable supply?

closed 2019-07-31 by JamesH65