raspberrypi/linux #2511: apt-get update freeze system after upgrade (RPi3)

E3V3A, 2018-04-16:
After successful upgrade of system (via `sudo apt-get upgrade`) and reboot, running `sudo apt-get updates` stalls and freeze system, mouse pointer still respond, but nothing else works or responds. Power unplug reboot necessary.

```bash
# sudo apt-get update

Hit:1 https://deb.nodesource.com/node_9.x stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Get:3 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:4 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Get:5 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Contents (deb) [35.5 MB]
82% [4 Packages store 0 B] [5 Contents-armhf 27.7 MB/35.5 MB 78%]
82% [4 Packages store 0 B] [5 Contents-armhf 27.7 MB/35.5 MB 78%]^C^C
```

pelwell, 2018-04-16:
1. Was this a one-off, i.e. what happens if you retry?
2. Does `journalctl --system` show any messages for that time period?
3. If you are experiencing under-voltage (#2512) then it might be affecting USB integrity, which could explain a networking failure. Try getting a better power supply and moving USB peripherals onto a powered hub.

E3V3A, 2018-04-16:
Hi @pelwell 
There is nothing wrong with my PSU. It's been running on medium to heavy load for months without any problem. I've updated half dozen times before, and never experienced this problem. This happened now just after updating. So there is some kind of other regression (apart from the other issue I raised.) 

Although a second run worked, I noticed the process that seem slow and prone to stalling is:
```
_apt      5538 79.4  0.5   8988  5148 pts/0    R+   12:45   1:07 /usr/lib/apt/methods/store
``` 
So perhaps its just your servers, not always responding?  
Either way, the system should not block CTRL-C (and everything else) from exiting the update, even if that is a bad thing to do.

JamesH65, 2018-04-16:
Can you answer Phil's questions number 1? Does it do it all the time, or was its just the first boot after the update?

E3V3A, 2018-04-17:
Just after first boot after update.

pelwell, 2018-04-17:
Any comments on 2 and 3?

E3V3A, 2018-04-18:
@pelwell 

> Any comments on 2 and 3?

1. It was a one-time occurence only after the kernel updates. After a 2nd reboot it worked. 
2. Since device access was completely disabled by this, there was no way to check `journalctl --system`. So no there was nothing in logs after reboot. 
3. I'm using WiFi, and AFAICR it is not using the USB bus. And I have power control disabled in kernel. `dwc_otg.lpm_enable=0`

wocka-wocka, 2018-04-25:
Similar to the above, I have a brand new RPi3B+ ... I've loaded raspian and I find I am unable to update it because "apt-get update" results in the system hanging completely.  I can't use apt-get to update or install *any* package.  Every attempt results in something like this displayed, followed by the machine completely freezing;

root@raspberrypi:~# apt-get update
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.3 kB]
Get:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease [15.0 kB]
0% [1 InRelease gpgv 25.3 kB]                                         

This is a brand new system with a 2.5 Amp USB Power Supply.  Wifi has not been enabled.  I simply took it out of the box, popped a configured MicroSD card in and then tried to load a few packages.  I've tried raaspbx with identical results.

JamesH65, 2018-04-26:
Its not the latest kernel, but fairly new.  Might be worth trying the latest Raspbian on a new SD card to see if the error still occurs. Could be the SDRAM frequency issue, you could try adding sdram_freq=450 to the config.tx file.

wocka-wocka, 2018-05-13:
FWIW, I replaced the Rpi3B+ hardware and found my problems went away.

paulwratt, 2020-05-23:
this is still an active problem, just thjat most people wont see it.
I am using last Stretch on RPi3 and it is very noticable
I am using latest Buster on RPi4 and it is there but not as noticable

My understanding of this issue and the way repos work is that; "store" is (remotely) added to the list that (manually) contains "main contrib non-free rpi". it is the defunct RPi Store repo. that "folder" still exists, but it does NOT contain **an empty repo**. and that is the reason for the long timeout

it is a P I T A if you have to deal with console only and script automation (it can take upto 5 mins to timeout)