raspberrypi/linux #2515: Found 4 unpatched upstream vulnerabilities in ext4

chan0415, 2018-04-17:
While scanning for unpatched  vulnerabilities in opensource projects. I found  4 unpatched upstream vulnerabilities in ext4 module.

* **CVE-2018-1092**
Description: The `ext4_iget` function in `fs/ext4/inode.c` mishandles the case of a root directory with a zero i_links_count, which allows attackers to cause a denial of service (ext4_process_freed_data NULL pointer dereference and OOPS) via a crafted ext4 image.
upstream patch: https://github.com/torvalds/linux/commit/8e4b5eae5decd9dfe5a4ee369c22028f90ab4c44
RedHat bug report: https://bugzilla.redhat.com/show_bug.cgi?id=1560777

* **CVE-2018-1093**
Description: The `ext4_valid_block_bitmap` function in `fs/ext4/balloc.c` allows attackers to cause a denial of service (out-of-bounds read and system crash) via a crafted ext4 image because balloc.c and ialloc.c do not validate bitmap block numbers.
upstream patches: https://github.com/torvalds/linux/commit/7dac4a1726a9c64a517d595c40e95e2d0d135f6f
RedHat bug report: https://bugzilla.redhat.com/show_bug.cgi?id=1560782

* **CVE-2018-1094**
Description: The `ext4_fill_super` function in `fs/ext4/super.c`  does not always initialize the crc32c checksum driver, which allows attackers to cause a denial of service (ext4_xattr_inode_hash NULL pointer dereference and system crash) via a crafted ext4 image.
upstream patches: https://github.com/torvalds/linux/commit/a45403b51582a87872927a3e0fc0a389c26867f1
https://github.com/torvalds/linux/commit/18db4b4e6fc31eda838dd1c1296d67dbcb3dc957
RedHat bug report: https://bugzilla.redhat.com/show_bug.cgi?id=1560788

* **CVE-2018-1095**
Description: The `ext4_xattr_check_entries` function in `fs/ext4/xattr.c` does not properly validate xattr sizes, which causes misinterpretation of a size as an error code, and consequently allows attackers to cause a denial of service (get_acl NULL pointer dereference and system crash) via a crafted ext4 image.
upstream patch: https://github.com/torvalds/linux/commit/54dd0e0a1b255f115f8647fc6fb93273251b01b9
RedHat bug report: https://bugzilla.redhat.com/show_bug.cgi?id=1560793

popcornmix, 2018-04-17:
I'm sure these will appear in upstream 4.14 tree in due course, and we'll have kernel builds with them in soon after.
Cherry picking upstream commits (which may not have been tested on older kernels) has some risk, so we'd prefer to wait until the kernel maintainers backport the commits.

Ferroin, 2018-04-17:
Also, not to downplay the severity of security issues, but all those CVE's require either a social engineering attack, or physical access to the system to exploit, which would (in theory at least) make them somewhat lower priority fixes.

nathanchance, 2018-04-25:
These are all fixed in 4.14.36 except for CVE-2018-1093, which caused a regression and has [a pending fix](https://patchwork.ozlabs.org/patch/903359/).

* https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/commit/?h=linux-4.14.y&id=8e0e94683f8449f4e83b4b563b80eb9c76b9e18f
* https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/commit/?h=linux-4.14.y&id=26dbb30c58ffb85bc015bd5e58831483d50f7d18
* https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/commit/?h=linux-4.14.y&id=b2623d8166563a19bee0cc8292e4ec951a42a6da
* https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/commit/?h=linux-4.14.y&id=e7793f2a2ac85f912690a69f7964907e31731ae7

JamesH65, 2018-05-10:
If I am reading patchwork correctly, all four are now accepted, so can this be closed?

nathanchance, 2018-05-10:
Yes, all fixes are now present in rpi-4.14.y.

* b39430ea068797bb45b72429db3743064280b1be and ae0db58dabe5463dcbcce095e83177536ed22d6a
* 8e0e94683f8449f4e83b4b563b80eb9c76b9e18f
* 26dbb30c58ffb85bc015bd5e58831483d50f7d18
* b2623d8166563a19bee0cc8292e4ec951a42a6da
* e7793f2a2ac85f912690a69f7964907e31731ae7