raspberrypi/linux #2516: brctl kernel internal errror

thecalle, 2018-04-18:
**4.14.30-v7+ #1102**
installed: bridge-utils (1.5-13+deb9u1) (apt-get -yf install **bridge-utils**)
run: brctl addbr br0 , OK
run **brctl addif br0 eth0**
effect:

Message from syslogd@raspberrypi at Apr 18 04:51:07 ...
**kernel:[ 2181.018431] Internal error: Oops: 17 [#1] SMP ARM**

Message from syslogd@raspberrypi at Apr 18 04:51:07 ...
kernel:[ 2181.066793] Process brctl (pid: 3206, stack limit = 0xb6b36210)

Message from syslogd@raspberrypi at Apr 18 04:51:07 ...
kernel:[ 2181.070141] Stack: (0xb6b37d28 to 0xb6b38000)

Message from syslogd@raspberrypi at Apr 18 04:51:07 ...
kernel:[ 2181.073476] 7d20: b9daf000 00000000 b6b37d5c b6b37d40 8058784c 8057fe78

**system LOST. Need hardware reboot.**

pelwell, 2018-04-18:
1. Which model of Pi is this?
2. What does `uname -a` return?

thecalle, 2018-04-18:
Linux raspberrypi 4.14.30-v7+ #1102 SMP Mon Mar 26 16:45:49 BST 2018 armv7l GNU/Linux
Raspberry Pi 3 Model B+

pelwell, 2018-04-18:
That kernel pre-dates a fix I added to the lan78xx Ethernet driver that may prevent the crash. Please run `sudo rpi-update` to get the latest kernel, then retest.

thecalle, 2018-04-18:
rpi-update move the kernel to last, but no kernel-headers to build from sources.
I need to build the DW-131 driver non present in kernel.

pelwell, 2018-04-18:
You're in luck. Apparently we've just (today) updated the Raspbian kernel and kernel headers packages, so if you `sudo apt-get update` and reinstall them you should get a compatible pair including the Ethernet driver fixes.

thecalle, 2018-04-18:
Firmware upgraded and Bridge is working correctly.
Sadly no build in headers:
make[1]: *** /lib/modules/4.14.34-v7+/build: No such file or directory.  Stop.
Can't build my drivers.

thecalle, 2018-04-18:
of course. the /build folder is missing

thecalle, 2018-04-18:
**SOLVED:**
root@raspberrypi:~# apt-get upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  **libraspberrypi-bin libraspberrypi-dev libraspberrypi-doc libraspberrypi0 raspberrypi-bootloader** **raspberrypi-kernel**
6 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
**REBOOT**
**apt-get install raspberrypi-kernel-headers**
root@raspberrypi:/home/pi# dpkg -S /lib/modules/4.14.34-v7+/build
raspberrypi-kernel-headers: /lib/modules/4.14.34-v7+/build
At the end, I got the build DIR.

closed 2018-04-18 by thecalle