raspberrypi/linux #2522: Pi B+ can not be pinged or reached after a while

bill-orange, 2018-04-20:
Using the latest updated Raspien release with my shiny new B+, I have run into a problem. After a random number of hours of operation, the PI B+ is not reachable or pingable from other network computers. This is especially weird since, the offending PI can see other devices in the network and can see the internet. My other PIs and PCs do not have this problem. If I disconnect and reconnect WiFi the problem is resolved for a while. I have tried reducing processor and memory speed. It has made no difference. My power supplies meet mfg. recommendations. 

I suspect that this issue may be some sort of adverse interaction with my UMBEE TimeWarner router.  However, if this was exclusively a router problem, I would expect my PI Model Bs and PI zero Ws to also have this problem. I am running Raspian on two PI B+. Both have this problem.

Can anyone reproduce this problem?  I found one other user on another forum with identical symptoms but that still leaves this as a pretty rare event.

pelwell, 2018-04-20:
[ So this is a 3B+, and the problem is seen over WiFi? If so, please make the issue title more precise. ]

Are you saying that even while the Pi can ping out (and get replies, presumably), other devices (even ones it has pinged) can't ping it?

bill-orange, 2018-04-20:
Will do.  I will edit the post title once I am on a PC and not my phone. 

Yes, this is a 3B+. You restated my problem correctly. It is very strange. As well as not being able to ping the offending PIs, I can not connect with VNC or see an apache2 default web page. All the while the PIs can ping other devices and show web pages such as CNN.com.

JamesH65, 2018-04-20:
Fixed the title. 

Note that the Pi3B+ has a new Wifi chip, although the kernel driver is the same, the underlying firmware will be different - so could well be a weird interaction specific to the new chip.

P33M, 2018-04-20:
How are you resolving the Pi's IP? Hostname?

What happens if you try to ping the (known) IP address instead of trying to resolve the host?

bill-orange, 2018-04-20:
I am pinging the IP address.

pelwell, 2018-04-20:
This is baffling.
1. Can you try continuously pinging the Pi from another device to see if that keeps the "link" up?
2. If 1 is successful, try pinging from a third device (if you have one) after what would have been long enough to expect a failure.
3. If 1 isn't successful (and you have a third device) try leaving both pinging simultaneously and see if they fail at the same time.

JamesH65, 2018-04-20:
Bit of a long shot, but is the IP address on the Pi changing for some
reason? Perhaps the interface going down and coming back with a different
address?

bill-orange, 2018-04-20:
I can try the two device test. Regarding test one, I wrote a shell script that pings the offending PI three times every five minutes and saves the result along with the time in a log file. When the failure happens the log entry goes from ‘3’ to ‘0’. After resetting the WiFi on the offending PI the log starts recording ‘3’ again. 

I should be able to run the script on a spare PI Zero  W and get two machines pinging away at it.

bill-orange, 2018-04-20:
James, good thought, but the IP address is not changing.

mkreisl, 2018-04-20:
@bill-orange 
Having sometimes same problem here (@JamesH65 , @pelwell  - don't panic, its not a Pi3B+ issue here :grin: )
Sometimes I can not ping a device (it's not only a Pi, happens with all WLAN clients). So, going to the not responding device and ping from there the other other client - and voila, ping is working again. Probably WLAN router does not handle ARP packets correctly, but I can live with that. 

Maybe you have same issue

bill-orange, 2018-04-20:
Well, I have to admit that your problem is similar. In my situation the only way to restore ping-ability is to disconnect and reconnect WiFi or reboot. Even forcing a DHCP renewal does not do the trick. Also, in my sitution only the PI 3B+ are affected.  Bs and Zero Ws work fine. 

I agree that this does have something to do with ARP. I can not think of anything else that would cause a one directional problem.

rodizio1, 2018-04-22:
I'd try looking at the traffic with wireshark/tshark/tcpdump and check if you can see ARP requests and replies.

Could this be power-saving/DTIM related maybe?

bill-orange, 2018-04-22:
I will try looking at the ARPs. I would be surprised if this is a power saver issue since it’s occurance is random. 

It also does not appear, to my surprise, to relate to the new Broadcom chip. 

I plugged in a spare Pi Zero W.  It ran fine no problem. I did an update-upgrade and installed apache2 and it started exhibiting the problem. So, it’s either something to do with the 3-13-18 release of Raspian or something to do with Apach2. I stopped Apache2 with sysctrl and I am waiting to see if the problem recurres. 

Bill

bill-orange, 2018-04-24:
I completely uninstalled apache2 after the problem reoccurred with Apache stopped.  I am waiting to see if the problem reoccurs.

@rodizio1 I loaded Wireshark on a computer connected via Ethernet and set the filter in Wireshark to ARP.  Is that correct?  What exactly am I looking for.  Wow, that thing sure shows a lot of packets flowing.  No wonder the router light continuously flickers.

bill-orange, 2018-04-25:
The problem still reoccurred with apach2 completely uninstalled. This was the result I expected but it’s good to be sure. 

This narrows the problem down to the 3-13-18 Raspian release. Hardware, power and apache2 are elemenated.

JamesH65, 2018-04-25:
Just to eliminate it, can you confirm there are no voltage low warnings in dmesg or that you do not see the lightning bolt symbol at top tight of a connected display?

bill-orange, 2018-04-25:
No lightening bolts and no syslog or dmesg entries suggesting a power problem.  I am using a 2.5A supply.

JamesH65, 2018-04-25:
Hmm, bit of a loss at what to suggest. It is most odd, and clearly not common or we would be inundated with reports.

JamesH65, 2018-04-25:
Did you try a different router?

bill-orange, 2018-04-25:
Yes,  It did seem to increase the time to failure, but since this is a somewhat random, the increase could be a red herring. In any event, even if the router was partially the culprit, the problem started with 3-13-18 and there are plenty of UBEE routers out there.  I would not expect  Raspian 3-13-18 to have router dependent difficulties.

Trotter73, 2018-09-24:
Hi,

I was seeing similar things after I changed Router, my Pis could ping however I could not ping the Pis, for me IPV6 seemed to be the culprit. I disabled IPV6 completely by adding ipv6.disable=1 to cmdline.txt since then no issues...

The router I was seeing issues with was a BT HH6

Worth a go as its easy to trial....

vrodic, 2019-09-29:
Probably related to this forum post
https://www.raspberrypi.org/forums/viewtopic.php?t=32398

"Wifi ARP requests lost" - lists a possible workaround from 

> davidasailor26

For anyone that comes across this problem in the future: I still have no effective fix. For now, I'm working around it by adding a cron job to another host I have on the same subnet. Basically this cron job sends a broadcast ARP request to the Pi each minute. If no response is received, the job sends a deauthenticate packet to the Pi. The deauthenticate forces the Pi to reassociate with the AP, which takes about 2 seconds and fixes the issue. This does require that the other host have a wireless card capable of injecting raw packets. The script for the cron job is:
Code: Select all

I will reply with any further breakthroughs I have.
`if ! /sbin/arping -b -f -c 5 -I em1 [IP Addr of Pi] > /dev/null ; then /sbin/aireplay-ng -0 1 -a [AP BSSID] -c [MAC Addr of Pi] wlan0 > /dev/null ; /bin/date >> /var/log/fixPi ; fi`

and to these bugs at least

https://github.com/raspberrypi/linux/issues/2677

https://github.com/raspberrypi/linux/issues/2732

scyto, 2019-11-21:
I have same issue on both a pi4 and pi zero.
I took the card from the pi4 and put it in the pi zero.
I only seem to have the issue on 2.4ghz networks
When it happens if i have both wlan0 and eth0 connected both stop responding to traffic.
I noted eventually my pi was going off line when the DHCP lease expired (i had short lease of 34 mins).
I have a Windows Server 2019 DHCP server.
As the pi is up a pole i can't connect with monitor and keyboard to see if it is still routing traffic externally from it or not.

I have increased the lease time to see if the issue goes away or just happens on the new lease expiry of 1 day.

dasl-, 2020-09-14:
@bill-orange did you ever solve this problem? I have basically the same issue you have. Addressed in more detail here: https://www.raspberrypi.org/forums/viewtopic.php?t=218167#p1719192

I have been unable to solve the problem so far.

bill-orange, 2020-09-14:
Not in any satisfactory way. I have my PIs set up to reboot nightly. I switched routers. My WiFi network is less congested. I am running the latest Raspian. Taken all together, the problem is minimal. 

Hardly a satisfying solution, without a solid known cause.

scyto, 2020-09-14:
For me the fix was to hardcode a true static IP address on the pi, after that i never had issue again after my previous post; i am convinced there is an issue with how the pi does dhcp acks to offers that some APs don't like - i literally saw the AP dropping the packets because.  My thesis is because they were not unicast acks (i was sniffing both sides of the internal AP interface).

 (not a dhcp reservation, my pet peeve is folks who say a dhcp reservation is a static IP, it isn't, though i know everyone in linux land calls it that these days, i am too old and grumpy and remember when it was never called that ;-) )

dasl-, 2020-09-15:
I was able to do some wireshark packet captures to get some more information. I am still stumped, but maybe this information will be helpful.

### My network setup:
router: 192.168.1.1
raspberry pi: 192.168.1.6
laptop: 192.168.1.5
phone: 192.168.1.4

### Problem description
The pi is connected to the internet via wifi. 

After a period of time (days to weeks), I am unable to access the pi from any computer on my network:
* I am unable to ssh onto the pi,
* web pages hosted on the pi are inaccessible

The output of `arp -a` on my laptop during this issue might look like:
`? (192.168.1.6) at (incomplete) on en0 ifscope [ethernet]`

If I hook up a keyboard and monitor to the pi during this issue, I will see that the pi appears to be working fine. Nothing in `dmesg` or `/var/log/messages` etc. The output of `ip addr show`on the pi will [look like this](https://gist.github.com/dasl-/567d8718653daa10aee080082d633f7b). Furthermore, the pi is able to access the network. It can hit external websites, such as google.com, and it can ping hosts on my network. If I ping from the pi to my laptop, then suddenly my laptop is able to access the pi again.

That is, after running `ping 192.168.1.5` on the pi, my laptop can ssh onto the pi and hit webpages hosted on the pi. My phone still will not be able to access the pi though. But after the pi pings my phone, my phone will be able to access web pages on the pi also.

After doing this ping from pi -> laptop, the `arp -a` entry on my laptop for the pi is "fixed", no longer incomplete.

### wireshark captures
#### capture pair 1
Here's the output of running `sudo tshark -i wlan0 -f arp -w tshark_sync.out` on the pi:
[image]

As we can see, there are a couple of arp requests and responses from the pi to the router.

At the same time, I ran a packet capture on my laptop: `sudo tshark -i en0 -f arp -w tshark_sync.out`:
[image]

As we can see, there are a few arp request broadcasts for the pi's IP address (192.168.1.6) that are coming from my phone (192.168.1.4) as it tries to hit the web page hosted on the pi.

So this is interesting: the packet capture on my laptop saw the arp request broadcast from my phone, but the packet capture running simultaneously on the pi did not see this arp request broadcast.

#### capture pair 2
Next, I rebooted the pi to "fix" the messed up network on it (it will likely have issues again within a few days to weeks).

Here's the output of running `sudo tshark -i wlan0 -f arp -w tshark_sync.out` on the pi:
[image]

As we can see, the pi (192.168.1.6) received an arp request broadcast from my phone (192.168.1.4). The pi responded properly to this arp request.

At the same time, I ran a packet capture on my laptop: `sudo tshark -i en0 -f arp -w tshark_sync.out`:
[image]

Again, we can see my laptop saw the arp request broadcast from my phone.  In this example though, everything worked properly because the pi had been recently rebooted.

### conclusion
So I am still confused about what is going on here. It seems like the wifi on the pi stops receiving broadcast messages after some random amount of time. I am not sure how to debug this further.

pelwell, 2020-09-15:
Disabling power saving might help - "sudo iwconfig wlan0 power off" will do that. Note that this setting does not persist across a reboot - appending it to /etc/rc.local will cause it to run after every reboot:
```
...
iwconfig wlan0 power off
exit 0
```
(`sudo` is not required because /etc/rc.local is run as root).

You can check the status of power saving at any time with `iwconfig wlan0`:
```
pi@raspberrypi:~$ iwconfig wlan0
wlan0     IEEE 802.11  ESSID:"Pi Towers"
...
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=66/70  Signal level=-44 dBm
...
```

dasl-, 2020-09-15:
Thanks, I will try this and report in a weeks if it seems to have helped.

I thought power management was already disabled though? Based on some random things I was googling: https://www.raspberrypi.org/forums/viewtopic.php?t=194619
And the output in dmesg:
```
pi@pifi:~ $ dmesg -T | grep -i brcm
[Tue Sep 15 00:56:34 2020] brcmfmac: F1 signature read @0x18000000=0x15264345
[Tue Sep 15 00:56:34 2020] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[Tue Sep 15 00:56:34 2020] usbcore: registered new interface driver brcmfmac
[Tue Sep 15 00:56:35 2020] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[Tue Sep 15 00:56:35 2020] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2
[Tue Sep 15 00:56:38 2020] brcmfmac: power management disabled
```

Although sure enough, after running `sudo iwconfig wlan0 power off`, the output of `iwconfig wlan0` changes from `Power Management:on` to `Power Management:off`.

max17048, 2020-09-15:
> I thought power management was already disabled though? Based on some random things I was googling: https://www.raspberrypi.org/forums/viewtopic.php?t=194619

Tx, thats also my knowledge. But I have to run the rc.local-workaround (on 3B).

BTW, maybe iwconfig runs out of support in a couple of years. Use:
```
iw dev wlan0 set power_save off  # in rc.local to set and
iw dev wlan0 get power_save      # to check -> Power save: off
```

pelwell, 2020-09-15:
>> I thought power management was already disabled though? Based on some random things I was googling: https://www.raspberrypi.org/forums/viewtopic.php?t=194619
>
> Tx, thats also my knowledge. But I have to run the rc.local-workaround (on 3B).

It depends on the age of your kernel - the workaround to disable power management was removed in late January.

max17048, 2020-09-15:
> It depends on the age of your kernel - the workaround to disable power management was removed in late January.

Currently I use the actual kernel (Raspian-Lite) on all devices (3B, ZeroW, 4B). But after reboot, "iw dev wlan0 get power_save" replys "on".

dasl-, 2020-09-15:
fwiw, I have:
```
pi@pifi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

```
pi@pifi:~ $ cat /etc/rc.local
#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi

iwconfig wlan0 power off
exit 0
```

And after I reboot the pi, I see power management is off:
```
pi@pifi:~ $ iwconfig wlan0
wlan0     IEEE 802.11  ESSID:"PRISM_5G"
          Mode:Managed  Frequency:5.18 GHz  Access Point: 50:6A:03:C9:72:80
          Bit Rate=433.3 Mb/s   Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=62/70  Signal level=-48 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

max17048, 2020-09-15:
Yes, this works for me since years.

dasl-, 2020-09-19:
> Disabling power saving might help - "sudo iwconfig wlan0 power off" will do that. Note that this setting does not persist across a reboot - appending it to /etc/rc.local will cause it to run after every reboot:

Hi all, (cc @pelwell ), disabling power saving did not help. The bug has struck again, 3 days after I disabled power save. Confirmed with additional packet captures that the symptoms are exactly the same as I described in my first post ( https://github.com/raspberrypi/linux/issues/2522#issuecomment-692462125 ).

Any other ideas :S

pelwell, 2020-09-21:
You could try enabling the brcmfmac firmware logging, in case it is reporting a problem - add `brcmfmac.debug=0x100000` to cmdline.txt (keeping everything in a single line).

dasl-, 2020-09-21:
Thanks, I've done that and will report back next time this bug resurfaces.
```
pi@pifi:~ $ cat /boot/cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait brcmfmac.debug=0x100000
```

dasl-, 2020-09-26:
Hi @pelwell , the issues have manifested again exactly the same as I described in my first post ( https://github.com/raspberrypi/linux/issues/2522#issuecomment-692462125 ).

This time I had brcmfmac firmware logging enabled. I cannot find any logs in `dmesg` though. Is it possible that the firmware logging didn't get enabled properly?

Here is my [`dmesg` output](https://gist.github.com/dasl-/46bbfb5518c535ed05109be45bd683be).

Here is my [`/boot/cmdline.txt` contents](https://gist.github.com/dasl-/25c57ebe6c24783b5a582b6d2cdd3b64) (using `cat -E` to show line endings).

btory, 2021-03-01:
Using Arch here but my 3B+ started doing this after a system upgrade (kernel 4.14.x -> 5.10.x). It would consistently fail to respond to other devices on the same subnet over wifi. tcpdump showed that all ARP requests to it went unanswered.

Tried some things others suggested:
- Disabling power management
- Disabling IPv6 stack
- Disabling bluetooth
- Disabling WMM, ASPD on the AP

None of those made any difference. The two things that did work for me, individually, were:
1. Enabling promiscuous mode on the wifi interface (`ip link set wlan0 promisc on`).
    - Without promisc mode the 3B+ doesn't see ARP requests: `tcpdump -i wlan0 -p arp`
    - With promisc mode it sees the ARP requests and responds: `tcpdump -i wlan0 arp`
2. Adding static ARP entries for the 3B+ on other devices.

CandyAngel, 2021-09-08:
> @CandyAngel I will test your fix at the next opportunity. I am now running 3D printers with Klipper firmware. Klipper off-loads the "heavy lifting" of the PID loops and printer kinematics onto the PI rather than running these tasks on the printers microcontroller.

Me too. I came across this issue as I was setting my printers back up and migrating them to Klipper. I now have my Ender 3 and FLSun QQ running on Klipper! My FLSun QQ-S is next :)

> Clearly, a periodic reboot is not a option since this would rather dramatically abort the print. If the PI disappears from the network routine monitoring of the print is impossible. I assume the two lines you added go inside network={ }.

On my setup, those options are defined in `/etc/network/interfaces.d/bond` and `ifenslave` has been patched to include handling for the `all_slaves_active` option as it is missing from upstream. I believe you can set these options on module load too though e.g. `options bonding all_slaves_active=1 use_carrier=0`.

You can check if this will fix your issue by setting the options while the bond device is up:

```
echo 1 | sudo tee /sys/class/net/bond0/bonding/all_slaves_active
echo 0 | sudo tee /sys/class/net/bond0/bonding/use_carrier
```

(those paths are off the top of my head so might need adjusting). If your remote wifi clients can then talk to the Pi, it's just a case of getting those options set by whatever network configuration system you are using (`network = {}` sounds familiar but I can't recall which system uses it..)

> @CandyAngel is your fix only applicable when one is bonding eth and wlan on the pi? In my setup, I only used wifi (no hardwired ethernet connection to the pi).

Mhm. This only fixes the issue of wlan devices not being able to communicate with a device which has their eth/wlan bonded and eth is active. What happens is the AP gets a packet on the wifi and then tries to pass it on via the wifi which just gets ignored as it is received on an inactive slave. Oddly enough, I've had this issue for ages on my setup at home (every device that can have eth/wlan bonding, does) but it never really bothered me because I very rarely actually use wifi!

Are you seeing a similar issue where your Pi can connect out via wifi, but cannot be contacted by any other devices on the network (wired or wireless)?

dasl-, 2021-09-08:
>Are you seeing a similar issue where your Pi can connect out via wifi, but cannot be contacted by any other devices on the network (wired or wireless)?

Yes, that is what I am seeing. But two caveats:
1) I have not tested connecting to the pi from a wired device when this happens
2) I have not reproduced this scenario since wifi updates described in [this post](https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=291609) were released. My setup has changed and I no longer have a raspberry pi 3B+ in my house.

dasl-, 2022-05-22:
Hello everyone, and especially @bill-orange ! I have a potential solution to these wifi problems. Please see my [post on the forums](https://forums.raspberrypi.com/viewtopic.php?t=334876) for all the details.

TLDR: add `over_voltage=2` to your `/boot/config.txt` file and / or get a better power supply to prevent your pi from occasionally becoming unreachable on wifi.

I would be interested to hear from you all if setting `over_voltage=2` solves your wifi issues.

bill-orange, 2022-05-22:
I will test it on two PIs and let you know what happens in a few days.

bill-orange, 2022-05-23:
I have five PIs running full time, one weather server, one Kodi, and three 3D printers).  All are randomly unreachable.  I added `over_voltage=2` to two of them.  both are printers.  

Sadly, this morning both were unreachable, as is often the case.  All my PIs have more than adequate power supplies with 1A excess capacity.  

I can consistently reach the offending PIs by going to ADMIN>WiFi on my SPECTRUM gateway router and changing the WiFi channel back and forth.  In around a minute I can see the PIs again.  Other devices, such as TLink remote wall plugs, don't disconnect and are not troubled by this procedure. 

I suspect that the SPECTRUM router is at least part of the problem and that there is some sort of unwholesome interaction between it and PIs.

bill-orange, 2022-05-23:
I believe so but I will repeat the test just to be sure.

bill-orange, 2022-05-23:
Oops no need to repeat the test.  I remember now.  I shut down both PIs, remove their SD cards, modified the config files on a WIN10 PC, replaced the cards and restarted the PIs.

The altered config.txt file had to load.

Bill

marbles-pi, 2022-08-28:
> I can consistently reach the offending PIs by going to ADMIN>WiFi on my SPECTRUM gateway router and changing the WiFi channel back and forth. In around a minute I can see the PIs again. Other devices, such as TLink remote wall plugs, don't disconnect and are not troubled by this procedure.

I have exactly the same issue. I got it to working again becasue of your post about changing the WiFi channel.
I have a Pi 3B with standard Bullseye on it - I haven't tried updating it yet.

Did you discover anything new since your last post?

bill-orange, 2022-10-11:
Nope. I still have the problem. I have three Raspberry PIs running Klipper on three 3D printers.

Two of the three routinely become unreachable. The the PI located at the greatest distance from other WiFi devices does not. I wonder if this hints as to the cause.

btory, 2024-11-05:
Bumping this ancient issue because I found a solution.

My 3B+ wasn't responding to ARP requests via WiFi because systemd was assigning a generated MAC address to the WLAN interface. The generated MAC address didn't match the permanent hardware MAC address and this was causing ARP packets to be dropped unless the interface was in promiscuous mode. If you are experiencing this issue, make sure the MAC address hasn't been changed.

bill-orange, 2024-11-05:
Interesting and logical. How do you make the comparison?Sent from my iPhone

btory, 2024-11-06:
> How do you make the comparison?

`ip a` then check the output under your WLAN interface.

Bad:
```
link/ether XX:XX:XX:XX:XX:XX brd YY:YY:YY:YY:YY:YY permaddr ZZ:ZZ:ZZ:ZZ:ZZ:ZZ
```
Good:
```
link/ether ZZ:ZZ:ZZ:ZZ:ZZ:ZZ brd YY:YY:YY:YY:YY:YY
```