raspberrypi/linux #2523: aarch64: rpi3: reboot doesn't work, hangs indefinitely

kees, 2018-04-20:
(I can't re-open https://github.com/raspberrypi/linux/issues/1640 so I'm creating this one instead.)

When requesting a reboot, the system hangs after the kernel thinks it has done a reboot:

[ 4609.960504] reboot: Restarting system

I'm seeing this problem on aarch64 kernels. Happens with both upstream kernel (at least up to 4.15) and Ubuntu kernel. Currently running:

`Linux version 4.15.0-15-generic (buildd@bos02-arm64-019) (gcc version 7.3.0 (Ubuntu/Linaro 7.3.0-14ubuntu1)) #16-Ubuntu SMP Wed Apr 4 14:00:16 UTC 2018
`

Updating to latest firmware (as of raspberrypi/firmware@a154f21) didn't change anything.

kees, 2018-04-22:
That fixed it! I wonder why that isn't getting auto-loaded. (Also, what about _loading_ it fixes this? It's not shown as being "in use" in "lsmod"...)

kees, 2018-04-22:
dts has:
                watchdog@7e100000 {
                        compatible = "brcm,bcm2835-pm-wdt";
                        reg = <0x7e100000 0x28>;
                };

modinfo says:
# modinfo bcm2835_wdt | grep -i alias
alias:          of:N*T*Cbrcm,bcm2835-pm-wdtC*
alias:          of:N*T*Cbrcm,bcm2835-pm-wdt

If I explicitly add it to /etc/modules.conf "systemctl status systemd-modules-load" reports:
Jan 28 15:58:18 pie3 systemd-modules-load[389]: Module 'bcm2835_wdt' is blacklisted

Nothing I can find in /etc/modprobe.d nor /usr/lib/modules-load.d has it blacklisted. *scratch head* I assume this is a OS or systemd issue...

kees, 2018-04-22:
Found it in /lib/modprobe.d/blacklist_linux_4.15.0-15-generic.conf under the heading "# Autogenerated watchdog blacklist". I will open a bug against Ubuntu.

Sorry for the noise!

closed 2018-04-22 by kees

lategoodbye, 2018-04-22:
@kees I'm happy that you found the solution. Btw this isn't the best place to report upstream/aarch64 issues. Please use linux-rpi-kernel@lists.infradead.org next time.