raspberrypi/linux #2524: Transmission Remote GUI (package: transgui) does not work with stretch

agl7, 2018-04-22:
Hello,

when I start transgui under Raspbian(stretch) there is created a space in the tray but with no icon. and there is nothing after some time on the screen.

When I start transgui form a command line I get ca. 200 times the same error:
(transgui:10168): Gtk-CRITICAL **: IA__gtk_widget_realize: assertion 'GTK_WIDGET_ANCHORED (widget) || GTK_IS_INVISIBLE (widget)' failed

Best Regards
Andreas (agl7)

agl7, 2018-04-22:
What I forget: Under jessie all works well.

Andreas

JamesH65, 2018-04-23:
@spl237 Any thoughts, or nothing to do with us?

spl237, 2018-04-23:
Presumably the authors of Transmission haven't updated it for Stretch - nothing to do with us.

JamesH65, 2018-04-23:
As above, please report a bug to the transmission maintainers, as this is not something we deal with here.

closed 2018-04-23 by JamesH65

agl7, 2018-04-23:
The transgui-Projekt have nothing to do with the Transmission-suite. It it's own project and for Debian  Stretch also armfh-binaries produced.
I will contact the maintainer of transgui (Andreas Noteng <andreas@noteng.no>) in the Debian project. Hope he can help.

Transgui works without any problems on my LinuxMint-Notebook. (LinuxMint 18.3, x64)

With Raspbian(jessie) there are no problems, all works very well.

Andreas

JamesH65, 2018-04-23:
What kernel version does your Mint notebook run? This is all going to be down to versions I suspect.

agl7, 2018-04-23:
$ uname -a (on my Notebook "specht" with LinuxMint 18.3)
Linux specht 4.13.0-38-generic #43~16.04.1-Ubuntu SMP Wed Mar 14 17:48:43 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux