raspberrypi/linux #2527: failing to mark a gpio as an IRQ

dpaape, 2018-04-23:
on version 4.14.30 - evidently the gpio i/f underwent some changes since my previous version.  the crux of the matter is that .../drivers/gpio/gpiolib.c checks to make sure that a gpio is -not- marked as an output before flagging it as an IRQ.  If we're going to attempt a universal interface to something as varied as gpio, then checking pin direction is erroneous.  Specifically, if one is using a gpio as a mailbox interrupt, the pin must be set as an output.  Could the check in gpiochip_lock_as_irq() be eliminated?

Dan

dpaape, 2018-04-23:
my previous version is irrelevant.  the change was to code that now prevents me from requesting an irq via a gpio device tree property and probably not gpiolib.c at all.  however, now i am forced to use gpio lib.  bottem line - all rabbit trails aside, why would a gpio need to be marked as an input in order to be used as an irq?  I will attempt to remedy with "upstream" as you call it.
thank you for the link
Dan

JamesH65, 2018-04-23:
It's not just us who call the mainline kernel development `upstream'. It's standard terminology for the mainline kernel development - i.e. that overseen by Linus Torvalds. ie THE canonical Linux.

pelwell, 2018-04-23:
Attacking the problem from a different direction, can you not enabling the interrupt while the pin is an input, then change the direction afterwards?

(And it turns out this issue has been raised before: https://github.com/raspberrypi/linux/issues/808)

dpaape, 2018-04-24:
i did not intent to disparage when quoting upstream.  i wasn't sure entirely what it meant other than the developers at linux kernel.org.  re: phil's posts - yes i tried to use gpiod_direction_input() directrly prior to the request_irq() call with no success.  i'ma little miffed at the constraint since we had to resort to using gpio as a mailbox int (instantiated from a dma descriptor) because the dma ints on r-pi behaved so poorly.  was there a functional issue that necessitated this?  otherwise it seems a bit arbitrary.  i'm still attempting to contact linux-gpio@vger.kernel.org.  if no success reversing this decision, we will just have to modify our kernel to remove the offending check.  thank you all for the help.
Dan

pelwell, 2018-04-24:
Rereading my comments from three years ago - https://github.com/raspberrypi/linux/issues/808#issuecomment-73381021, https://github.com/raspberrypi/linux/issues/808#issuecomment-73381422 - perhaps my initial response in this thread could have been more sympathetic. The rpi- kernel branches already contain a number of patches to and reversions of bits of upstream policy code that we don't agree with. Given that this bit of code has been unchanged since 3.18, I'm prepared to take on the burden of maintaining a patch the disables these checks.

pelwell, 2018-04-24:
Noddy patch pushed to rpi-4.14.y.

dpaape, 2018-04-24:
thanks Phil
Dan

closed 2018-04-24 by dpaape