raspberrypi/linux #2528: Network booted RPI 3B+ modprobe bcm2835-v4l2 freezes machine

jonshouse1, 2018-04-23:
NFS root RPI3B+ 

I have a working card image, works with my v4l app, camera works fine.  

I made a network booting image of the same files, boots ok its just now it hard locks loading the module.  Modprobe of other modules is ok.

Linux raspberrypi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux

I can switch between booting form the sd card or the LAN, sdcard always works, LAN boot always locks if I try and load the module.
Even tried copying the modules to a tempfs mount point and using "modprobe -d", still locks up.

jonshouse1, 2018-04-23:
Just tried upgrading

root@raspberrypi:~# uname -a
Linux raspberrypi 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux
root@raspberrypi:~# modprobe bcm2835-v4l2

Still freezes machine, Keyboard unresponsive, does not ping etc etc...

jonshouse1, 2018-04-23:
Did an rpi-update, still same,  tried a couple of different PSU for good luck as well

```
root@raspberrypi:/boot# /opt/vc/bin/vcgencmd version
Apr 16 2018 18:21:52 
Copyright (c) 2012 Broadcom
version af8084725947aa2c7314172068f79dad9be1c8b4 (clean) (release)
root@raspberrypi:/boot# uname -a
Linux raspberrypi 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux
```

6by9, 2018-04-24:
This looks like a duplicate of #2335 (ignore the title).

pelwell, 2018-04-24:
Testing with a 3B+ booting from SD card but with an NFS root partition I see no problems - it just works.

Are you booting:
a) with no SD card, TFTP for bootcode.bin and start.elf, NFS root, 
b) bootcode.bin on SD card, then TFTP boot for start.elf and NFS root, or
c) firmware and kernel on SD card, root on NFS drive?

pelwell, 2018-04-24:
Now tested with a fully network-booted 3B+ - again, no problems.

6by9, 2018-04-24:
With #2335 I did have it failing on a 2 or 3B with only bootcode.bin on the SD card and talking to PiServer.
I couldn't determine a reason for it stalling NFS, but not having called bcm_host_init resulted in the VCHI calls during bcm2835-v4l2 probe to have nasty side-effects. I'll see if I can fire up my PiServer install again and reproduce.

6by9, 2018-04-24:
Updated my PiServer from 4.9.xx to 4.14.34 (latest from apt).
Reproducable on a 3B IFF you have the camera plugged in. Without the camera it checks the number of cameras detected (0) and aborts. Most odd as both bits use the same MMAL framework.

jonshouse1, 2018-04-24:
> > Are you booting:
> a) with no SD card, TFTP for bootcode.bin and start.elf, NFS root,

No SD card.

I have firmware in  a directory with the name of the Pi serial number "25bcbfa6"

`root@mail:/disks/vol1/tftpboot/25bcbfa6# md5sum *
e3ab88859237eeb7f8deb8ef9f9af2bd  bcm2708-rpi-0-w.dtb
3501aebf7f08798dfa44f38422b30550  bcm2708-rpi-b.dtb
8154d0a768147413019fa98c18ac9398  bcm2708-rpi-b-plus.dtb
b07a7eccd032e98e5ee866c02f415f84  bcm2708-rpi-cm.dtb
4c73875b9db6a1e55824e3b3f005406f  bcm2709-rpi-2-b.dtb
a81f9a6a9bbc7b83209512021c9c248b  bcm2710-rpi-3-b.dtb
c2e9f69eff3551644a600721eed4ea07  bcm2710-rpi-3-b-plus.dtb
5d0680a605e98ba5ff7e3716ecab27d0  bcm2710-rpi-cm3.dtb
17efaf1c1ef89289168d71cdc8194982  bootcode.bin
935439b96482e274890752e0a4c16946  cmdline.OLD
0875c9770bb841b69915a74a82e4eb3d  cmdline.txt
e3f0a57ceef109eb13d94df7f8b95132  config.txt
d7810fab7487fb0aad327b76f1be7cd7  COPYING.linux
253315e7f9ca4b64dbe69a62f2456c50  fixup_cd.dat
da5a8b0ad03be4dc6deaad85eca8e52d  fixup.dat
59775794d283ab35d83532fc8fdefe46  fixup_db.dat
c1b836cccef2a337f8e236154c5d15eb  fixup_x.dat
bdc6492254aec36a53f90506a22ba6a9  issue.txt
2926f6994558d63e70d6f3999a082a90  kernel7.img
d092c0f9628577f8970adbf7eecc49c1  kernel.img
4a4d169737c0786fb9482bb6d30401d1  LICENCE.broadcom
bcd91fdfffb0cf7518016189fca363ee  LICENSE.oracle
md5sum: overlays: Is a directory
e203c4d048030f412acbf7034dc4886a  start_cd.elf
c8c3486368d8be92ffe1673eccc68911  start_db.elf
8a8e2c3cf9b6ce177e50f748eb91e816  start.elf
9fc019c14638acd381f9027317cc855a  start_x.elf
`
Root file system is then via NFS.

Board is a Pi 3B+, yes with the camera attached.  The idea is to have NFS booted cameras, I have purchases 10 camera modules so it would be great if this could be sorted.  I assure the issue is real and can provide unexciting video if needs be.
Thanks, Jon.

6by9, 2018-04-24:
pelwell's just confirmed that he had no camera attached, so that sort of confirms my earlier debug.

I have seen a little mismatch in resource allocation/free, but don't see how that would cause these issues. It's odd as there isn't anything fundamentally different between SD card and network booting.
I've currently broken my kernel build of the V4L2 driver totally so that it won't load, therefore need to fix that first :-(

pelwell, 2018-04-24:
I can also confirm that it wedges with a camera attached.

With some VCHIQ logging enabled at the same I see the following kernel logging:
```
[  123.209851] media: Linux media interface: v0.10
[  123.245878] Linux video capture interface: v2.00
[  123.279314] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[  123.280499] vchiq: Open Service mmal SrcPort:2
[  123.280526] vchiq: Sent Msg OPEN(2) to ???? s:2 d:0 len:12
[  123.280588] vchiq: Rcvd Msg OPENACK(3) from mmal s:86 d:2 len:2
[  123.280597] vchiq: Rcvd: 00000000: 10 00                                           ..
[  123.280647] vchiq: Sent: 00000000: 6d 6d 61 6c 04 00 00 00 00 00 00 00 01 00 00 00 mmal............
[  123.280654] vchiq: Sent Msg DATA(5) to mmal s:2 d:86 len:160
[  123.281024] vchiq: Rcvd Msg DATA(5) from mmal s:86 d:2 len:44
[  123.281039] vchiq: Rcvd: 00000000: 6d 6d 61 6c 04 00 00 00 44 ce 3d 3f 01 00 00 00 mmal....D.=?....
[  123.281098] vchiq: Sent: 00000000: 6d 6d 61 6c 08 00 00 00 6d 6d 61 6c 02 00 00 00 mmal....mmal....
[  123.281103] vchiq: Sent Msg DATA(5) to mmal s:2 d:86 len:36
[  123.281155] vchiq: Rcvd Msg DATA(5) from mmal s:86 d:2 len:316
[  123.281168] vchiq: Rcvd: 00000000: 6d 6d 61 6c 08 00 00 00 44 ce 3d 3f 02 00 00 00 mmal....D.=?....
[  123.281361] vchiq: Sent: 00000000: 6d 6d 61 6c 0f 00 00 00 00 00 00 00 03 00 00 00 mmal............
[  123.281365] vchiq: Sent Msg DATA(5) to mmal s:2 d:86 len:40
[  123.281432] vchiq: Rcvd Msg DATA(5) from mmal s:86 d:2 len:180
[  123.281447] vchiq: Rcvd: 00000000: 6d 6d 61 6c 0f 00 00 00 44 ce 3d 3f 03 00 00 00 mmal....D.=?....
[  123.281484] vchiq: Sent: 00000000: 6d 6d 61 6c 05 00 00 00 00 00 00 00 04 00 00 00 mmal............
[  123.281488] vchiq: Sent Msg DATA(5) to mmal s:2 d:86 len:28
[  123.281606] vchiq: Rcvd Msg DATA(5) from mmal s:86 d:2 len:28
[  123.281628] vchiq: Rcvd: 00000000: 6d 6d 61 6c 05 00 00 00 44 ce 3d 3f 04 00 00 00 mmal....D.=?....
[  123.281809] vchiq: Open Service mmal SrcPort:3
[  123.281818] vchiq: Sent Msg OPEN(2) to ???? s:3 d:0 len:12
[  123.281866] vchiq: Rcvd Msg OPENACK(3) from mmal s:88 d:3 len:2
[  123.281873] vchiq: Rcvd: 00000000: 10 00                                           ..
[  123.281913] vchiq: Sent: 00000000: 6d 6d 61 6c 04 00 00 00 00 00 00 00 01 00 00 00 mmal............
[  123.281917] vchiq: Sent Msg DATA(5) to mmal s:3 d:88 len:160
[  123.283885] WARN::dwc_otg_hcd_handle_hc_fsm:2458: Unexpected IRQ state on FSM transaction:dev_addr=4 ep=2 fsm=6, hcint=0x00000002
[  123.315864] WARN::dwc_otg_hcd_handle_hc_fsm:2458: Unexpected IRQ state on FSM transaction:dev_addr=4 ep=2 fsm=6, hcint=0x00000002
[  123.331860] WARN::dwc_otg_hcd_handle_hc_fsm:2458: Unexpected IRQ state on FSM transaction:dev_addr=4 ep=1 fsm=6, hcint=0x00000002
[  125.450280] vchiq: Rcvd Msg DATA(5) from mmal s:88 d:3 len:44
[  125.450337] vchiq: Rcvd: 00000000: 6d 6d 61 6c 04 00 00 00 f4 82 3d 3f 01 00 00 00 mmal......=?....
[  125.450431] vchiq: Sent: 00000000: 6d 6d 61 6c 08 00 00 00 6d 6d 61 6c 02 00 00 00 mmal....mmal....
[  125.450442] vchiq: Sent Msg DATA(5) to mmal s:3 d:88 len:36
[  125.450510] vchiq: Rcvd Msg DATA(5) from mmal s:88 d:3 len:316
[  125.450533] vchiq: Rcvd: 00000000: 6d 6d 61 6c 08 00 00 00 f4 82 3d 3f 02 00 00 00 mmal......=?....
[  125.450590] vchiq: Sent: 00000000: 6d 6d 61 6c 08 00 00 00 6d 6d 61 6c 03 00 00 00 mmal....mmal....
[  125.450600] vchiq: Sent Msg DATA(5) to mmal s:3 d:88 len:36
[  125.450656] vchiq: Rcvd Msg DATA(5) from mmal s:88 d:3 len:316
[  125.450680] vchiq: Rcvd: 00000000: 6d 6d 61 6c 08 00 00 00 f4 82 3d 3f 03 00 00 00 mmal......=?....
[  125.450735] vchiq: Sent: 00000000: 6d 6d 61 6c 08 00 00 00 6d 6d 61 6c 04 00 00 00 mmal....mmal....
[  125.450744] vchiq: Sent Msg DATA(5) to mmal s:3 d:88 len:36
[  125.450815] vchiq: Rcvd Msg DATA(5) from mmal s:88 d:3 len:316
[  125.450838] vchiq: Rcvd: 00000000: 6d 6d 61 6c 08 00 00 00 f4 82 3d 3f 04 00 00 00 mmal......=?....
[  125.450896] vchiq: Sent: 00000000: 6d 6d 61 6c 08 00 00 00 6d 6d 61 6c 05 00 00 00 mmal....mmal....
[  125.450905] vchiq: Sent Msg DATA(5) to mmal s:3 d:88 len:36
[  125.450961] vchiq: Rcvd Msg DATA(5) from mmal s:88 d:3 len:316
[  125.450986] vchiq: Rcvd: 00000000: 6d 6d 61 6c 08 00 00 00 f4 82 3d 3f 05 00 00 00 mmal......=?....
[  125.451041] vchiq: Sent: 00000000: 6d 6d 61 6c 08 00 00 00 6d 6d 61 6c 06 00 00 00 mmal....mmal....
[  125.451050] vchiq: Sent Msg DATA(5) to mmal s:3 d:88 len:36
[  125.451106] vchiq: Rcvd Msg DATA(5) from mmal s:88 d:3 len:316
[  125.451129] vchiq: Rcvd: 00000000: 6d 6d 61 6c 08 00 00 00 f4 82 3d 3f 06 00 00 00 mmal......=?....
[  125.451187] vchiq: Sent: 00000000: 6d 6d 61 6c 0e 00 00 00 54 7a 52 b7 07 00 00 00 mmal....TzR.....
[  125.451196] vchiq: Sent Msg DATA(5) to mmal s:3 d:88 len:80
[  125.451305] vchiq: Rcvd Msg DATA(5) from mmal s:88 d:3 len:28
[  125.451328] vchiq: Rcvd: 00000000: 6d 6d 61 6c 0e 00 00 00 f4 82 3d 3f 07 00 00 00 mmal......=?....
[  125.451379] vchiq: Sent: 00000000: 6d 6d 61 6c 0f 00 00 00 00 00 00 00 08 00 00 00 mmal............
[  125.451388] vchiq: Sent Msg DATA(5) to mmal s:3 d:88 len:40
[  125.451500] vchiq: Rcvd Msg DATA(5) from mmal s:88 d:3 len:92
[  125.451523] vchiq: Rcvd: 00000000: 6d 6d 61 6c 0f 00 00 00 f4 82 3d 3f 08 00 00 00 mmal......=?....
[  125.451575] vchiq: Sent: 00000000: 6d 6d 61 6c 0e 00 00 00 00 00 00 00 09 00 00 00 mmal............
[  125.451584] vchiq: Sent Msg DATA(5) to mmal s:3 d:88 len:44
[... cut ...]
[  125.473226] vchiq: Rcvd: 00000000: 6d 6d 61 6c 06 00 00 00 f4 82 3d 3f 50 00 00 00 mmal......=?P...
[  125.473288] bcm2835-v4l2: Broadcom 2835 MMAL video capture ver 0.0.2 loaded.
[  134.630326] nfs: server 10.3.14.8 not responding, still trying
[  137.990337] lan78xx 1-1.1.1:1.0 eth0: Failed to read stat ret = 0xffffff92
[  152.790326] nfs: server 10.3.14.8 not responding, still trying
[  155.910319] lan78xx 1-1.1.1:1.0 eth0: Failed to read stat ret = 0xffffff92
```
Note the (presumable fatal) dwc_otg messages, and the subsequent NFS death.

jonshouse1, 2018-04-24:
> It's odd as there isn't anything fundamentally different between SD card and network booting.
I am not a kernel hacker for linux but I do write embedded firmware, for my 2c worth I suspect it is either an assumption about ISR order or frequency.

jonshouse1, 2018-04-28:
Any update on this ? 

I did an rpi-update, now running 
Linux raspberrypi 4.14.37-v7+ #1111 SMP Thu Apr 26 13:56:59 BST 2018 armv7l GNU/Linux

This is much the same, except it returns the prompt before crashing after "modprobe bcm2835-v4l2" .  PS this new kernel/firmware version is slightly less reliable NFS booting than before. Sometimes stops on the IP details summary and fails to continue..... odd ...

Thanks,
Jon

jonshouse1, 2018-05-11:
18 days zero comments.  Am I correct that the issue is within the Broadcom binary b******ks (tm)?  I am assuming this is not an issue I can investigate ?

6by9, 2018-05-11:
Sorry, busy on other issues/tasks. It got an initial triage and is now in the queue to investigate.

> Am I correct that the issue is within the Broadcom binary b******ks (tm)? I am assuming this is not an issue I can investigate ?

Incorrect as noted in https://github.com/raspberrypi/linux/issues/2335#issuecomment-360093829
Anything calling bcm_host_init before the bcm2835-v4l2 module is loaded appears to resolve the issue. Source for bcm_host_init is in the [normal place for userland code](https://github.com/raspberrypi/userland/blob/master/host_applications/linux/libs/bcm_host/bcm_host.c#L83) if you wished to investigate exactly which bit is making the module happy.

jonshouse1, 2018-05-11:
>  if you wished to investigate exactly which bit is making the module happy
Ok, I will have a look, what is the best place to browse the source of the specific RPI shipped version of the "bcm2835-v4l2" module ?

I don't know how the vc code interacts with the kernel code.
I note that  bcm_host_init() is protected from multiple init calls "if (initted) return;"  so a hack (rather than a true fix) would be to arrange it that the module called bcm_host_init() before it started 'real work' and see if that improves it.  I need to read up on getting enough of a build environment to modify the module.....

jonshouse1, 2018-05-11:
```
// gcc bcmhi.c -I/opt/vc/include -L/opt/vc/lib -lbcm_host -o bcmhi 

int main ()
{
	bcm_host_init();
}

```

Compiled the following as "bcmhi", when run before the modprobe on the NFS booted machine the module loads ok.

I noticed this on the debug output above. Doesn't the CSI interface depend on an ISR resident in the vc code ?

[  123.283885] WARN::dwc_otg_hcd_handle_hc_fsm:2458: Unexpected IRQ state on FSM transaction:dev_addr=4 ep=2 fsm=6, hcint=0x00000002

Thanks,
Jon

jonshouse1, 2018-05-11:
> ??! How did you get from an error message from the USB driver (dwc_otg) to worrying over interrupts from the camera?

Largely by failing to understand what I am reading, as I said I am not a linux kernel hacker, plus we seem to differ in terminology, to me the "ISR" is the code called when the IRQ is triggered, the interrupt control logic would no relevance to that? I will keep reading.... last time I did anything Linux kernel related it ran on a 486, my "Understanding the Linux Kernel" book is dated 2001 :-)

6by9, 2018-05-11:
> I assume that is as far down as I see along that path as I cant find source for vchiq_initialise()?

https://github.com/raspberrypi/userland/blob/master/interface/vchiq_arm/vchiq_lib.c#L161

Although the first stage would be to copy/paste the implementation of bcm_host_init into your app to see how far through you need to go to make the module loading work correctly. Once you've done that, then start following the last required call down (or just report back as you'll have saved us a good half hour or so of debug).

jonshouse1, 2018-05-11:
[](http://jonshouse.co.uk/bcmhi.tar.gz)

I copied some bits around, no makefile "./compile" generates a binary "bcmhi"

In bcm_host.c i modified bcm_host_init(),  adding a printf.

I move the return down the function one call at a time re-booting between each attempt, reached pretty much the end of the function before the modprobe worked. 

Output looked liked this:
```

 /root@raspberrypi:/bcmhi# ./bcmhi 
 did vcos init()
 did vcos_initialise()
 did vchi_initialise()
 did vchi_create_connection()
 did vchi_connect()
 did vc_vchi_gencmd_init()
 did vc_vchi_dispmanx_init()
 did vc_vchi_tv_init()
 did vc_vchi_cec_init()
 did vc_gencmd()
 root@raspberrypi:/bcmhi# modprobe bcm2835-v4l2
 root@raspberrypi:/bcmhi# 

```

Not sure if that helps, I cant see source for vc_gencmd() so I assume that is as far as I can follow that line ?

jonshouse1, 2018-05-11:
Sorry left an absolute path in ./compile, should look like this:

`gcc bcm_host.c -I/opt/vc/include -I/opt/vc/include/interface/vmcs_host/  -L/opt/vc/lib -lvcos -lvchiq_arm -lrt -lm -lpthread -lbcm_host -lGLESv2 -DGLES2 `pwd`/libilclient.a  -o bcmhi`
`

jonshouse1, 2018-05-12:
Found vc_gencmd() ... 
https://github.com/raspberrypi/userland/blob/a1b89e91f393c7134b4cdc36431f863bb3333163/interface/vmcs_host/vc_vchi_gencmd.c

seems github search does not show the function definition in preference to callers... will egrerp -r locally instead from now on, me bad ...

6by9, 2018-05-24:
I've finally had a chance to dig into this. The minimum set of stuff from bcm_host_init is
```
   vchi_connect(&global_connection, 1, global_initialise_instance);
  
   vc_vchi_gencmd_init (global_initialise_instance, &global_connection, 1);

   if ( success == 0 )
   {
      success = vc_gencmd( response, sizeof(response), "set_vll_dir /sd/vlls" );
      vcos_assert( success == 0 );
   }
}
```
The camera component does query VLLs, so I suspect that something in the internal "filesystem" on the VPU is blowing up due to not having the VLL directory set sensibly. I'll trace it further in the firmware now.

6by9, 2018-05-29:
Firmware patch created to fix this issue. It'll be reviewed internally, but I'd expect it to be in the next rpi-update release.

6by9, 2018-06-27:
Workaround was merged 5 June, released via rpi-update 7 June.
@jonshouse1 Please test, otherwise this issue will get closed in 30 days.

JamesH65, 2018-06-27:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

jonshouse1, 2018-06-27:
I am trying to figure out how to update the NFS root image without backtracking to the original SD card image then copying it all back over.

I this super crap error (invalid hash of what?, from where? referenced by what ?)
```
root@raspberrypicamera:~# rpi-update 
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Invalid hash given
```

6by9, 2018-06-27:
You're on a network booted system so need to update the files that your TFTP server is handing out.

Go to https://github.com/Hexxeh/rpi-firmware, download the start*.elf files and fixup.* files, and put them in the directory that your TFTP server is using. Reboot the Pi. The firmware should now all be updated, and hopefully the issue resolved.

pelwell, 2018-06-27:
Alternatively you can (if your TFTP server is sharing its files) mount the TFTP server directory for your Pi on /boot and use rpi-update as usual.

jonshouse1, 2018-06-27:
"You're on a network booted system so need to update the files that your TFTP server is handing out."
Yes, I know :-)

I login to the NFS booted Pi via ssh - as default it is read only

I issue
  # mount -o remount rw /

Check it
root@raspberrypicamera:~# ls >x
root@raspberrypicamera:~# rm x

Yep, it is read/write ok.

My NFS root has a "/boot" this is the same contents as the firmware directory the Pi booted from
(on the NFS server itself)
root@mail:/disks/vol1/tftpboot/rpi_cctv/boot# cp -av ../../25bcbfa6/* .

Now rpi-update can see a normal read/write "/" and a true representation of "/boot" - in the past I simply ran rpi-update at this point and all was good, then copied the firmware back into the serial number directory.

jonshouse1, 2018-06-27:
> "Go to https://github.com/Hexxeh/rpi-firmware, download the start*.elf files and fixup.* files, and put them in the directory that your TFTP server is using"
Thanks, I will give that a go.... in case the overlays have changed I will try replacing all files with a new one and copying back config.txt and cmdline.txt as I think those are the only user settings ?

pelwell, 2018-06-27:
One thing to be aware of is that this won't update the bootcode.bin downloaded by the bootrom, which doesn't support the serial number subdirectories,

jonshouse1, 2018-06-27:
>  One thing to be aware of is that this won't update the bootcode.bin downloaded by the bootrom, which doesn't support the serial number subdirectories,

Sorry, I don't understand.  Isn't the bootcode.bin the same binary however it is loaded by the Pi... what does not support the serial number subdirectories?, you have lost me at this point.

pelwell, 2018-06-27:
When the boot ROM downloads bootcode.bin from the TFTP server (as a 3B+ can when booting with no SD card) it only looks for `bootcode.bin`, not `<serial>/bootcode.bin`, despite the fact that bootcode.bin itself will then search for `<serial>/start.elf` etc. This isn't a problem because the same image will run on all models and will be backwards compatible, but it may catch you out if you thought you were updating it when actually you were just overwriting `<serial>/bootcode.bin`.

jonshouse1, 2018-06-27:
"when actually you were just overwriting <serial>/bootcode.bin."
Great, I understand now, thanks :-)

Glad you mentioned that, I was looking at the ftfboot directory just now thinking I must delete some of these stray files....

jonshouse1, 2018-06-27:
Yay ...

Thanks :-)

```
root@raspberrypicamera:~# modprobe bcm2835-v4l2 
root@raspberrypicamera:~# /opt/vc/bin/vcgencmd version
Jun 21 2018 14:07:17 
Copyright (c) 2012 Broadcom
version 8885266d85a7c3c03894fcb03c9724020f9cd77e (clean) (release)

```

jonshouse1, 2018-06-27:
PS I uploaded the camera client/server code to github for safe keeping, bit rough at the moment, but works ok. It looks static on the video but is updating 14 frames a second.

My next job is top try and figure out how to write my date/time stamp over the camera frame then have the Pi jpeg encode the result. At the moment I use v4l to read the frame as YUV, convert to RGB, overlay the text then jpeg encode and send the image to the server.

Linux desktop viewer.
https://www.youtube.com/watch?v=ilA25ho_u-k&feature=youtu.be

The code,  server, client and desktop apps.
https://github.com/jonshouse1/jcamsys

6by9, 2018-06-27:
> My next job is top try and figure out how to write my date/time stamp over the camera frame then have the Pi jpeg encode the result. At the moment I use v4l to read the frame as YUV, convert to RGB, overlay the text then jpeg encode and send the image to the server.

Eek! Why do people hate YUV so much?
The camera can spit out RGB in a number of flavours (conversion done in vector code on the VPU), so no need to do the conversion in your app.
Hardware accelerated JPEG encode requires YUV, so it has to convert back from your RGB frame. Memory says there are some mechanisms for JPEG to contain RGB, but they are relatively uncommon.

It's a slight shame that you chose to use V4L2, as the camera subsystem can do basic annotations, as demonstrated in https://www.youtube.com/watch?v=WnnKrVSMbng. They're always top and centre at the moment as I haven't had the inclination to add the plumbing to move it around.

jonshouse1, 2018-06-27:
> It's a slight shame that you chose to use V4L2, as the camera subsystem can do basic annotations

Nice demo video :-)

I have an RPI specific version of the camera code, that use v4l at the moment but can work in any way the Pi is happy with.  It uses memory mapped I/O to read the state of the IR light level and to toggle the IRCUT filter on the camera module I am using, so it is non portable at that point:

https://github.com/jonshouse1/jcamsys/tree/master/jcamsys_camera_rpi/jcamsys_camera_rpi_v0.4

What I really need from an API is just a way to scribble into the framebuffer between the camera capture and some form of hrdware Jpeg encode.  Does any API exist for my to plug into the YUV buffer and still let the Pi hardware encode the jpeg.  I did look for demos/code and did not come up with a great deal.

I could not find any good examples of encoding a camera jpeg on the Pi video engine for example, I only want one jpeg at a time as I need to decide to send it or drop it depending on network conditions. I am using TCP and cant afford for the buffer to bloat to the point it becomes non real-time.

As for RGB24 rather than YUV, it's mostly because I can work with RGB in my head hence that is what most of my code references, I can probably work with YUV but only with several google pages open ;-)

jonshouse1, 2018-06-27:
It has been so long since I looked at this, I was wrong, the generic V4l code uses YUV if opening the device for RGB fails OR (and it is an important or) the size of the RGB image is wrong. Seems V4l has a fault on some cameras, claims it opens in RGB but does not, in that case the code 'falls back' to YUV and adds a conversion to RGB.
The code overlaying the text is RGB though and the Jpeg encode is via libjpeg rather than the v4l API.
See lines 672+

Video of configuration tool, changes server settings and also passes settings to any connected cameras.
https://youtu.be/8yGgJwwZ700

JamesH65, 2019-07-31:
Closing this issue as questions answered/issue resolved.

closed 2019-07-31 by JamesH65