raspberrypi/linux #2529: Unexpected interrupt: 8 on SC16IS752

hardikharpal, 2018-04-24:
Hi,

I am Interfacing SC16IS752(custom board) on SPI0 of raspberry pi 3B.I have latest stretch lite version as on 24-April-2018.
I created dts file and generated dtbo file and copied dtbo to /boot/overlay. Also edited /boot/config.txt to load overlay.
`/dts-v1/;
/plugin/;
/ {
        compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

        fragment@0 {
        target = <&spidev0>;
        __overlay__ {
            status = "disabled";
        };
    };

        fragment@1 {
                target = <&gpio>;
                __overlay__ {
                        spi0_pins: spi0_pins {
                                brcm,pins = <9 10 11>;
                                brcm,function = <4>; /* alt 0 */
                        };

                        spi0_cs_pins: spi0_cs_pins {
                                brcm,pins = <8>;
                                brcm,function = <1>; /* output */
                        };
                };
        };

        fragment@2 {
        target = <&spi0>;
        frag1: __overlay__ {
            #address-cells = <1>;
            #size-cells = <0>;
            pinctrl-names = "default";
            pinctrl-0 = <&spi0_pins &spi0_cs_pins>;
            cs-gpios = <&gpio 8 1>;
            status = "okay";

            sc16is752: sc16is752@0 {
                compatible = "nxp,sc16is752";
                reg = <0>; /* CE0 */
                clocks = <&sc16is752_clk>;
                interrupt-parent = <&gpio>;
                interrupts = <24 2>; /* IRQ_TYPE_EDGE_FALLING */
                gpio-controller;
                #gpio-cells = <2>;
                spi-max-frequency = <4000000>;

                sc16is752_clk: sc16is752_clk {
                   compatible = "fixed-clock";
                   #clock-cells = <0>;
                   clock-frequency = <14745600>;
                };
            };
        };
        };
        
        fragment@3 {
                target = <&aux>;
                __overlay__ {
                        status = "okay";
                };
        };

    __overrides__ {
      int_pin = <&sc16is752>,"interrupts:0";
    };
};`
My problem is when open port on minicom . IRQ pin remains low for all time and UART communication not working.
Getting below message on dmesg .

> [1293.489435] sc16is7xx spi0.0: ttySC0: Unexpected interrupt: 8
[ 1293.490328] sc16is7xx spi0.0: ttySC0: Unexpected interrupt: 0
[ 1293.490460] sc16is7xx spi0.0: ttySC0: Unexpected interrupt: 0

Getting bellow output for `sudo vcdbg log msg `

> 002047.676: dtparam: spi=on
002062.027: brfs: File read: 1630 bytes
002075.726: brfs: File read: /mfs/sd/overlays/sc16is752-spi0.dtbo
002097.865: Loaded overlay 'sc16is752-spi0'

So I tried with SC16IS750 (sparkfun breakout board) which has one UART channel. I have connected to same pins on raspberry pi and made changes(just replaced sc16is752 with sc16is750) to dts file and generated dtbo file copied to /boot/overlay and edited /boot/config.txt to load overlay.
And It works perfectly able to send data and receive data.

Am I missing something ? Any Hint will be Helpful.
I read here [https://raw.githubusercontent.com/janztec/empc-arpi-linux/rpi-3.18.y/drivers/tty/serial/sc16is7xx.c](url)
that sc16is7xx_startup() one comment is there

> /* ama 2016-12: fixed error message "unexpected interrupt: 8" in dmesg by added mdelay(1)
 without delay, after enabling the interrupts in IER, set_baud/set_termios is immediatly called, 
  enables enhanced register ("config mode") with LCR = 0xBF and the first interrupt occurs at
 the same time, resulting in reading the IIR interrupt status register in the wrong mode. 
 This problematic time window, from enabling the interrupts to handling them, is about 100µs, so a
 delay of 1000µs=1ms was choosen*/
mdelay(1);

But in case of raspberry pi driver for SC16IS7XX [https://github.com/raspberrypi/linux/blob/rpi-4.14.y/drivers/tty/serial/sc16is7xx.c](url) the delay is no there.

Do you think this is the cause of my issue? 

Thanks and Regards,
Hardik Harpal

6by9, 2018-04-25:
Can you verify that your SC16IS752 works on SPI1 as there is already an overlay for that which is known to work. There was work on the driver with #2241 in February that solved some problems with it.

Your random hack is from a 3.18 kernel - a lot has changed between 3.18 and 4.14.

pelwell, 2018-04-25:
I'm guessing the custom board has a 40-pin header, in which case changing the SPI bus is hard.

I do now have an SC16IS752 board, so I'll take a quick look.

pelwell, 2018-04-25:
I've not seen the "Unexpected interrupt: 8" message that you get, so I can't really test the patch. However, building a custom kernel doesn't take so long with a 3B or 3B+ - a few hours, not overnight. Follow our [kernel building guide](https://www.raspberrypi.org/documentation/linux/kernel/building.md), modifying the sc16is7xx driver by adding the delay, then see if the message goes away.

hardikharpal, 2018-04-25:
@pelwell  sure I will do that and I will update the result here. I am newbie to linux/raspberypi . 

I have query out of this issue: Is it possible to read/write GPIO0-GPIO7 of  SC16IS752?  How can I do that?

closed 2018-04-25 by hardikharpal

pelwell, 2018-04-25:
The various sc16is7xx overlays declare the devices as GPIO controllers, and the driver supports this feature, so Linux creates an entry in /sys/class/gpio (for me it's gpiochip504). Using this you can export and control each of the GPIO pins.

The sysfs GPIO mechanism is documented [here](https://www.kernel.org/doc/Documentation/gpio/sysfs.txt).

hardikharpal, 2018-05-03:
After Adding  `mdelay(1);`  "Unexpected interrupt: 8" is not coming when I open port.

pelwell, 2018-05-03:
Thanks for the feedback - I'll patch our kernel tree.

paulenuta, 2018-05-10:
Update:
I have resurrected an older pi installation and using same custom board with 2 x SC16IS752IBS on I2C bus I have no "Unexpected interrupt: 0" in dmesg.

```
root@raspberrypi:/home/pi# ./serial_write.py /dev/ttySC0
/dev/ttySC0  was setup

 Start with:  /dev/ttySC0
<
/dev/ttySC0  opened
@Write line:  0
 data was sent to printer /dev/ttySC0
/dev/ttySC0  closed
>

^CTraceback (most recent call last):
  File "./serial_write.py", line 81, in <module>
    time.sleep(2)
KeyboardInterrupt
root@raspberrypi:/home/pi# uname -a
Linux raspberrypi 4.9.33-v7+ #95 SMP Sun Jun 25 20:36:46 CEST 2017 armv7l GNU/Linux

root@raspberrypi:/home/pi# cat /etc/*release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
root@raspberrypi:/home/pi#

root@raspberrypi:/home/pi# dmesg -w
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.33-v7+ (peter@lamerx) (gcc version 4.8.3 201401                                                                                                                                                             06 (prerelease) (crosstool-NG linaro-1.13.1-4.8-2014.01 - Linaro GCC 2013.11) )                                                                                                                                                              #95 SMP Sun Jun 25 20:36:46 CEST 2017
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instructio                                                                                                                                                             n cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x3a800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 241664
[    0.000000] free_area_init_node: node 0, pgdat 80c6ee00, node_mem_map b9faa00                                                                                                                                                             0
[    0.000000]   Normal zone: 2124 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 241664 pages, LIFO batch:31
[    0.000000] percpu: Embedded 14 pages/cpu @b9f63000 s25600 r8192 d23552 u5734                                                                                                                                                             4
[    0.000000] pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pag                                                                                                                                                             es: 239540
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm27                                                                                                                                                             08_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_siz                                                                                                                                                             e=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUU                                                                                                                                                             ID=c1a66bc0-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet                                                                                                                                                              splash plymouth.ignore-serial-consoles
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 936296K/966656K available (7168K kernel code, 483K rwdata                                                                                                                                                             , 2012K rodata, 1024K init, 778K bss, 22168K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
    lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x80800000   (8160 kB)
      .init : 0x80b00000 - 0x80c00000   (1024 kB)
      .data : 0x80c00000 - 0x80c78fbc   ( 484 kB)
       .bss : 0x80c7a000 - 0x80d3ca64   ( 779 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[... 202 lines cut ...]
[... cut ...]
[   10.106402] Bluetooth: SCO socket layer initialized
[   10.122021] Bluetooth: HCI UART driver ver 2.3
[   10.122030] Bluetooth: HCI UART protocol H4 registered
[   10.122033] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   10.122176] Bluetooth: HCI UART protocol Broadcom registered
[   10.413761] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   10.413768] Bluetooth: BNEP filters: protocol multicast
[   10.413783] Bluetooth: BNEP socket layer initialized
[   59.046084] random: crng init done

```

paulenuta, 2018-05-10:
I will made a test from a fresh image of Jessie to see how it works.

```
root@raspberrypi:/home/pi# lsmod
Module                  Size  Used by
bnep                   11959  2
hci_uart               20480  1
btbcm                   7597  1 hci_uart
bluetooth             370485  22 hci_uart,bnep,btbcm
brcmfmac              208935  0
rtc_ds1307             13454  0
brcmutil                9159  1 brcmfmac
sc16is7xx              13373  1
hwmon                  10569  1 rtc_ds1307
cfg80211              542965  1 brcmfmac
snd_bcm2835            23042  1
rfkill                 21072  4 bluetooth,cfg80211
snd_pcm                97920  1 snd_bcm2835
snd_timer              23802  1 snd_pcm
tpm_tis_spi             4066  0
tpm_tis_core            7814  1 tpm_tis_spi
tpm                    32243  2 tpm_tis_spi,tpm_tis_core
spidev                  7314  0
snd                    69677  5 snd_timer,snd_bcm2835,snd_pcm
i2c_bcm2835             6331  0
bcm2835_gpiomem         3823  0
spi_bcm2835             7456  0
uio_pdrv_genirq         3788  0
uio                    10278  1 uio_pdrv_genirq
fixed                   3061  0
i2c_dev                 7159  0
fuse                  100998  3
ipv6                  409752  48
root@raspberrypi:/home/pi# sudo vcdbg log msg
001316.670: HDMI:EDID error reading EDID block 0 attempt 0
001317.933: HDMI:EDID error reading EDID block 0 attempt 1
001319.192: HDMI:EDID error reading EDID block 0 attempt 2
001320.450: HDMI:EDID error reading EDID block 0 attempt 3
001321.709: HDMI:EDID error reading EDID block 0 attempt 4
001322.967: HDMI:EDID error reading EDID block 0 attempt 5
001324.226: HDMI:EDID error reading EDID block 0 attempt 6
001325.484: HDMI:EDID error reading EDID block 0 attempt 7
001326.743: HDMI:EDID error reading EDID block 0 attempt 8
001328.001: HDMI:EDID error reading EDID block 0 attempt 9
001329.023: HDMI:EDID giving up on reading EDID block 0
001346.875: HDMI:Setting property pixel encoding to Default
001346.896: HDMI:Setting property pixel clock type to PAL
001346.915: HDMI:Setting property content type flag to No data
001346.934: HDMI:Setting property fuzzy format match to enabled
001590.836: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001590.862: hdmi: HDMI:>>>>>>>>>>>>>Rx sensed, reading EDID<<<<<<<<<<<<<
001591.169: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
001592.432: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
001593.695: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
001594.958: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
001596.221: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
001597.485: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
001598.749: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
001600.012: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
001601.276: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
001602.539: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
001603.566: hdmi: HDMI:EDID giving up on reading EDID block 0
001603.600: hdmi: HDMI: No lookup table for resolution group 0
001603.616: hdmi: HDMI: hotplug attached with DVI support
001603.645: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001603.939: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
001605.202: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
001606.464: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
001607.727: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
001608.989: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
001610.251: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
001611.514: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
001612.776: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
001614.038: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
001615.301: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
001616.326: hdmi: HDMI:EDID giving up on reading EDID block 0
001618.669: hdmi: HDMI: hotplug deassert
001618.684: hdmi: HDMI: HDMI is currently off
001618.698: hdmi: HDMI: changing mode to unplugged
001618.719: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001621.761: *** Restart logging
001626.801: Read command line from file 'cmdline.txt'
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=c1a66bc0-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
001963.293: Loading 'kernel7.img' to 0x8000 size 0x45d270
001966.947: No kernel trailer (run mkknlimg to fix) - assuming DT-capable
001973.294: Loading 'bcm2710-rpi-3-b.dtb' to 0x465270 size 0x44d8
002091.908: dtparam: i2c_arm=on
002101.167: dtparam: spi=on
002131.939: Loaded overlay 'letstrust-tpm'
002132.042: dtparam: audio=on
002188.021: Loaded overlay 'gpio-poweroff'
002188.036: dtparam: gpiopin=27
002188.709: dtparam: active_low=true
002244.818: Loaded overlay 'i2c-rtc'
002244.834: dtparam: mcp7941x=true
002290.746: Loaded overlay 'sc16is752-i2c-02'
002352.712: Loaded overlay 'sc16is752-i2c-01'
003513.045: Device tree loaded to 0x2effb200 (size 0x4d7d)
003518.612: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
004715.705: vchiq_core: vchiq_init_state: slot_zero = 0xfa980000, is_master = 1
004719.817: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
root@raspberrypi:/home/pi# ls -l /dev/ttyS*
crw--w---- 1 root tty       4, 64 May 10 10:36 /dev/ttyS0
crw-rw---- 1 root dialout 241,  0 May 10 10:43 /dev/ttySC0
crw-rw---- 1 root dialout 241,  1 May 10 10:36 /dev/ttySC1
crw-rw---- 1 root dialout 241,  2 May 10 10:36 /dev/ttySC2
crw-rw---- 1 root dialout 241,  3 May 10 10:36 /dev/ttySC3
root@raspberrypi:/home/pi#

```

paulenuta, 2018-05-18:
Update:
I have test Strech Raspbian Lite release from 18-04-2018 and 13-03-2018.
Error seem to manifest only on 18-04-2018.

**2018-03-13-raspbian-stretch-lite.img works fine**
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux
pi@raspberrypi:~ $ cat /etc/*release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs
```
on which I run:
```
# get Pi up to date
sudo apt-get clean -y; sudo apt-get update -y; sudo apt-get upgrade -y; sudo apt-get dist-upgrade -y; sudo apt-get autoremove -y
sudo apt-get clean -y; sudo apt-get install avrdude mc pigpio build-essential git python3-pip python-pip python-dev python-smbus sysbench -y
pip install python-periphery; pip install pyserial; pip install spidev
....
sudo sh -c "echo 'sc16is7xx' >> /etc/modules"
sudo dtc -@ -I dts -O dtb -o sc16is752-i2c-01.dtbo sc16is752-i2c-01.dts
sudo dtc -@ -I dts -O dtb -o sc16is752-i2c-02.dtbo sc16is752-i2c-02.dts
sudo cp sc16is752-i2c-01.dtbo /boot/overlays/
sudo cp sc16is752-i2c-02.dtbo /boot/overlays/
echo 'Add in config.txt dtoverlay section:'
sudo sh -c "echo '' >> /boot/config.txt"
sudo sh -c "echo '# Enable PI_OS LivePIN' >> /boot/config.txt"
sudo sh -c "echo 'dtoverlay=gpio-poweroff,gpiopin=27,active_low \n' >> /boot/config.txt"
sudo sh -c "echo '# Enable MCP74912 RTC' >> /boot/config.txt"
sudo sh -c "echo 'dtoverlay=i2c-rtc,mcp7941x \n' >> /boot/config.txt"
sudo sh -c "echo '# Enable I2C UARTEXP 2 modules' >> /boot/config.txt"
sudo sh -c "echo '# ttySC2, ttySC3' >> /boot/config.txt"
sudo sh -c "echo 'dtoverlay=sc16is752-i2c-02' >> /boot/config.txt"
sudo sh -c "echo '# ttySC0, ttySC1' >> /boot/config.txt"
sudo sh -c "echo 'dtoverlay=sc16is752-i2c-01 \n' >> /boot/config.txt"
sudo sh -c "echo '# Enable PI UART' >> /boot/config.txt"
sudo sh -c "echo 'enable_uart=1 \n' >> /boot/config.txt"

```

and test with:
```
#!/usr/bin/env python      
import sys
import time
from serial import *
from periphery import GPIO

if len(sys.argv) > 1:
	ports = sys.argv
	del ports[0]
	#print ports
else:
	ports = ['/dev/ttyUSB0', '/dev/ttyS0', '/dev/ttySC0', '/dev/ttySC1', '/dev/ttySC2', '/dev/ttySC3']
	#ports = ['/dev/ttyUSB0', '/dev/ttySC1']
	#ports = ['/dev/ttySC1']

counter=0
# How to get GPIO ??
# gpiochip496 -> ../../devices/platform/soc/3f804000.i2c/i2c-1/1004d/gpio/gpiochip496
# DSRA = GPIO4 = 500 | DSRB = GPIO0= 496
# gpiochip504 -> ../../devices/platform/soc/3f804000.i2c/i2c-1/10048/gpio/gpiochip504
# GPIO0-8 > 504, 505, 506, 507, 508
# DSRA = GPIO4 = 508 | DSRB = GPIO0= 504

esc_buzz = b'\x10' + b'\x14' + b'\x03' + b'\x00' + b'\x00' + b'\x01' + b'\x00'
esc_drawer = b'\x1B' + b'\x70' + b'\x00' + b'\x0A' + b'\xFF'
esc_cutter = b'\x1D' + b'\x56' + b'\x42'
to_send = b'\x0A'

OpenPorts = []
for port in ports:
	try:
		#print port
		ser = Serial()
		ser.port = port
		ser.baudrate = 9600
		ser.xonxoff = True
		ser.rtscts = False
		ser.timeout = 1
		print ser.name, ' was setup'
		ser.close()
		OpenPorts.append(ser)
		#print OpenPorts
	except SerialException:
		print 'serial port unavailable', port
		continue

if ( ser.name == '/dev/ttySC0' ):
	DSR = GPIO(508, "in")
	
if ( ser.name == '/dev/ttySC1' ):
	DSR = GPIO(504, "in")

if ( ser.name == '/dev/ttySC2' ):
	DSR = GPIO(500, "in")
	
if ( ser.name == '/dev/ttySC3' ):
	DSR = GPIO(496, "in")

while 1:
	for ser in OpenPorts:
		ser.open()
		print '\n Start with: ', ser.name
		if ser.isOpen():
			print '<'
			print ser.name, ' opened'
			if ( DSR.read() == 0 ):
				print '@Write line: ', counter
				ser.write('@Write line: %d \n'%(counter))
				ser.write('\n')
				ser.write('\n')
				ser.write(to_send)
				ser.write(esc_drawer)
				ser.write(esc_cutter)
				counter += 1			
				print ' data was sent to printer', ser.name
			else:
				print ' CTS_{:s} is NotSet - Printer Error @ '.format(ser.name[-1:]), ser.name
		ser.close()
		print ser.name, ' closed'
		print '> \n'
		time.sleep(2)

```

**2018-04-18-raspbian-stretch-lite.img throws "Unexpected interrupt" on every serial or gpio call to the sc16is7xx**
```
[  627.437693] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: 8
[  632.199867] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  634.227692] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  636.248385] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  638.269045] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  640.290448] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  642.311459] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a

```

paulenuta, 2018-05-18:
Any advice?

pelwell, 2018-05-18:
I requested a Pull Request from @hardikharpal in #2542, but I've not received a response.

Since this issue is affecting quite a few people I started to prepare a patch but didn't like the approach taken elsewhere. The issue is caused by the fact that the SC16IS752 has an Enhanced Feature Register which is aliased at the same address as the Interrupt Identification Register; accessing it requires that a magic value is written to the Line Configuration Register. If an interrupt is raised while the EFR is mapped in then the interrupt handler won't be able to access the IIR, leading to the error messages you've seen.

The workaround discussed here is to add a delay between enabling interrupts from the device and returning from the startup method, the idea being that the set_baud method which accesses the EFR is called shortly afterwards. However, this doesn't protect against other calls to set_baud. My preferred solution is to explicitly mask the interrupt from the device for the critical period while the EFR register is swapped in, but I don't have the means to test the change.

Is anyone running the -v7 kernel (Pi2, Pi3, Pi3+, CM3) willing to try a special build of the module including the interrupt masking? You can download the module from [here](https://drive.google.com/open?id=0B_P-i4u-SLBXYTJUbnBmYkFaUHM).

N.B. You will first need to `sudo rpi-update` to get the latest kernel version, then with the new module in the current working directory I suggest you run:
```
$ sudo mv /lib/modules/`uname -r`/kernel/drivers/tty/serial/sc16is7xx.ko{,.orig}
$ sudo cp sc16is7xx.ko /lib/modules/`uname -r`/kernel/drivers/tty/serial/
```
Then reboot and see if the interface works and doesn't report errors.

After the test you can revert to the original version with:
```
$ sudo mv /lib/modules/`uname -r`/kernel/drivers/tty/serial/sc16is7xx.ko{.orig,}
```
I look forward to some feedback.

paulenuta, 2018-05-18:
> The workaround discussed here is to add a delay between enabling interrupts from the device and returning from the startup method, the idea being that the set_baud method which accesses the EFR is called shortly afterwards. 

I have tried [mdelay(1)](https://raw.githubusercontent.com/janztec/empc-arpi-linux/rpi-3.18.y/drivers/tty/serial/sc16is7xx.c) as showed above on a kernel re-build on which I also activate TPM driver as documented [here](https://letstrust.de/archives/9-Howto-Enable-TPM-Support-on-a-Raspberry-PI-0,-0W,-1,-2,-3-and-make-it-work-with-the-LetsTrust-TPM.html), but it just reduce the number of error messages.

> Is anyone running the -v7 kernel (Pi2, Pi3, Pi3+, CM3) willing to try a special build of the module including the interrupt masking? 

Is this OK for you?
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux
pi@raspberrypi:~ $ cat /etc/*release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs
```
I also have a card with 2018-04-18-raspbian-stretch-lite.img. I can test on Pi3, Pi3+.

pelwell, 2018-05-18:
Modules have to match the running kernel fairly precisely, so your current, old kernel won't work. Fortunately you can revert to your current kernel+firmware using `sudo rpi-update 936a8dc`, so it is safe to update to the 4.14 kernel using `sudo rpi-update` in order to apply the patched module, as described above.

paulenuta, 2018-05-18:
Now on;
```
root@raspberrypi:/home/pi# uname -a
Linux raspberrypi 4.14.39-v7+ #1112 SMP Sat May 5 12:01:33 BST 2018 armv7l GNU/Linux
```
with the module donloaded from above [link](https://drive.google.com/open?id=0B_P-i4u-SLBXYTJUbnBmYkFaUHM) still there are errors.
```
[   12.601004] Bluetooth: HCI UART driver ver 2.3
[   12.601019] Bluetooth: HCI UART protocol H4 registered
[   12.601025] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   12.602247] Bluetooth: HCI UART protocol Broadcom registered
[   12.878142] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   12.878150] Bluetooth: BNEP filters: protocol multicast
[   12.878167] Bluetooth: BNEP socket layer initialized
[  167.593284] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: 8
[  177.647082] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  182.669447] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  187.693995] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  196.718369] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  201.742118] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  206.761328] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  211.783646] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  216.805630] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  241.909223] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  246.930364] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  256.969775] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a

```

pelwell, 2018-05-18:
Would you say this is an improvement over the original, unpatched driver? It's possible I've missed another place that needs protection.

paulenuta, 2018-05-18:
I have run

> After the test you can revert to the original version with:
> 
> $ sudo mv /lib/modules/`uname -r`/kernel/drivers/tty/serial/sc16is7xx.ko{.orig,}

then reboot
and this is the output (script is waiting one second + ser.out_waiting between writes):
```
[   12.341142] Bluetooth: HCI UART driver ver 2.3
[   12.341156] Bluetooth: HCI UART protocol H4 registered
[   12.341162] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   12.341359] Bluetooth: HCI UART protocol Broadcom registered
[   12.518602] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   12.518609] Bluetooth: BNEP filters: protocol multicast
[   12.518622] Bluetooth: BNEP socket layer initialized
[   76.652563] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[   88.693679] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[   88.695194] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[   94.717621] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[   94.719134] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  106.762782] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  106.764304] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  112.786679] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  112.788184] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  112.789047] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  124.831409] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  124.832972] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  130.854054] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  130.855567] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a

```
I got 14 "Unexpected..."

And this is the output after applying you patched module and restart (no other changes in system or script)
```
[   25.940200] Bluetooth: HCI UART driver ver 2.3
[   25.940218] Bluetooth: HCI UART protocol H4 registered
[   25.940224] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   25.940449] Bluetooth: HCI UART protocol Broadcom registered
[   26.100663] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   26.100670] Bluetooth: BNEP filters: protocol multicast
[   26.100686] Bluetooth: BNEP socket layer initialized
[  131.862362] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  137.885442] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  143.911167] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  149.931522] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  149.933485] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  155.956271] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  161.976532] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  168.000119] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  174.023469] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  180.046545] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  192.090521] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a

```
I got 11 "Unexpected..."
Script has a counter (initialized at 0) and port is opened and data sent until counter is <=11.
So 14(original) to 11(patched).

pelwell, 2018-05-18:
The set_termios is another user of the EFR register and needed the same treatment. Try [this module](https://drive.google.com/open?id=0B_P-i4u-SLBXYTJUbnBmYkFaUHM) (yes, the URL is unchanged - it's using versioning in Google Drive)

paulenuta, 2018-05-18:
I replace the module and this are the results:

- after first reboot  got 10 errors
- after second reboot got 8 errors
- after third reboot got 10 errors
- after forth reboot got 8 errors
- and a second run without reboot give 12 errors
- the third run without reboot give 11 errors.

( I have not use copy/paste in above lines :) )
```
[   12.438135] Bluetooth: BNEP socket layer initialized
[   95.605189] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  101.629754] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  113.670660] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  125.714546] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  131.737532] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  143.781530] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  149.804654] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  155.827732] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a

[  760.060005] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  766.084792] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  772.109343] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  778.133892] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  784.158429] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  790.183436] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  796.208341] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  802.232890] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  808.257814] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  814.282367] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  820.303415] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  826.330438] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a

[  886.036865] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  892.061106] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  898.086195] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  904.110817] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  910.136787] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  916.156922] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  928.204194] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  934.228931] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  940.253443] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  946.278117] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a
[  952.302655] sc16is7xx 1-0048: ttySC1: Unexpected interrupt: a

```
In the script I'm also [using chip GPIO as a CTS pin because I couldn't read CTS state](https://www.raspberrypi.org/forums/viewtopic.php?f=44&p=1067687&t=165137&sid=d38de7b3fa70918db066ccd7d50037d5)

```
#!/usr/bin/env python      
import sys
import time
import serial
from serial import *
from periphery import GPIO

if len(sys.argv) > 1:
	ports = sys.argv
	del ports[0]
	#print ports
else:
	ports = ['/dev/ttyUSB0', '/dev/ttyS0', '/dev/ttySC0', '/dev/ttySC1', '/dev/ttySC2', '/dev/ttySC3']
	#ports = ['/dev/ttyUSB0']
	#ports = ['/dev/ttySC0']
	#ports = ['/dev/ttySC1']

counter=0
# How to get GPIO ??
# gpiochip496 -> ../../devices/platform/soc/3f804000.i2c/i2c-1/1004d/gpio/gpiochip496
[... 67 lines cut ...]
					ser.write(line1)
				else:
					ser.write(esc_OVR)
					ser.write('Write line: %d'%(counter))
				time.sleep(1)
				# ser.write('\n')
				# ser.write(to_send)
				# ser.write(esc_drawer)
				# ser.write(esc_cutter)
				counter += 1			
				# print ' data was sent to printer', ser.name
				# print 'Data on buffer '
				while ser.out_waiting:
					ser_Buff.append(ser.out_waiting)
				# print ''.join('{:02X} |\t'.format(a) for a in (ser_Buff))
			ser.close()
			ser_Buff = []
			# print ser.name, ' closed'
			# print '> \n'
	else:
		# print ' CTS_{:s} is NotSet - Printer Error @ '.format(ser.name[-1:]), ser.name
		counter = 0
	time.sleep(5)

```

pelwell, 2018-05-18:
I don't suppose you have some spare hardware I could borrow for a while, do you?

paulenuta, 2018-05-18:
I have available and I can talk and get approval for sending, how do we move on this?

pelwell, 2018-05-18:
Email me at phil@raspberrypi.org and we can arrange shipping (and any guarantees you need).

paulenuta, 2018-06-28:
In my use case and with bellow test script I have not received "Unexpected interrupt", so for me the patch is correct.

```
#!/usr/bin/env python
import sys
import time
import threading
from serial import *

if len(sys.argv) > 1:
	ports = sys.argv
	del ports[0]
	#print ports
else:
	ports = ['/dev/ttySC0', '/dev/ttySC1', '/dev/ttySC2', '/dev/ttySC3']

counter = 1
maxCtr = 10 + counter
baud = 9600
out = "[ABCDEFGHIJKLMNOPQRSTUVXYZabcdefghijklmnopqrstuvxyz0123456789]\n"
inp = ""

for i in range(0, 3, 2):
	ser0 = Serial(ports[i], baud, timeout=1, rtscts=0)
	ser1 = Serial(ports[i+1], baud, timeout=1, rtscts=0)
	ser0.flush()
	ser1.flush()
	while counter < maxCtr:
		print '-------- Start --------'
		print '|{:^21}|'.format(counter)
		print '#{:03d} > {} {}.'.format(counter, out, ser0.name)
		print 'Snd {} bytes.'.format(ser0.write( out ))
		time.sleep(1)
		inp = ser1.readline()
		print '#{:03d} < {:s} {}.'.format(counter, inp, ser1.name)
		print 'Rcv {} bytes.'.format(len(inp))
		if (inp == out):
			print '|{:^21}|'.format('Received OK.')
		else:
			print '|{:^21}|'.format('Received ERROR!')
			exit()
		print '-------- Stop  --------'
		print '\r\n'
		counter += 1
		time.sleep(2)

	ser0.flush()
	ser1.flush()
	ser0.close()
	ser1.close()
	counter = 1

	def serialDataSnd():
		global counter
		ser0 = Serial(ports[i], baud, timeout=1, rtscts=0)
		while counter < maxCtr:
			print "#{:03d} Sending  > {:s}".format( counter, out )
			ser0.write(out)
			time.sleep(2)

	def serialDataRcv():
		global counter
		ser1 = Serial(ports[i+1], baud, timeout=1, rtscts=0)
		while counter < maxCtr:
			time.sleep(1)
			line = ser1.readline()
			sys.stdout.write('#{:03d} Received < {:s}\r\n'.format(counter, line))
			counter += 1

	def SerialTestMain():
		thread1 = threading.Thread(target = serialDataSnd)
		thread2 = threading.Thread(target = serialDataRcv)
		thread1.start()
		thread2.start()
		thread1.join()
		time.sleep(2)

	SerialTestMain()
	ser0.close()
	ser1.close()
	counter = 1

exit()
```

JamesH65, 2018-06-28:
@paulenuta So are you happy for this issue to be closed?

paulenuta, 2018-06-28:
@JamesH65 Yes.