raspberrypi/linux #2530: rpi 3b+: cat /sys/class/thermal/thermal_zone0/temp: invalid argument

m1027, 2018-04-25:
On my rapberry 3b+ I cannot read the cpu temperature:
```
cat /sys/class/thermal/thermal_zone0/temp: invalid argument
```
Kernel is 4.14.34 from https://github.com/raspberrypi/linux.git with
```
CONFIG_THERMAL=y                                                               
CONFIG_THERMAL_HWMON=y
CONFIG_THERMAL_OF=y
CONFIG_THERMAL_DEFAULT_GOV_STEP_WISE=y                                         
CONFIG_THERMAL_GOV_STEP_WISE=y                                                                              
CONFIG_BCM2835_THERMAL=y
```

Firmware is on
commit ae9a493932e47e08cabb25a2728037298075fd00
Author: popcornmix <popcornmix@gmail.com>
Date:   Tue Apr 10 18:34:46 2018 +0100

    kernel: LAN78xx VLAN checksum fix
from https://github.com/raspberrypi/firmware.git.

FYI: I have some rpi 3 (non-plus version) around with similar setup where reading the temperature works.

JamesH65, 2018-05-10:
@m1027 Any luck with the suggestions above?

m1027, 2018-05-10:
> @m1027 Any luck with the suggestions above?

Thanks for asking.

I haven't tried yet since starting with a default kernel might fix it but
later stripping it down to what is actually required here is a huge task.

I thought I'd get some clues here to circumvent that way.

Feel free to close this issue or put it in a wait state. I'll come back on at
some point in the future.

m1027, 2018-05-13:
Without using the default kernel, after a simple update of firmware and kernel
(4.14.34 to 4.14.39) the temperature value 'temp' can be read out properly.

Thanks

closed 2018-05-13 by m1027