raspberrypi/linux #2532: old firmware soc issue with kernel 4.14.34

procount, 2018-04-26:
I have a kernel / firmware compatibility problem with PINN (fork of NOOBS) that results in the following error messages:

```
Jan  1 00:00:03 recovery kern.info kernel: [    2.042377] raspberrypi-firmware soc:firmware: Attached to firmware from 2017-04-05 11:51
Jan  1 00:00:03 recovery kern.err kernel: [    2.052410] raspberrypi-firmware soc:firmware: Request 0x00030046 returned status 0x80000001
Jan  1 00:00:04 recovery kern.err kernel: [    4.074719] raspberrypi-firmware soc:firmware: Request 0x00030046 returned status 0x80000001
Jan  1 00:00:06 recovery kern.err kernel: [    6.154717] raspberrypi-firmware soc:firmware: Request 0x00030046 returned status 0x80000001
Jan  1 00:00:08 recovery kern.err kernel: [    8.234750] raspberrypi-firmware soc:firmware: Request 0x00030046 returned status 0x80000001
Jan  1 00:00:10 recovery kern.err kernel: [   10.324808] raspberrypi-firmware soc:firmware: Request 0x00030046 returned status 0x80000001
Jan  1 00:00:12 recovery kern.err kernel: [   12.394737] raspberrypi-firmware soc:firmware: Request 0x00030046 returned status 0x80000001
Jan  1 00:00:14 recovery kern.err kernel: [   14.474718] raspberrypi-firmware soc:firmware: Request 0x00030046 returned status 0x80000001
Jan  1 00:00:16 recovery kern.err kernel: [   16.554704] raspberrypi-firmware soc:firmware: Request 0x00030046 returned status 0x80000001
Jan  1 00:00:18 recovery kern.err kernel: [   18.634705] raspberrypi-firmware soc:firmware: Request 0x00030046 returned status 0x80000001
```

Before the 3B+ came out, we used kernel 4.4.50 (e223d71ef728c559aa865d0c5a4cedbdf8789cfd) with rpi-firmware (22de0bb68d34fd210ba9d086c6a1fc5e90f0bfbb) and everything was good.

When the 3B+ arrived, we used kernel 4.9.50 (f263ce844727c06fcd06242745e72ca385ccf90d) with rpi-firmware (3347884c7df574bbabeff6dca63caf686e629699). This worked fine for any OS that had been upgraded to work with the 3B+. But existing OSes that used old firmware would not boot from PINN (or NOOBS). I summised an incompatibility was introduced between bootcode.bin of PINN and start.elf of the older OSes. So I added the option to downgrade PINN's firmware (on models prior to the 3B+) to the 22de0bb version so that the older OSes would continue to boot.

NOOBS has now moved onto kernel 4.14.34, so I followed suit, upgrading the kernel to (f70eae405b5d75f7c41ea300b9f790656f99a203) and rpi-firmware to (5db8e4e1c63178e200d6fbea23ed4a9bf4656658). I still allowed pinn's firmware to be downgraded to 22de0bb, but with the 4.14.34 kernel, this now produces the above error messages.

Has something fundamentally changed in kernel 4.14.34 from 4.9.50 which prevents it working with the older 22de0bb firmware?

When I downgrade the firmware, I only change bootcode.bin and recovery.elf. Do I also need to change something else, like the dtb files?

firmware: Request 0x00030046 is **get_throttled** and the error seems related to linux/drivers/firmware/raspberrypi.c

6by9, 2018-04-26:
raspberrypi-firmware is complaining when it shouldn't (just return the error, possibly with a trace level debug message).
TBH get_throttled_poll in the firmware driver should check the error code, log an error, and not reschedule itself if it gets the unsupported error from the mailbox.

procount, 2018-04-26:
Thanks! I look forward to the fix.
It's not critical for me and it's good to know it's not a major compatibility problem, just that the messages get in the way so it's annoying. At least I will be able to keep with the 4.14 branch.

JamesH65, 2018-04-26:
Been thinking about this. Since its the upper layer that makes the call's responsibility to check for errors, I do wonder whether this should actually remain at error level. Clearly it is an error, because something is trying to call something that isn't there. So should we continue to log it as an error and just fix the caller so that it doesn't continue to make the call after the first error? Or drop the logging level down from error to debug or similar?

6by9, 2018-04-26:
So should an I2C adapter shout in the kernel log on every error it sees? That would include every probe for a device that is potentially on the bus. Or for every address not present during an i2cdetect? Lots of error messages!
Definite "no" from me. Both should be considered as a transparent pipes that report errors and let the caller log a message should it fail.

JamesH65, 2018-04-26:
OK, will drop the error log to dbg.

pelwell, 2018-04-26:
I would vote for logging it once as a warning.

JamesH65, 2018-04-27:
RFC

https://github.com/raspberrypi/linux/pull/2534

closed 2018-04-28 by pelwell