raspberrypi/linux #2533: Mouse event from touchscreen stops

lbenini, 2018-04-26:
We have found a particularly strange problem with RPI3 and a touchscreen.
On a "macro" level the touchscreen stops to work "after a while"
We have done quite a lot of tests that hopefully can help:

1. The problem exists on Jessie and Stretch (both updated)
1. On Jessie the problem exists **with and without X**

 (Qt application in eglfs mode has the same behavior); on Stretch we have not tested without Xorg
1. The majority of the tests we have done are on an iiyama PROLITE T2252MSC-B1, but is not isolated to Quanta Controller
1. We have used already the monitor and the touch with success with other Linux-based boards without any problem
1. Normally the kernel send messages on both /dev/input/mouse? and /dev/input/event? but when the problem appears the /dev/input/mouse? stops receiving messages. 
   Touch messages continue to be received on /dev/input/event?
1. With an additional USB mouse connected is it possible to send mouse events also when is stopped
1. Powering off and on the touchscreen doesn't resolve the problem 
1. We have been able to isolate and reproduce the problem moving a finger in a circular fashion on the screen (see video) for a while
1. The problem "disappears" if we touch with two fingers (it will anyway represent himself after a while)

We have prepared a video that shows the situation.
I cannot upload here so you can find it here: https://www.youtube.com/watch?v=PhPwJ6Vm40k
The terminal on the left shows the touch events, the one in the middle show the mouse events and one the right the mouse events coming only from the USB mouse

```
root@raspberrypi:~# tvservice -n
device_name=IVM-PL2252MT
```

```
root@raspberrypi:~# lsusb
Bus 001 Device 004: ID 0408:3008 Quanta Computer, Inc.
Bus 001 Device 006: ID 413c:2107 Dell Computer Corp.
Bus 001 Device 005: ID 045e:0040 Microsoft Corp. Wheel Mouse Optical
Bus 001 Device 007: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
root@raspberrypi:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.34-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1110 SMP Mon Apr 16 15:18:51 BST 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] On node 0 totalpages: 242688
[    0.000000] free_area_init_node: node 0, pgdat 80c84e40, node_mem_map ba3a1000
[    0.000000]   Normal zone: 2133 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:31
[    0.000000] random: fast init done
[    0.000000] percpu: Embedded 17 pages/cpu @ba34a000 s38720 r8192 d22720 u69632
[    0.000000] pcpu-alloc: s38720 r8192 d22720 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=0e663c18-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait plymouth.ignore-serial-consoles
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 940236K/970752K available (7168K kernel code, 575K rwdata, 2072K rodata, 1024K init, 706K bss, 22324K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
                   lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x80800000   (8160 kB)
                     .init : 0x80b00000 - 0x80c00000   (1024 kB)
[... 468 lines cut ...]
B: MSC=10

I: Bus=0003 Vendor=413c Product=2107 Version=0110
N: Name="DELL Dell USB Entry Keyboard"
P: Phys=usb-3f980000.usb-1.1.3/input0
S: Sysfs=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/0003:413C:2107.0003/input/input1
U: Uniq=
H: Handlers=sysrq kbd leds event1
B: PROP=0
B: EV=120013
B: KEY=10000 7 ff9f207a c14057ff febeffdf ffefffff ffffffff fffffffe
B: MSC=10
B: LED=7

I: Bus=0003 Vendor=0408 Product=3008 Version=0110
N: Name="Quanta OpticalTouchScreen"
P: Phys=usb-3f980000.usb-1.2/input0
S: Sysfs=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:0408:3008.0001/input/input3
U: Uniq=0000
H: Handlers=mouse1 event2
B: PROP=2
B: EV=b
B: KEY=400 0 0 0 0 0 0 0 0 0 0
B: ABS=2608000 3
```

E3V3A, 2018-04-28:
Is this a problem that suddenly appeared (e.g. after an update/upgrade) or something that has been around since first install? If it appeared after update, please provide more info about your OS version and try to pinpoint what was updated when it started.

thooge, 2018-05-09:
Same problem here with Iiyama ProLite T2252MSC.
Tried with fresh install of Raspbian, Raspi 3B and Raspi3 B+.
```
$uname -a
Linux raspberrypi 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux

$lsusb
Bus 001 Device 006: ID 046d:c05a Logitech, Inc. M90/M100 Optical Mouse
Bus 001 Device 004: ID 2575:0401  
Bus 001 Device 005: ID 046a:0011 Cherry GmbH G83 (RS 6000) Keyboard
Bus 001 Device 007: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
$cat /proc/bus/input/devices
I: Bus=0003 Vendor=2575 Product=0401 Version=0110
N: Name="Weida Hi-Tech                CoolTouchR System           "
P: Phys=usb-3f980000.usb-1.2/input0
S: Sysfs=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:2575:0401.0001/input/input3
U: Uniq=
H: Handlers=mouse1 event2 
B: PROP=2
B: EV=b
B: KEY=400 0 0 0 0 0 0 0 0 0 0
B: ABS=2608000 3
```
Tried on screen keyboard _onboard_. Unusable, seems not to send KeyUp events?

Same configuration with normal PC and Debian Stretch works as a charme.

JamesH65, 2018-05-10:
Reference : https://www.raspberrypi.org/forums/viewtopic.php?t=179663

lbenini, 2018-05-15:
I come back late, but I've been able to conduct more test only today:

1. The software is a raspbian stretch updated to today (it behaves exactly as described before)
1. Using a powered USB HUB doesn't resolve the problem
1. After the hang is it possible to recover the touchscreen by clicking with two fingers (as in the video) but also by removing and inserting again the hid-multitouch module
1. The following boot options have NO effect on the problem:
             * hid.mousepool (various settings)
             * dwc_otg.speed (1/0)
1. Same behaviour on RPI3 B+

trondjl, 2018-08-01:
Hi,
I have the same issue. I can trigger it by clicking with three fingers and fix it with two. It also seems to hang by itself sometimes when left alone, but also it might be the first click then is misinterpreted...
Using evdev driver instead of libinput seems to be more stable in X, but still hangs with three fingers in my javafx app.
Using rpi3b with updated Jessie and an 10 inch waveshare screen.
Anyone found a good workaround or fix yet?

JamesH65, 2019-09-19:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

thooge, 2019-09-19:
Still waiting for a fix...

thurban, 2019-09-19:
I was able to fix it by adding the following text to the end of the **/boot/cmdline.txt**:
`usbhid.mousepoll=0`
The PROLITE T2252MSC and PROLITE T2454MSC both work just fine afterwards.

JamesH65, 2019-12-20:
@thooge Did the above workaround fix your issue?

closed 2020-02-13 by JamesH65