raspberrypi/linux #2539: pifacecommon not up-to-date

spiro-trikaliotis, 2018-05-01:
The package python3-pifacecommon reports itself as version 4.2.1. However, it differs in one patch from the version from here: https://github.com/piface/pifacecommon.

It misses the patch d4c568926e, cf. https://github.com/piface/pifacecommon/commit/d4c568926e49dc75a983e2fed41b79ebf317b10d?diff=unified. That one is needed in order to run piface (for example the digital IO) on newer kernels:

Can you include that one in a Raspbian package so the piface can be run without a local patch?

Regards
Spiro Trikaliotis

spiro-trikaliotis, 2018-05-01:
I am sorry, I am in the wrong bug tracker. Closing here and searching for the right one.

closed 2018-05-01 by spiro-trikaliotis

XECDesign, 2018-05-01:
The package was provided by piface. The correct place to report it is https://github.com/piface/pifacecommon. The last reply I saw from them was that they no longer support the debian package and tell people to use pip instead. I also think that they're having somebody look into providing .debs again.

spiro-trikaliotis, 2018-05-01:
Thank you for your reply.
I just tried with pip (more precisely, pip3). It still gives the "wrong" spi.py file, without the patch. Thus, this is no solution, either.