raspberrypi/linux #2540: hfsplus: "getfattr -d ..." returns error "No such attribute"

llemtt, 2018-05-02:
Hi,

issuing the command "getfattr -d " on a file/dir without extended attributes and without FinderInfo returns the error "No such attribute".

This is traceable directly to hfsplus listxattr function that returns ENODATA instead of returning a more correct empty attribute list.

Such a behavior causes lots of problems including:

sharing an hfsplus volume via samba using the vfs_fruit module won't let you rename a file/dir without extended attributes or FinderInfo.

JamesH65, 2018-05-02:
Is this specific to Raspberry Pi or a more upstream issue? AFAIK, we make no changes to this area, so I suspect this need to be reported upstream.

llemtt, 2018-05-02:
I traced it on Raspberry Pi only but I think it's a more general issue!

How can I report upstream?

JamesH65, 2018-05-02:
You need to find check the maintainers file for your particular driver (root of the Linux tree). I've had a quick look, and this is what it says for HFS, so you need to email list below. You may want to check the list archive first to see if is already reported.

```
HFS FILESYSTEM
L:      linux-fsdevel@vger.kernel.org
S:      Orphan
F:      Documentation/filesystems/hfs.txt
F:      fs/hfs/

HFSPLUS FILESYSTEM
L:      linux-fsdevel@vger.kernel.org
S:      Orphan
F:      Documentation/filesystems/hfsplus.txt
F:      fs/hfsplus/
```

llemtt, 2018-05-02:
Many thanks!

Anyway it will be a long way for APFS to completely replace HFS+ given that it doesn't support a currently supported OSX version like El Capitan or even TimeMachine backups...

closed 2018-05-02 by llemtt