raspberrypi/linux #2541: 7" touchscreen does not stretch the picture to fullscreen anymore

bouwew, 2018-05-02:
I'm using the official 7" 800x480 LCD-screen in an application.
Before, I could use the framebuffer-parameters in /boot/config.txt to show a low-res-picture (chrome-browser in kiosk-mode) in a higher resolution and the screen would be stretched full-screen. As described here: https://www.raspberrypi.org/forums/viewtopic.php?t=120710&p=817889#p1196487
I am using higher numbers and the picture would still be stretched to full-screen.

Recently, I've update my Pi3 and the effect is gone, I have now unsymmetrical large black borders around the picture.
Is this something related to chromium-browser? Or is this happening due to a firmware update? If this is because of a firmware-update, can this be reverted, please? It makes the official 7" screen completely unusable.

pelwell, 2018-05-02:
Please post your config.txt (or at least the parts relevant to the display).

bouwew, 2018-05-03:
The relevant lines from my config.txt:

```
# disable_overscan=1
framebuffer_width=1154
framebuffer_height=640
framebuffer_aspect=-1

```

bouwew, 2018-05-03:
Update: after I just updated my older image, the kiosk-screen is still being shown fullscreen :)
After checking my config, I noticed I am still using Midori as the browser in kiosk-mode.
So, this likely means that the problem is chromium-browser-related. 
Previously, I had installed chromium-browser to my old image, to test the updated version. An older version did not work well so I switched to Midori. 

I will install chromium-browser once more to see the effect. More info to follow.

bouwew, 2018-05-03:
Update 2: I have installed chromium-browser and am using it in kiosk-mode.
The display is still fullscreen :)

I will do some more experimenting to see if I can reproduce the problem, but for now there is no problem. Therefore ,I close the issue.

closed 2018-05-03 by bouwew