raspberrypi/linux #2542: sc16is7xx.c HW flow control not working

hardikharpal, 2018-05-03:
I able to communicate with SC16IS752(custom board). Without HW flow control everything works fine.
When I enable HW flow control. When I send  data on /dev/ttySC0 it did not get transmitted through TXA pin of  SC16IS752(checked with DSO no movment on signal). 

So I checked voltage level CTS pin. It is low so there  is no problem there. 

I searched for this issue and landed to [https://www.spinics.net/lists/linux-serial/msg21881.html](https://www.spinics.net/lists/linux-serial/msg21881.html) and  using this guide [KERNEL BUILDING](https://www.raspberrypi.org/documentation/linux/kernel/building.md)  I edited sc16is7xx.c and able to communicate with HW Flow Control Enabled.

Here are the changes I done 
 in `sc16is7xx_set_baud()`  replace ` sc16is7xx_port_write(port,SC16IS7XX_EFR_REG,SC16IS7XX_EFR_ENABLE_BIT);` with  `sc16is7xx_port_update(port,SC16IS7XX_EFR_REG,SC16IS7XX_EFR_ENABLE_BIT,SC16IS7XX_EFR_ENABLE_BIT);`
as this clears Auto CTS Auto RTS bits of EFR register. 

and 
another change in  `sc16is7xx_set_termios()` replace `if (termios->c_cflag & CRTSCTS)
flow |= SC16IS7XX_EFR_AUTOCTS_BIT |	SC16IS7XX_EFR_AUTORTS_BIT;`
with `if (termios->c_cflag & CRTSCTS){
flow |= SC16IS7XX_EFR_AUTOCTS_BIT | SC16IS7XX_EFR_AUTORTS_BIT;
port->status |= UPSTAT_AUTOCTS;	
}`
 
as  In sc16is6xx_set_termios() upper layer is informed that this driver support automatic HW flow control. Without this knowledge the two layers eventually ended up in different states, causing communication to stop.

Its working fine.

hardikharpal, 2018-05-03:
For quick test short(link) RTC CTS pin of SC16IS7XX breakout board to each other and also short TX RX to each other. Test with minicom, hw flow control enable.

hardikharpal, 2018-05-04:
It worked after one more change in `sc16is7xx_get_mctrl()` function replace `return TIOCM_DSR | TIOCM_CAR;` by `return TIOCM_DSR | TIOCM_CAR | TIOCM_RI | TIOCM_CTS;`.

Now if TX RX shorted,
if RTS CTS shorted it send and receive data properly.
if RTS CTS not shorted it does not transmit data.

hardikharpal, 2018-05-08:
The change I made, Will it break other functionality of driver? Any comment on this?

JamesH65, 2018-05-08:
If the change is ONLY in the sc16is7xx driver, then I would not expect it to affect any other drivers. However, this is not a driver we maintain, you should really be finding out who maintains it (check the maintainer file in root of Linux tree) and communicate with them - usually via a kernel mailing list.

pelwell, 2018-05-08:
James is correct - this is an upstream driver, and patches should go upstream - but I already have one downstream patch in our kernel trees and I'm prepared to take another. I've been trying to resurrect my test setup so I can try this patch and #2529, but I think the hardware is broken in some way, so I'm prepared to take it on trust that neither patch is harmful and apply them speculatively.

pelwell, 2018-05-08:
To save errors and to ensure the patch has the correct attribution, please submit a Pull Request for this and #2529, either separately or together.

paulenuta, 2018-05-10:
Hi,
Please review my comments from [#2529](https://github.com/raspberrypi/linux/issues/2529).

Thank you!
Paul

JamesH65, 2018-06-26:
@pelwell @paulenuta @hardikharpal Little bit confused. Can this issue also be closed if the referenced issue #2529 has been closed?

pelwell, 2018-06-26:
I was hoping for some feedback - I notified @paulenuta of the likely fix on Jun 8th and, aside from a brief acknowledgement, I've not heard anything since.

paulenuta, 2018-06-26:
Sorry for the delay, by the end of the week I'll finish my tests and come back with feedback.
Paul

paulenuta, 2018-06-28:
@JamesH65 , @pelwell  I have return feedback in #2529 https://github.com/raspberrypi/linux/issues/2529#issuecomment-400966175.
Any way I don't think my tests are relevant with #2542, aside the fact that I couldn't test with "rtscts=1" (hw flow control). Maybe @hardikharpal will give us some feedback?

Paul

pelwell, 2018-06-28:
The streams have got somewhat crossed. #2529 can now be closed. For this issue I asked for the proposed patches to be submitted as a Pull Request, but so far that hasn't happened.

JamesH65, 2019-07-31:
@hardikharpal Did you ever
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested. have time to do that pull request?

beikeland, 2020-05-13:
Issue still still present in kernel v5.4.40-v7l  (ee6afe0c70e93e6c9da398f045c42cc1ea2a2f81) which has the efr_lock mutex from #2529. Applying the changes @hardikharpal have suggested seems to resolve the hardware flow control issue for me as well.

pelwell, 2020-05-13:
Would you mind looking at newer kernel versions to see if the issue has been fixed their? git log <branch> <file> is a quick way to see if any changes have been made.

pelwell, 2020-05-13:
I've looked ahead and can't see any relevant changes. Combining this post and the original kernel patch gets four changes:
```
diff --git a/drivers/tty/serial/sc16is7xx.c b/drivers/tty/serial/sc16is7xx.c
index fe486498530d..7eda5b1eb1d9 100644
--- a/drivers/tty/serial/sc16is7xx.c
+++ b/drivers/tty/serial/sc16is7xx.c
@@ -523,8 +523,9 @@ static int sc16is7xx_set_baud(struct uart_port *port, int baud)
 
        /* Enable enhanced features */
        regcache_cache_bypass(s->regmap, true);
-       sc16is7xx_port_write(port, SC16IS7XX_EFR_REG,
-                            SC16IS7XX_EFR_ENABLE_BIT);
+       sc16is7xx_port_update(port, SC16IS7XX_EFR_REG,
+                             SC16IS7XX_EFR_ENABLE_BIT,
+                             SC16IS7XX_EFR_ENABLE_BIT);
        regcache_cache_bypass(s->regmap, false);
 
        /* Put LCR back to the normal mode */
@@ -846,7 +847,7 @@ static unsigned int sc16is7xx_get_mctrl(struct uart_port *port)
        /* DCD and DSR are not wired and CTS/RTS is handled automatically
         * so just indicate DSR and CAR asserted
         */
-       return TIOCM_DSR | TIOCM_CAR;
+       return TIOCM_DSR | TIOCM_CAR | TIOCM_RI | TIOCM_CTS;
 }
 
 static void sc16is7xx_set_mctrl(struct uart_port *port, unsigned int mctrl)
@@ -933,14 +934,19 @@ static void sc16is7xx_set_termios(struct uart_port *port,
        regcache_cache_bypass(s->regmap, true);
        sc16is7xx_port_write(port, SC16IS7XX_XON1_REG, termios->c_cc[VSTART]);
        sc16is7xx_port_write(port, SC16IS7XX_XOFF1_REG, termios->c_cc[VSTOP]);
-       if (termios->c_cflag & CRTSCTS)
+       if (termios->c_cflag & CRTSCTS) {
                flow |= SC16IS7XX_EFR_AUTOCTS_BIT |
                        SC16IS7XX_EFR_AUTORTS_BIT;
+               port->status |= UPSTAT_AUTOCTS;
+       };
        if (termios->c_iflag & IXON)
                flow |= SC16IS7XX_EFR_SWFLOW3_BIT;
        if (termios->c_iflag & IXOFF)
                flow |= SC16IS7XX_EFR_SWFLOW1_BIT;
 
+       /* Always set enable enhanced */
+       flow |= SC16IS7XX_EFR_ENABLE_BIT;
+
        sc16is7xx_port_write(port, SC16IS7XX_EFR_REG, flow);
        regcache_cache_bypass(s->regmap, false);
 ```
@beikeland Do you have all these changes or just a subset?

beikeland, 2020-05-13:
```diff
diff --git a/drivers/tty/serial/sc16is7xx.c b/drivers/tty/serial/sc16is7xx.c
index aa1bfd40f790..5f51a5eea127 100644
--- a/drivers/tty/serial/sc16is7xx.c
+++ b/drivers/tty/serial/sc16is7xx.c
@@ -523,8 +523,9 @@ static int sc16is7xx_set_baud(struct uart_port *port, int baud)

        /* Enable enhanced features */
        regcache_cache_bypass(s->regmap, true);
-       sc16is7xx_port_write(port, SC16IS7XX_EFR_REG,
-                            SC16IS7XX_EFR_ENABLE_BIT);
+       sc16is7xx_port_update(port,SC16IS7XX_EFR_REG,
+                               SC16IS7XX_EFR_ENABLE_BIT,
+                               SC16IS7XX_EFR_ENABLE_BIT);
        regcache_cache_bypass(s->regmap, false);

        /* Put LCR back to the normal mode */
@@ -846,7 +847,7 @@ static unsigned int sc16is7xx_get_mctrl(struct uart_port *port)
        /* DCD and DSR are not wired and CTS/RTS is handled automatically
         * so just indicate DSR and CAR asserted
         */
-       return TIOCM_DSR | TIOCM_CAR;
+       return TIOCM_DSR | TIOCM_CAR | TIOCM_RI | TIOCM_CTS;
 }

 static void sc16is7xx_set_mctrl(struct uart_port *port, unsigned int mctrl)
@@ -933,9 +934,11 @@ static void sc16is7xx_set_termios(struct uart_port *port,
        regcache_cache_bypass(s->regmap, true);
        sc16is7xx_port_write(port, SC16IS7XX_XON1_REG, termios->c_cc[VSTART]);
        sc16is7xx_port_write(port, SC16IS7XX_XOFF1_REG, termios->c_cc[VSTOP]);
-       if (termios->c_cflag & CRTSCTS)
+       if (termios->c_cflag & CRTSCTS) {
                flow |= SC16IS7XX_EFR_AUTOCTS_BIT |
                        SC16IS7XX_EFR_AUTORTS_BIT;
+               port->status |= UPSTAT_AUTOCTS;
+       }
        if (termios->c_iflag & IXON)
                flow |= SC16IS7XX_EFR_SWFLOW3_BIT;
        if (termios->c_iflag & IXOFF)
```
is what I applied and seems to work consistently across rpi-5.6.y (9bd27a5b83190f0abedcd67239f064627a66fd63) and HEAD (f0e620550b8b422fef4adcabb2d0e8e69f1fec75) Not sure which version is the latest, HEAD seemed to be 4.x. In any case neither work without the patch, all work with the patch. (Trying to use hciattach with a hci uart which needs flow control)

pelwell, 2020-05-13:
Would you mind adding the other change and re-testing:
```diff
        if (termios->c_iflag & IXOFF)
                flow |= SC16IS7XX_EFR_SWFLOW1_BIT;
 
+       /* Always set enable enhanced */
+       flow |= SC16IS7XX_EFR_ENABLE_BIT;
+
        sc16is7xx_port_write(port, SC16IS7XX_EFR_REG, flow);
        regcache_cache_bypass(s->regmap, false);
```
I'll merge one of the two sets of changes after your feedback.

beikeland, 2020-05-13:
Sure thing, I'll start with the rpi-5.6.y branch, let me know if there is a specific branch you want me to test it against.

pelwell, 2020-05-13:
Either 5.6 or 5.4 - there's been little change, and I'm happy that if it works for one it will work for the other.

beikeland, 2020-05-13:
Still works with `flow |= SC16IS7XX_EFR_ENABLE_BIT;` as far as I can tell.

pelwell, 2020-05-13:
Thanks - that's now in rpi-5.4.y and rpi-5.6.y.