raspberrypi/linux #2543: raspberrypi-firmware soc:firmware: Request 0x00030046 returned status 0x80000001

seniorgod, 2018-05-06:
Dear all,

i have builded a new RT kernel:

4.14.34-rt27-v7+ #1 SMP PREEMPT RT Sun May 6 16:25:04

by following the "standard procedure" described at rasberrypi.org: 

git clone --depth=1 -b rpi-4.14.y-rt  https://github.com/raspberrypi/linux
cd linux
KERNEL=kernel7
make -j4 zImage modules dtbs
sudo cp arch/arm/boot/dts/*.dtb /boot/
sudo cp arch/arm/boot/dts/overlays/*.dtb* /boot/overlays/
sudo cp arch/arm/boot/dts/overlays/README /boot/overlays/
sudo scripts/mkknlimg arch/arm/boot/zImage /boot/$KERNEL.img
shutdown -r now

everything went without problems.

But now i get following messages every two seconds: 

**May  6 16:53:32 raspberrypi kernel: [   69.203312] raspberrypi-firmware soc:firmware: Request 0x00030046 returned status 0x80000001**

What to do? 

best regards 
Matthias

pelwell, 2018-05-06:
Just download the latest firmware from https://github.com/raspberrypi/firmware/tree/master/boot (skipping the kernels) and the messages will stop.

seniorgod, 2018-05-07:
ok,
i clone the repository git clone --depth=1 https://github.com/raspberrypi/firmware
than i deleted the kernels
i make all executable with chmod a+x -R *
i copied all under firmware/boot, **except the deleted kernel7.img and kernel.img**, to /boot 
i copied all under firmware/boot/overlays  to /boot/overlays 

shutdown -r now 

the errors are gone 
thanks!

closed 2018-05-07 by seniorgod