raspberrypi/linux #2548: Bluetooth won't start on Pi3

WayneKeenan, 2018-05-09:
h/w:   RPi 3 B v1.2
OS Base image:  2018-03-13-raspbian-stretch.zip

```
$ uname -a
Linux erxpi 4.14.39-v7+ #1112 SMP Sat May 5 12:01:33 BST 2018 armv7l GNU/Linux
$ cat /boot/.firmware_revision 
252a11c1c90275ead84ffe2e0f0e6d2c573a3bd7
```

Error:

```
 sudo systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
Condition: start condition failed at Wed 2018-05-09 09:32:56 UTC; 1min 24s ago
           └─ ConditionPathIsDirectory=/sys/class/bluetooth was not met
     Docs: man:bluetoothd(8)
```

I'm also seeing similar symptoms on the latest `2018-04-18-raspbian-stretch.zip` vanilla and with `rpi-update` applied, but I didn't capture the logs, the above state is my current 'game'.

WayneKeenan, 2018-05-09:
Trying to launch `bluetoothctl` just hangs.
Manually restarting the `bluetooth` service 'helps', I could scan on `2018-04-18-raspbian-stretch` , but `hciconfig`  on `2018-03-13-raspbian-stretch` shows nothing.

WayneKeenan, 2018-05-09:
Same issue after running `sudo BRANCH=next rpi-update` :
```
cat  /boot/.firmware_revision 
585219e716362e9c82add2ec08510bfab4226781
```

WayneKeenan, 2018-05-09:
please ignore, appears to be a h/w issue, trying the same SDcard in another Pi (exactly same variant of Pi), bluetooth works straight away.

closed 2018-05-09 by WayneKeenan

pelwell, 2018-05-09:
How are you powering your Pi 3? What does "vcgencmd get_throttled" return?

[ Don't run BRANCH=next rpi-update without very good reason - it may be a trick version, or it may actually be older than the current master release (which is the case now) ]

pelwell, 2018-05-09:
Oh, OK. Unless you are using exactly the same supply and cable it might still be worth investigating the power source.

WayneKeenan, 2018-05-09:
Same power supply (one of the official ones I ordered to go with some Pi3 B+'s)

Ah ok, it 'seems' to be ok so far, will revert.   Thanks for asking and mentioning that.

All the best
Wayne