raspberrypi/linux #2551: USB-modem + USB-webcam on RPi3 causes kernel panics

jetibest, 2018-05-12:
I have ten Raspberry Pi 3's with a USB 4G-modem and a USB webcam plugged in. And I managed to get a log which is similar to what @l-spitz describes at #1272 (but I have not been able to reproduce the issue with the dwc_otg buspower toggle) - see down below.

When the webcam is used every 5 minutes using crontab, the problem occurs every 24~48 hours. A kernel oops shows up, and after half an hour of no 4G connection (most of the times the RPi's are offline for half an hour before coming back online after an eventual reboot - however, in rare cases this lasts up to a few hours), the system reboots itself. There is definitively no power supply issue, especially since we've recently tested using an externally powered USB-hub which still does not solve the problem.

Interestingly, the problem *only occurs when **both*** the webcam and modem are plugged in (and used). I have another ten RPi's that have been up and running for about a year now 24/7, using the exact same modem and power supply etc, but without the webcam, and those RPi's have never had any problems.

Next to that, every now and then a warning occurs in the kernel logs: "uvcvideo: Non-zero status (-5) in status completion handler." So far, only when that warning occurs every few hours, I can consistently predict that a reboot will occur. This is an important observation: If the warning does not show up in the logs, for example, when the modem is not plugged in, then the reboot/kernel oops will not occur either. If the warning occurs, I know that some day soon, the rpi will reboot.

Unfortunately I haven't been able to reproduce the problem, it seems to have the properties of a race condition of some sorts. Using the webcam every 5 minutes instead of every hour gives a higher probability of the problem showing up. However, any script where I have tried to force the problem with a shorter interval did not increase the frequency of the problem. I have also tried stressing the RPi with high CPU and other things, but to no avail. Additionally, when the webcam is plugged in but it is not used, the problem does not occur.

Since at first I thought the problem was only related to the webcam, I have tried several uvcvideo modprobe options such as quirks mode and nodrop and several other combinations. These did not have an effect on the problem.

Some more information on the RPi's (Raspbian GNU/Linux 8):
```
> uname -a
Linux weti-anc1711-00 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux
> lsusb
Bus 001 Device 111: ID 05a3:9422 ARC International 
Bus 001 Device 110: ID 12d1:14dc Huawei Technologies Co., Ltd. 
Bus 001 Device 109: ID 1a40:0101 Terminus Technology Inc. 4-Port HUB
Bus 001 Device 108: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 107: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Here are the kernel logs I managed to extract from one of the RPi's (I don't have too many logs, since I have been using `tmpfs` on `/var/log` to reduce SD-card corruption):
```
[568461.998997] uvcvideo: Non-zero status (-5) in status completion handler.
[569970.962252] uvcvideo: Non-zero status (-5) in status completion handler.
[571877.689924] uvcvideo: Failed to resubmit video URB (-1).
[573977.726977] uvcvideo: Failed to resubmit video URB (-1).
[574877.998712] uvcvideo: Failed to resubmit video URB (-1).
[575778.459269] uvcvideo: Failed to resubmit video URB (-1).
[576082.931273] uvcvideo: Failed to resubmit video URB (-1).
[576082.935186] uvcvideo: Failed to resubmit video URB (-1).
[578671.852022] uvcvideo: Non-zero status (-5) in status completion handler.
[579716.748041] uvcvideo: Failed to resubmit video URB (-1).
[584777.153730] uvcvideo: Failed to resubmit video URB (-1).
[586578.011536] uvcvideo: Failed to resubmit video URB (-1).
[595277.433798] Unable to handle kernel NULL pointer dereference at virtual address 00000104
[595277.440310] pgd = b4c50000
[595277.444031] [00000104] *pgd=00000000
[595277.447739] Internal error: Oops: 805 [#1] SMP ARM
[595277.451799] Modules linked in: cdc_ether sr_mod cdrom hci_uart btbcm bluetooth brcmfmac brcmutil cfg80211 rfkill sg evdev snd_bcm2835 bcm2835_gpiomem i2c_bcm2835 uvcvideo snd_usb_audio snd_hwdep snd_usbmidi_lib videobuf2_vmalloc snd_rawmidi videobuf2_memops snd_seq_device videobuf2_v4l2 snd_pcm videobuf2_core snd_timer videodev snd media uio_pdrv_genirq fixed uio i2c_dev fuse ipv6
[595277.471881] CPU: 0 PID: 147 Comm: systemd-journal Not tainted 4.9.24-v7+ #993
[595277.480474] Hardware name: BCM2835
[595277.484650] task: b4d53b00 task.stack: b4d32000
[595277.488994] PC is at uvc_queue_next_buffer+0x7c/0xf0 [uvcvideo]
[595277.493456] LR is at 0x100
[595277.498072] pc : [<7f1e33dc>]    lr : [<00000100>]    psr: 60000193
[595277.498072] sp : b4d33aa8  ip : 00000100  fp : b4d33ac4
[595277.506335] r10: ad446400  r9 : b4f48398  r8 : 00000010
[595277.508553] r7 : b6956400  r6 : b4f484e8  r5 : b4f48398  r4 : b30c2600
[595277.510877] r3 : b4f48398  r2 : 00000200  r1 : 00000000  r0 : 60000193
[595277.513259] Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[595277.517812] Control: 10c5383d  Table: 34c5006a  DAC: 00000055
[595277.520057] Process systemd-journal (pid: 147, stack limit = 0xb4d32210)
[595277.522425] Stack: (0xb4d33aa8 to 0xb4d34000)
[595277.524678] 3aa0:                   00000001 b30c2600 b4f48000 ad446500 b4d33afc b4d33ac8
[595277.529189] 3ac0: 7f1e6ab4 7f1e336c 7f1f06c0 b30c2600 00000000 b30c2600 ad446400 b4f48000
[595277.533654] 3ae0: b4f484e8 80b79340 00000000 00000001 b4d33b1c b4d33b00 7f1e56b8 7f1e69a8
[595277.538091] 3b00: ad446400 60000113 00000000 00000000 b4d33b3c b4d33b20 80556fe8 7f1e5640
[595277.543118] 3b20: 97af5d80 ad446400 ad446400 00000000 b4d33b5c b4d33b40 80557114 80556f74
[595277.548000] 3b40: b638ea74 97af5d80 b59d0400 ad446400 b4d33b84 b4d33b60 80581084 805570d4
[595277.552983] 3b60: b4d33ba4 20000113 804eb324 b59a4ac0 b59a4ac4 00000000 b4d33b94 b4d33b88
[595277.557836] 3b80: 8058fe40 80581024 b4d33bbc b4d33b98 80123070 8058fe2c 80122ffc 80c02080
[595277.562546] 3ba0: 00000000 b4d32000 00000100 b5805180 b4d33c24 b4d33bc0 8010169c 80123008
[595277.567385] 3bc0: 8017400c 80177754 00000001 00400100 80c02d00 038bdb6b 0000000a 80c7a980
[595277.572177] 3be0: 80c71e38 80c0312c 80b793b8 80c02080 b4d33bc0 00000001 80172ecc 80b7cec0
[595277.577441] 3c00: 00000000 00000000 00000001 b5805180 b4d32000 024200ca b4d33c34 b4d33c28
[... 184 lines cut ...]
[595601.666903] 3c80: 00000081 b4d33dd8 80c6f440 00400000 00000000 00000081 80c043a8 00000000
[595601.671693] 3ca0: 024200ca b4d33d5c b4d33d60 b4d33cc8 80214cec 80213a28 a0000113 ffffffff
[595601.676472] [<8071893c>] (__irq_svc) from [<80213a28>] (get_page_from_freelist+0x18/0xb0c)
[595601.681257] [<80213a28>] (get_page_from_freelist) from [<80214cec>] (__alloc_pages_nodemask+0xf0/0xdd0)
[595601.686047] [<80214cec>] (__alloc_pages_nodemask) from [<802420cc>] (handle_mm_fault+0xb58/0xde4)
[595601.690836] [<802420cc>] (handle_mm_fault) from [<807193f4>] (do_page_fault+0x33c/0x3b0)
[595601.695607] [<807193f4>] (do_page_fault) from [<801011e8>] (do_DataAbort+0x48/0xc4)
[595601.700378] [<801011e8>] (do_DataAbort) from [<80718c24>] (__dabt_usr+0x44/0x60)
[595601.705135] Exception stack(0xb4d33fb0 to 0xb4d33ff8)
[595601.707512] 3fa0:                                     76084000 00000000 0006fff8 00000000
[595601.712182] 3fc0: 56768598 7ee04da0 00000000 76e1188c 7ee04c68 7ee04c68 ffffffff 7ee04da0
[595601.716801] 3fe0: 00000000 7ee04c00 00000000 76f910dc 20000010 ffffffff
[595613.690064] INFO: rcu_sched detected stalls on CPUs/tasks:
[595613.692416]         3-...: (1 GPs behind) idle=6bb/140000000000000/0 softirq=25589465/25589472 fqs=16770 
[595613.696982]         (detected by 0, t=33627 jiffies, g=24633640, c=24633639, q=14188)
[595613.701553] Task dump for CPU 3:
[595613.703829] avconv          R  running task        0  7121      1 0x00000003
[595613.706219] [<80713980>] (__schedule) from [<8025ea44>] (__slab_free+0x1f8/0x3a4)
[595676.739299] INFO: rcu_sched detected stalls on CPUs/tasks:
[595676.741616]         3-...: (1 GPs behind) idle=6bb/140000000000000/0 softirq=25589465/25589472 fqs=19920 
[595676.746111]         (detected by 0, t=39932 jiffies, g=24633640, c=24633639, q=16746)
[595676.750647] Task dump for CPU 3:
[595676.752908] avconv          R  running task        0  7121      1 0x00000003
[595676.755229] [<80713980>] (__schedule) from [<8025ea44>] (__slab_free+0x1f8/0x3a4)
```

pelwell, 2018-06-08:
I would recommend trying a newer kernel - we are currently shipping Raspbian based on 4.14.34. Amongst a slew of improvements is a back-port of a patch that fixes a memory corruption bug caused  by the USB driver - see https://github.com/raspberrypi/linux/commit/8960ba68b681a62a523cc593773d5d03a413a075 .

You can either grab the current image, update the relevant Raspbian packages (`sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel`) or use rpi-update to fetch the very latest firmware (`sudo rpi-update`).

jetibest, 2018-06-08:
@pelwell We are currently testing the latest update for which we used `rpi-update`. However, does this fix mean that it should be reproducable using the following code repeatedly?
```
echo 0 >  /sys/bus/platform/drivers/dwc_otg/20980000.usb/buspower
sleep 5
echo 1 >  /sys/bus/platform/drivers/dwc_otg/20980000.usb/buspower
```
See also: #1272

pelwell, 2018-06-08:
Perhaps - it's worth a try.

jetibest, 2018-06-09:
@pelwell As I suspected - unfortunately, the latest update did not fix the problem. The kernel oops still occurs, even after latest `rpi-update`.

    > uname -a
    Linux rpi 4.14.48-v7+ #1118 SMP Thu Jun 7 16:40:27 BST 2018 armv7l GNU/Linux

jetibest, 2018-06-09:
Here are the `dmesg -k` logs after booting the system from the kernel oops frozen state:

	[    0.000000] Booting Linux on physical CPU 0x0
	[    0.000000] Linux version 4.14.48-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1118 SMP Thu Jun 7 16:40:27 BST 2018
	[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
	[    0.000000] CPU: div instructions available: patching division code
	[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
	[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
	[    0.000000] Memory policy: Data cache writealloc
	[    0.000000] cma: Reserved 8 MiB at 0x36800000
	[    0.000000] On node 0 totalpages: 225280
	[    0.000000] free_area_init_node: node 0, pgdat 80c84fc0, node_mem_map b603a000
	[    0.000000]   Normal zone: 1980 pages used for memmap
	[    0.000000]   Normal zone: 0 pages reserved
	[    0.000000]   Normal zone: 225280 pages, LIFO batch:31
	[    0.000000] percpu: Embedded 17 pages/cpu @b5fe3000 s38720 r8192 d22720 u69632
	[    0.000000] pcpu-alloc: s38720 r8192 d22720 u69632 alloc=17*4096
	[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
	[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 223300
	[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait noswap
	[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
	[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
	[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
	[    0.000000] Memory: 871216K/901120K available (7168K kernel code, 575K rwdata, 2072K rodata, 1024K init, 706K bss, 21712K reserved, 8192K cma-reserved)
	[    0.000000] Virtual kernel memory layout:
		vector  : 0xffff0000 - 0xffff1000   (   4 kB)                                                                                                                                  
		fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
		vmalloc : 0xb7800000 - 0xff800000   (1152 MB)
		lowmem  : 0x80000000 - 0xb7000000   ( 880 MB)
		modules : 0x7f000000 - 0x80000000   (  16 MB)
		  .text : 0x80008000 - 0x80800000   (8160 kB)
		  .init : 0x80b00000 - 0x80c00000   (1024 kB)
		  .data : 0x80c00000 - 0x80c8feb4   ( 576 kB)
		   .bss : 0x80c96f40 - 0x80d478b4   ( 707 kB)
	[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
	[    0.000000] ftrace: allocating 25253 entries in 75 pages
	[    0.000000] Hierarchical RCU implementation.
	[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
	[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
	[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
	[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
	[    0.000023] Switching to timer-based delay loop, resolution 52ns
	[    0.000272] Console: colour dummy device 80x30
	[    0.000804] console [tty1] enabled
	[    0.000844] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
	[    0.000884] pid_max: default: 32768 minimum: 301
	[    0.001214] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
	[    0.001247] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
	[    0.002192] Disabling memory control group subsystem
	[    0.002290] CPU: Testing write buffer coherency: ok
	[    0.002701] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
	[    0.003100] Setting up static identity map for 0x100000 - 0x10003c
	[    0.003233] Hierarchical SRCU implementation.
	[    0.003884] smp: Bringing up secondary CPUs ...
	[    0.004591] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
	[    0.005343] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
	[    0.006078] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
	[    0.006182] smp: Brought up 1 node, 4 CPUs
	[    0.006252] SMP: Total of 4 processors activated (153.60 BogoMIPS).
	[    0.006274] CPU: All CPU(s) started in HYP mode.
	[    0.006292] CPU: Virtualization extensions available.
	[    0.007150] devtmpfs: initialized
	[    0.016884] random: get_random_u32 called from bucket_table_alloc+0xfc/0x24c with crng_init=0
	[    0.017547] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
	[    0.017789] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
	[    0.017833] futex hash table entries: 1024 (order: 4, 65536 bytes)
	[    0.018407] pinctrl core: initialized pinctrl subsystem
	[    0.019168] NET: Registered protocol family 16
	[    0.021766] DMA: preallocated 1024 KiB pool for atomic coherent allocations
	[    0.026531] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[... 241 lines cut ...]
	[   15.807713] sr 1:0:0:0: [sr0] tag#0 Sense Key : 0x3 [current] 
	[   15.807719] sr 1:0:0:0: [sr0] tag#0 ASC=0x11 ASCQ=0x0 
	[   15.807727] sr 1:0:0:0: [sr0] tag#0 CDB: opcode=0x28 28 00 00 00 0f fe 00 00 02 00
	[   15.807734] print_req_error: critical medium error, dev sr0, sector 16376
	[   15.807760] attempt to access beyond end of device
	[   15.807765] sr0: rw=0, want=16384, limit=16376
[... cut ...]
	[   20.167064] scsi 1:0:0:0: Direct-Access     HUAWEI   TF CARD Storage  2.31 PQ: 0 ANSI: 2
	[   20.168114] sd 1:0:0:0: Attached scsi generic sg1 type 0
	[   20.176203] sd 1:0:0:0: [sdb] Attached SCSI removable disk
	[   26.102627] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
	[   26.102662] brcmfmac: power management disabled
	[   42.685677] uvcvideo: Failed to resubmit video URB (-1).
	[ 1803.057432] uvcvideo: Failed to resubmit video URB (-1).
	[ 2702.913553] uvcvideo: Failed to resubmit video URB (-1).

jetibest, 2018-06-17:
@pelwell We installed a fresh Arch Linux for the RPi 3B, which still led to the problem (I figure the kernel is just the same). Note that it is unlikely due to a bug in the USB-devices themselves, since the problem only occurs when a combination of both is used. Power supply is definitively not an issue either, as stated before.

It applies to all ten RPi 3B devices, so it is not a coincidental manufacturing error either. We are wasting a lot of time on this issue, if you have any idea to what might cause this issue or help to prevent it, please let us know.

I also see that in RPi 3B+ model some changes were made to the USB architecture, including an additional USB controller. Why was this changed? Could this have anything to do with this issue?

JamesH65, 2018-06-17:
@jetibest The USB connection on the SOC itself is the same. The new ethernet chip (to give us gig ethernet) now has two hubs internally, one hanging off the other, to give enough ports. Not sure why it was done like that, but that's how it is. We just use the chip!

jetibest, 2018-06-17:
@JamesH65 Thank you. We are now testing with a RPi 3B+, to see if the issue still occurs, just to make sure we tried everything before we start looking for alternatives to the RPi. If there is something else we could or should try, I am open to suggestions.

Will this issue be investigated? How do you typically deal with issues like these?

JamesH65, 2018-06-18:
The problem here is that you are using two external devices.  We are currently working full time on ensuring the on board devices work as expected, so there is little time for trying combinations of third party stuff. 

As for how we deal, generally, the bugs that affect a lot of people get investigated first. And of course, we always have new products to develop as well, and as we don't have a dedicated bug fixing team, there is a certain amount of flitting around between new product and bug fixing. 

We also like to have easy ways of replicating issues, on Raspbian. Easily replicated bugs get fixed faster. Bugs reported on other distro's, because we cannot be sure they also go wrong on our distro, tend to be lower on the priority list.

jetibest, 2018-06-18:
@JamesH65 Thank you for your swift reply. I understand, fair enough.

As far as reproducibility goes, when you have the time, you can do the following which has been very consistent for us:

 - RPi 3 Model B with NOOBS Raspbian installation
 - usb webcam (any uvcvideo)
 - usbmodem (e.g. Huawei E3372)

Set a crontab for every 5 minutes running the following script:

    avconv -f video4linux2 -input_format mjpeg -i /dev/video0 -codec:v copy "/tmp/frame.%d.jpg" &
    pid=$!
    sleep 5
    kill -INT $pid
    wait $pid
    rm -f /tmp/frame.*.jpg

Make sure to connect to the internet through the usbmodem, and keep an ssh-connection open (e.g. use `ServerAliveInterval 60` in `/etc/ssh/sshd.conf`). It will take by average ~3 days (of 24 hours) for the kernel oops to occur. Unfortunately we have been unable to reproduce this problem on a shorter notice.

Of course, for many people, an occasional kernel oops may only be a slight annoyance, and it is hard to track down the problem. Many times it may be indistinguishable from a bad power supply. I wouldn't be surprised if this is not reported too often, if at all. But I found these that might be similar:

ajohn370 commented on Dec 14, 2017 "Having a similar issue with an rpi 3 b. System runs smoothly but it freezes hangs and I loose ssh and VNC access." - https://github.com/raspberrypi/firmware/issues/247

l-spitz commented on Sep 26, 2017 "I'm getting a lot of issues similar to what you describe, when running webcams over the usb, namely sporadically get dmesg messages, and eventually get following, which requires a power cycle to recover from." - https://github.com/raspberrypi/linux/issues/1272

zerwes commented on Mar 29 "same issue here" - https://github.com/raspberrypi/linux/issues/2481

l-spitz, 2018-06-18:
Actually I could reproduce it in usually less than an hour, typically 20 mins (rarely a day long), by running two webcams, taking stills as fast as they could. Using a python script (calling linux using os.system), but suspect bash script would do the same .  Each webcam was in its own process.  

Each time it was same thing (kernel NULL pointer dereference at virtual address 0...)
I tried fswebcam and uvccapture both did the same.
There seemed a loose correlation between file size and crashes (eg. a low-res web-cam vs. high res).  
This was on the Rpi 3, not tried it since the B+ came out.

I came to the conclusion that trying to do more than one large file transfer over the usb at the same time was the trigger.  I re-arranged my application to avoid this situation.

JamesH65, 2018-06-18:
Thanks, both useful data points that will help when we get the time. @P33M Any thoughts?

jetibest, 2018-06-29:
I can now confirm that the problem is only reproducable when the webcam is repeatedly turned on and off again to take pictures (using avconv). When the webcam is permanently streaming (avconv mjpeg), there is no issue. It seems like turning the webcam on or off is the cause (however, only in combination with another usb-device).

@l-spitz do you still have your python script, or know what it consisted of? I would like to test your script on our system.

l-spitz, 2018-06-30:
Sure, my script is constantly turning two usb webcams on and off (as an accelerated test) so fits with what you found.  Hope it helps

`
	import sys, os
	import time
	import multiprocessing
	import subprocess	
       
	class OCamera():
		def __init__(self,name,filename,video):
			print(name)
			self.filename = filename
			if video == 1:
				self.device = '/dev/video1'
			elif video == 0:
				self.device = '/dev/video0'
			else:
				raise ValueError("device not supported")
				
			self.cmd_string = "fswebcam -d " + self.device + " -q -r 800x600 --no-banner -S 5 "
			#self.cmd_string = "uvccapture -d" + self.device + " -q81 -x640 -y480 -m -S45 -C30 -G80 -B-96 -o"
			print("command string",self.cmd_string)
		   
			self.killme = multiprocessing.Event()
			self.killme.set()
			self.name = name
			self.process = multiprocessing.Process(target=self.Process, args=(), name=name)
	
		def killProcess(self):
			self.killme.clear()
	
		def start(self):
			self.process.start()
	 
		def Process(self):
			while self.killme.is_set():
				print(self.name)
				os.system(self.cmd_string + self.filename)
				time.sleep(0.1)       
				
		
	
	if __name__ == "__main__":
		procs = list()
		
		try:
			procs.append(OCamera("Camera One",'/home/pi/Temp/cam1_test.jpg',1))
			procs.append(OCamera("Camera Two",'/home/pi/Temp/cam2_test.jpg',0))
	
			print("starting cameras")
			for wc in procs:
				wc.start()
	
		except:
			for wc in procs:
				wc.killProcess()
	
			for wc in procs:
				wc.process.join()
	

`

jetibest, 2018-07-06:
@l-spitz Thank you for the script. Ran it for an hour or so, and later again for 5 hours straight. Unfortunately could not reproduce any problem within a short period of time (even while the usb-modem was also still plugged in). I used two ELP camera modules with USB connectors, on the RPi 3 Model B.

After that I ran the same script for 3 hours, but now with a sleep of 3 seconds (to ensure the webcams are suspended momentarily, which I verified with `cat /sys/bus/usb/devices/1-1.4/power/*`), and I still could not reproduce it on short notice.

So the current status is that it is still not reproducable within hours, but only within days. It's very peculiar that the problem does not occur sooner for a higher frequency of accessing the USB webcams.

l-spitz, 2018-07-07:
What resolution was the camera, I was using a genius HD webcam F100, and also tried with others in the 5MP range (all on RPi 3 Model B).  I re-tried the script before posting, and had three crashes in a 6 hr period.  I had a hunch larger MP caused more crashes.

noafterglow, 2018-08-21:
I think I am having a similar issue.  I am using ELP 5MP webcams ELP-USB500W05G, currently on Rpi 3B but hopefully soon on 3B+ models.  We are using thousands of Pi's so this is a big deal for us. 

Here is the initial trace from kernel....  As you can see, there is some sort of USB bus crash, followed by a USB restart.

`Aug 21 09:00:45 raspberrypi kernel: [12920.270759] usb usb1-port1: disabled by hub (EMI?), re-enabling...
Aug 21 09:00:45 raspberrypi kernel: [12920.270777] usb 1-1: USB disconnect, device number 6
Aug 21 09:00:45 raspberrypi kernel: [12920.270783] usb 1-1.1: USB disconnect, device number 7
Aug 21 09:00:45 raspberrypi kernel: [12920.271116] smsc95xx 1-1.1:1.0 eth0: unregister 'smsc95xx' usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet
Aug 21 09:00:45 raspberrypi kernel: [12920.271175] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
Aug 21 09:00:45 raspberrypi kernel: [12920.331236] usb 1-1.3: USB disconnect, device number 8
Aug 21 09:00:45 raspberrypi kernel: [12920.331372] cdc_acm 1-1.3:1.0: failed to set dtr/rts
Aug 21 09:00:45 raspberrypi kernel: [12920.332048] usb 1-1.5: USB disconnect, device number 9
Aug 21 09:00:45 raspberrypi kernel: [12920.530924] Indeed it is in host mode hprt0 = 00001501
Aug 21 09:00:46 raspberrypi kernel: [12920.740744] usb 1-1: new high-speed USB device number 10 using dwc_otg
Aug 21 09:00:46 raspberrypi kernel: [12920.740844] Indeed it is in host mode hprt0 = 00001101
Aug 21 09:00:46 raspberrypi kernel: [12920.980837] usb 1-1: New USB device found, idVendor=0424, idProduct=9514
Aug 21 09:00:46 raspberrypi kernel: [12920.980844] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Aug 21 09:00:46 raspberrypi kernel: [12920.981246] hub 1-1:1.0: USB hub found
Aug 21 09:00:46 raspberrypi kernel: [12920.981295] hub 1-1:1.0: 5 ports detected
Aug 21 09:00:46 raspberrypi kernel: [12921.300789] usb 1-1.1: new high-speed USB device number 11 using dwc_otg
Aug 21 09:00:46 raspberrypi kernel: [12921.450965] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00
Aug 21 09:00:46 raspberrypi kernel: [12921.450981] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Aug 21 09:00:46 raspberrypi kernel: [12921.453840] smsc95xx v1.0.6
Aug 21 09:00:46 raspberrypi kernel: [12921.544026] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:f8:2f:76
Aug 21 09:00:47 raspberrypi kernel: [12921.640774] usb 1-1.3: new full-speed USB device number 12 using dwc_otg
Aug 21 09:00:47 raspberrypi kernel: [12921.732710] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
Aug 21 09:00:47 raspberrypi kernel: [12921.732934] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
Aug 21 09:00:47 raspberrypi kernel: [12921.782278] usb 1-1.3: New USB device found, idVendor=04d8, idProduct=f058
Aug 21 09:00:47 raspberrypi kernel: [12921.782294] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Aug 21 09:00:47 raspberrypi kernel: [12921.782302] usb 1-1.3: Product: Sensor S7 V1.0
Aug 21 09:00:47 raspberrypi kernel: [12921.782311] usb 1-1.3: Manufacturer: Agnetix Inc.
Aug 21 09:00:47 raspberrypi kernel: [12921.782320] usb 1-1.3: SerialNumber: EA2622F2D54C5490704C0403E12F2958
Aug 21 09:00:47 raspberrypi kernel: [12921.785171] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
Aug 21 09:00:47 raspberrypi kernel: [12922.080722] usb 1-1.5: new high-speed USB device number 13 using dwc_otg
Aug 21 09:00:47 raspberrypi kernel: [12922.216313] usb 1-1.5: New USB device found, idVendor=05a3, idProduct=9520
Aug 21 09:00:47 raspberrypi kernel: [12922.216327] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Aug 21 09:00:47 raspberrypi kernel: [12922.216336] usb 1-1.5: Product: HD USB Camera
Aug 21 09:00:47 raspberrypi kernel: [12922.216344] usb 1-1.5: Manufacturer: HD Camera Manufacturer
Aug 21 09:00:47 raspberrypi kernel: [12922.217250] uvcvideo: Found UVC 1.00 device HD USB Camera (05a3:9520)
Aug 21 09:00:47 raspberrypi kernel: [12922.348073] uvcvideo 1-1.5:1.0: Entity type for entity Extension 2 was not initialized!
Aug 21 09:00:47 raspberrypi kernel: [12922.348089] uvcvideo 1-1.5:1.0: Entity type for entity Processing 3 was not initialized!
Aug 21 09:00:47 raspberrypi kernel: [12922.348102] uvcvideo 1-1.5:1.0: Entity type for entity Camera 1 was not initialized!
Aug 21 09:00:47 raspberrypi kernel: [12922.348480] input: HD USB Camera as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.0/input/input2
Aug 21 09:00:48 raspberrypi kernel: [12923.151520] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Aug 21 09:00:48 raspberrypi kernel: [12923.153739] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC5E1
Aug 21 11:51:27 raspberrypi kernel: [23162.284857] usb usb1-port1: disabled by hub (EMI?), re-enabling...
Aug 21 11:51:27 raspberrypi kernel: [23162.284875] usb 1-1: USB disconnect, device number 10
Aug 21 11:51:27 raspberrypi kernel: [23162.284881] usb 1-1.1: USB disconnect, device number 11
Aug 21 11:51:27 raspberrypi kernel: [23162.285232] smsc95xx 1-1.1:1.0 eth0: unregister 'smsc95xx' usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet
Aug 21 11:51:27 raspberrypi kernel: [23162.285291] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
Aug 21 11:51:27 raspberrypi kernel: [23162.346627] usb 1-1.3: USB disconnect, device number 12
Aug 21 11:51:27 raspberrypi kernel: [23162.346829] cdc_acm 1-1.3:1.0: failed to set dtr/rts
[... cut ...]
We have several issues at this point.
1) as you can see it appears the kernel input system resources do not seem to be freed...
2) the camera moves between video0 and video1, we chase it in our python script.
3) the network gets taken down at the same time causing a whole host of other issues.

we have tried it both stock and with dwc_otg.fiq_enable=1 dwc_otg.fiq_fsm_enable=1 dwc_otg.fiq_fsm_mask=0xf.  No difference.

If you would like our exercise script, please let me know, and I will send it.

jetibest, 2018-08-21:
@noafterglow Thank you for your contribution to this issue! We worked around the problem exactly the same way. Since we need to grab frames, and stream video with different settings etc. I wrote a service that keeps the camera always running on MJPEG stream mode. Then another program can pipe these frames as input, using ffmpeg to generate either individual JPEG files, or e.g. a live H264 in TS stream - even at the same time. Using this modular approach, the webcam never needs to be disabled (and `v4l2-ctl` can change the settings in realtime), and actually gave the advantage of being able to take pictures at the same time as videostream etc etc.

Even though this workaround is stable, we would really prefer to keep power consumption down as we are using solar panels to power the system.

On the longer term, I also found that sometimes there are suddenly `/dev/video0` and `/dev/video1`. But I think this is unrelated to this specific issue, since the camera is permanently enabled now. Additionally this did not interfere with our application.

Could you provide your exercise script? It would be awesome to have a script that can consistently reproduce this problem in a shorter period of time - if you managed to do this. We sometimes had to wait up to a week to reproduce this issue, which I believe is the reason we could not solve the problem - let alone find the root cause. How fast are you able to consistently reproduce the issue on a single RPi?

noafterglow, 2018-08-22:
basically the camera was operated from cv2, with the following sequence (It was spread out over two files...:

self._capture = cv2.VideoCapture(port)

        self.default_width = self._capture.get(CAP_PROP_FRAME_WIDTH)
        self.default_height = self._capture.get(CAP_PROP_FRAME_HEIGHT)
        self.default_fps = self._capture.get(CAP_PROP_FPS)
        self._capture.set(CAP_PROP_FRAME_WIDTH, width)
        self._capture.set(CAP_PROP_FRAME_HEIGHT, height)
        self.width = self._capture.get(CAP_PROP_FRAME_WIDTH)
        self.height = self._capture.get(CAP_PROP_FRAME_HEIGHT)
        self.fps = self._capture.get(CAP_PROP_FPS)

for i in xrange(10):
     self.grab()

_, image = self._capture.retrieve()
self._capture.release()

noafterglow, 2018-08-22:
The disconnect happens at release().

noafterglow, 2018-11-06:
The power supply is a 3A 5.25V supply with with a great feedback arrangement, and class B EMI components.  I know because I designed it, and the test labs tested it, which is what I do all the time.  This is NOT and EMI issue either, unless the Pi board itself is poorly designed.  There is no way for the Pi to see any change in voltage as a result of the webcam being turned on and off.  There is way too much capacitance there for the webcam's measly 100mA draw to make any difference.  The camera is out of range of people touching it, so ESD is not it either.  This is a bug.  

I don't know what the Pi can do to make the root hub disconnect, but its doing it internally.  The issue happens at different rates on different Pi's, some have a couple of times a day, some every few minutes, some go days between events.  Mostly is a couple of times per day.  It seems to follow the Pi, and not the location.

jetibest, 2018-11-28:
In another attempt to fix this problem, I am going to modify the USB-driver configuration - apparently there are many options to try. More specifically, I will be changing the dwc_otg module (using modprobe or in /boot/cmdline.txt) parameters a.k.a. "DesignWare USB 2.0 OTG Controller (DWC_otg)".

References:
[Parameter reference](https://www.cl.cam.ac.uk/~atm26/ephemeral/rpi/dwc_otg/doc/html/module%20parameters.html)
[Parameter reference specifically for BCM2708](https://elinux.org/RPI_BCM2708_Parameters)
"dwc_otg.fiq_fsm_enable=0 has fixed the crash, but the audio quality is still very poor with this device." - [Issue 588: Kernel Oops when trying to record from USB Audio Mic](https://github.com/raspberrypi/linux/issues/588)

Obviously, messing with these parameters will require thorough testing, and many modifications may not work at all. Will report back in a few months.

@noafterglow this information may also be interesting to you.

ysirotin, 2019-03-23:
Did the USB driver modifications work for resolving the issue?

noafterglow, 2019-03-24:
I have not investigated the issue in the last 6 months.  We have just
loaded the latest kernel and will go looking at the logs again to see if
the problem persists.
Stay tuned.

Ihor Lys
617-470-2740

ysirotin, 2019-03-29:
We have narrowed this a similar issue down specifically to the USB port used.  When plugging the webcam into the 1-1.3 port on the 3B, we consistently get errors/resets like those documented above.  However, when plugging a webcam into 1-1.4 port, we do not see the errors (or the errors are much more infrequent).  I am honestly not sure what the difference is between these ports.

pelwell, 2019-03-29:
Is it the specific ports or the relative position of the webcam compared to the other devices? With 2 devices and 4 ports you have 12 possible combinations.

ysirotin, 2019-03-29:
We only have one thing (Logitech C920) plugged into the USB port of the Pi (no keyboard/mouse).  We did not exhaustively test each port yet.

noafterglow, 2019-03-29:
I suspect that this is actually driven by an issue in the hub itself. This
may be why the Pi foundation dumped that part on the latest pi.  The only
real way to know is to hook up a USB analyzer to both the internal port
which hooks to the hub as well as the port used for the webcam and look at
what happens during the event.

Now that that Pi is EOL, its over, and we only have the next version to
deal with.

JamesH65, 2019-03-29:
We didn't dump the part - still used on the Pi1, 2, and 3 which are still in production, not EOL. Yes, you can still get Pi 1 if you really want! For the Pi3B+ we simply moved to a part that supported gig ethernet.

noafterglow, 2019-03-29:
Ok.  Then perhaps setting up a test system is worth it.  As i
mentioned earlier, i have a 100% method for causing this problem,  using
the shutoff of a webcam from user mode.  I dont know if its the same
problem here, but it definitely has the same symptom.  I dont have the
proper equipment to snoop the intermediate buses to see what actually
happens, but id bet that a day of work on that would probably be
very useful. Otherwise i fear that this is really  nothing more than
guessing as to parameters i certainly dont understand.

I will set up my system and run it and let you know if the latest kernel
solves the problem.  However, we are upgrading to to latest pi on our next
build, as well as moving the camera to the csi connector, so we wont see
this exact issue again.

juliogonzalez, 2020-08-17:
I case it can be useful, I am having something very similar with an old Hauppage WinTV-USB (analog TV tuner, works as `v4l2` device) that works flawlessly connected to any PC, but suffers the `disabled by hub (EMI?), re-enabling...` when used for streaming on either a Raspberri Pi 3A+ or a Raspberry Pi 3B+

I tried pretty much everything: 
- Changing cables
- Changing power adapters
- Adding ferrite cores
- Buying a new MultiTT hub from UUGear.
- Changing to a Raspberry Pi 3B+ and connecting the device to one of the ports, without using any hub
- Playing with `cmdline.txt` and `dwc_otg`

All of it on different combinations, including everthing at the same time.

My only conclusion is that, being a USB 1.0 (according to `lsusb`, that also shows it consumes max 200mA) is causing some kind of problems. Maybe what's described at https://www.raspberrypi.org/documentation/hardware/raspberrypi/usb/README.md#knownissues `2. USB 1.1 webcams` (even if if in my case the analog tuner provides only 320x240 video)

I'd rule any kind of real EMI problem as if I place the USB cable following the same path, but connected to a PC instead of connected to a Raspberry, things work fine, with or without ferrite beads.

The exact same setup using a Terratec Cinergy Hybrid T USB XS (USB 2.0, includes an analog TV tuner supported as `v4l2` device, and consumes max 500mA) works just fine.

Not complaining! But if you need any more info, and you think it can be useful for this issue, just ping :-)

I wonder, is any one of the persons reporting problems using any USB 1.X device?

jboero, 2020-11-12:
Hi I know this is old but same issue here for months.  Happens randomly between boots, so I think it must be some kind of race condition.  That and the C920 webcam isn't officially supported by Linux.  Still nothing should ever cause a null dereference in the kernel.

Linux z640 5.8.18-200.fc32.x86_64 #1 SMP Mon Nov 2 19:49:11 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

jboero, 2020-11-12:
@jetibest I just found it.  Who needs to check pointers before dereference right?  Submitting a fix.  The webcam might error (still not supported in Linux) but it won't Oops the kernel.

https://github.com/raspberrypi/linux/blob/1c64f4bc22811d2d371b271daa3fb27895a8abdd/drivers/usb/core/usb.c#L276

JamesH65, 2020-11-12:
That's in unmodified upstream code - have you passed it on to the correct upstream Linux mailing list?

jboero, 2020-11-12:
I figured it was.  Still trying to hunt it down upstream.  Hold on the PR for now thanks!

6by9, 2020-11-12:
get_maintainer.pl is your friend.
```
:~/linux$ ./scripts/get_maintainer.pl drivers/usb/core/usb.c
Greg Kroah-Hartman <gregkh@linuxfoundation.org> (supporter:USB SUBSYSTEM,commit_signer:2/1=100%,authored:1/1=100%,added_lines:2/2=100%,removed_lines:3/3=100%)
Felipe Balbi <balbi@kernel.org> (commit_signer:1/1=100%)
linux-usb@vger.kernel.org (open list:USB SUBSYSTEM)
linux-kernel@vger.kernel.org (open list)
```
linux-kernel@vger.kernel.org is the mailing list of last resort, so go for Greg KH and linux-usb initially.

jboero, 2020-11-12:
Yep thanks I've submitted and sent the email to the usb list.
https://github.com/torvalds/linux/commit/a40519014549f60969c8e67a2fd91426db05fe04

glowfishDE, 2021-05-22:
Hi,
i´m having the same issue on the 4.9.201-tegra Kernel on the Nvidia Jetson Nano.

Any suggestion what i could do ?

[Sa Mai 22 21:23:18 2021] uvcvideo: Failed to resubmit video URB (-1).
[Sa Mai 22 22:23:12 2021] uvcvideo: Failed to resubmit video URB (-1).
[Sa Mai 22 22:23:13 2021] uvcvideo: Failed to resubmit video URB (-1).
[Sa Mai 22 22:23:13 2021] uvcvideo: Failed to resubmit video URB (-1).
[Sa Mai 22 22:23:13 2021] uvcvideo: Failed to resubmit video URB (-1).
[Sa Mai 22 22:23:13 2021] uvcvideo: Failed to resubmit video URB (-1).
^C
root@bellatrix:~# uname -r
4.9.201-tegra
root@bellatrix:~# uname -a
Linux bellatrix 4.9.201-tegra #1 SMP PREEMPT Wed May 5 09:31:36 PDT 2021 aarch64 aarch64 aarch64 GNU/Linux

JamesH65, 2021-05-23:
> Hi,
> i´m having the same issue on the 4.9.201-tegra Kernel on the Nvidia Jetson Nano.
>
> Any suggestion what i could do ?
>
> [Sa Mai 22 21:23:18 2021] uvcvideo: Failed to resubmit video URB (-1).
> [Sa Mai [...]

Ask on a Jetson forum?

closed 2021-05-25 by P33M

wolverin-a, 2023-01-26:
> @jetibest I just found it. Who needs to check pointers before dereference right? Submitting a fix. The webcam might error (still not supported in Linux) but it won't Oops the kernel.
> 
> https://github.com/raspberrypi/linux/blob/1c64f4bc22811d2d371b271daa3fb27895a8abdd/drivers/usb/core/usb.c#L276

did this solution help you solve the problem with the usb camera?

jetibest, 2023-01-26:
Good question, forgot all about this issue, since the workaround of keeping the webcam on at all times has been working flawlessly. But I might be testing this again, to check.

However, I am skeptical about the solution ( @jboero ). From my limited understanding, the `dev->actconfig` is used, but `dev` might be NULL? So this is fixed by checking if `dev` is `NULL` or not. But if that is really the problem, does an extra NULL-check at this point fix the issue?

In my view, the function should indicate in its usage documentation that the `dev` that is passed must not be NULL. Because let's take a look at the actual usage of this function, for example in the UVC driver: https://elixir.bootlin.com/linux/latest/source/drivers/media/usb/uvc/uvc_driver.c#L922

The function in question is called as `intf = usb_ifnum_to_if(udev, buffer[12+i]);`. So here, `udev` is presumably NULL when the issue occurs. If that is the case, then even if a check is applied, the `uvc_dbg` message that logically follows, should also have the same issue since it uses that same pointer `udev->devnum`. And there are also other functions in the same file, where `dev->udev` is used directly without any NULL-check, those would potentially fail in the same way.

wolverin-a, 2023-01-26:
I keep the camera on all the time, but that doesn't solve the problem

I increased the buffers of the uvcvideo.ko driver as suggested [https://forums.developer.nvidia.com/t/currupted-frames-from-ar1820-usb-camera/42919/9](here) and that didn't solve the problem either.

Checking for NULL, there is no linux kernel branch in 5.x or 6.x, so this is also not a solution.