raspberrypi/linux #2555: Kernel oops or hard freeze when streaming video on Zero W (and Pi 3B+)

balboah, 2018-05-13:
This is a summary from troubleshooting on https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=213423&p=1315106 

When trying to stream video on Zero W with the noir v2 camera module, I get many variants of kernel oopses and freezes. For example:

```
[  273.802531] random: crng init done
[  379.898672] ------------[ cut here ]------------
[  379.901131] kernel BUG at Returning to usermode but unexpected PSR bits set?:5!
[  379.903359] Internal error: Oops - BUG: 0 [#1] ARM
[  379.905866] Modules linked in: cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic brcmfmac brcmutil cfg80211 snd_bcm2835(C) snd_pcm rfkill snd_timer snd uio_pdrv_genirq uio fixed ip_tables x_tables ipv6
[  379.913291] CPU: 0 PID: 826 Comm: modprobe Tainted: G         C      4.14.34+ #1110
[  379.917705] Hardware name: BCM2835
[  379.920005] task: d6a0e120 task.stack: d5662000
[  379.922689] PC is at no_work_pending+0x30/0x34
[  379.925152] LR is at 0xbea4f970
[  379.927229] pc : [<c000fe54>]    lr : [<bea4f970>]    psr: 20000013
[  379.929588] sp : d5663fa8  ip : d5663fa8  fp : 00000000
[  379.932763] r10: 00000000  r9 : d5662000  r8 : c000ff64
[  379.935421] r7 : 000000d9  r6 : 00d321d8  r5 : 00000020  r4 : 00000078
[  379.938231] r3 : c0939414  r2 : d5663fe4  r1 : b6e6f198  r0 : 00000000
[  379.940929] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  379.943654] Control: 00c5387d  Table: 15664008  DAC: 00000055
[  379.946464] Process modprobe (pid: 826, stack limit = 0xd5662188)
[  379.948699] Stack: (0xd5663fa8 to 0xd5664000)
[  379.951177] 3fa0:                   00d321b8 00000020 00000000 00d321d8 00008000 00000000
[  379.956042] 3fc0: 00d321b8 00000020 00d321d8 000000d9 00000002 0002b990 0003f030 bea4fb10
[  379.960686] 3fe0: 0003efa8 bea4f970 b6e6f198 b6e6f0b8 60000010 00000000 00000000 00000000
[  379.966304] Code: e9527fff e1a00000 e28dd048 e1b0f00e (e7f001f2)
[  379.969043] ---[ end trace ee6907230b405e54 ]---
```

It can easily be reproduced by these steps:

1. Reset to the 2018-04-18-raspbian-stretch-lite.img and enable the headless SSH and /boot/wpa_supplicant.conf
2. raspi-config -> enable camera module -> reboot
3. (terminal 1)$ raspivid -t 0 -w 1280 -h 720 -o /dev/null
4. (terminal 2)$ ssh pi@the-zero-ip
5. Repeat step 4 until crash (will happen within 5 minutes)

I have also had freezes on Pi 3B+, but have not been able to create a proper reproduce or catched any kernel messages. It might be more related to https://github.com/raspberrypi/linux/issues/2387

lategoodbye, 2018-05-14:
Could you please try http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2018-03-14/ with the Zero W?

balboah, 2018-05-15:
@lategoodbye 

When leaving a `watch ssh pi@192.168.1.201 date` in my terminal
after the pi had been running `raspivid -t 0 -w 1280 -h 720 -o /dev/null` for maybe 15 minutes, I got:

dmesg:
```
[ 1473.345882] systemd: 33 output lines suppressed due to ratelimiting
[ 1551.070104] systemd-journald[108]: Failed to send WATCHDOG=1 notification message: Connection refused
```

journald:
```
sshd[857]: Accepted publickey for pi from 192.168.1.14 port 54892 ssh2: RSA SHA256:xxx
sshd[857]: pam_unix(sshd:session): session opened for user pi by (uid=0)
systemd[1]: [35B blob data]
systemd[1]: :0, function (null)(). Aborting.
systemd[1]: Caught <ABRT>, dumped core as pid 861.
kernel: systemd: 33 output lines suppressed due to ratelimiting
systemd[1]: Freezing execution.
systemd-logind[251]: Failed to start session scope session-c19.scope: Message recipient disconnected from message bus without replying
sshd[857]: pam_systemd(sshd:session): Failed to create session: Connection timed out
sshd[867]: Received disconnect from 192.168.1.14 port 54892:11: disconnected by user
```

But it seems I'm unable to provoke a kernel oops!
I guess it's just something fishy with authenticating new logins too fast in this older version

amovitz, 2018-05-23:
I'm having this issue when streaming video for long periods of time, even if the CPU average usage is below 30%. The 3B+ hard locks some time between 30 minutes to a few hours, consistently. Using the exact same SD card image, it works perfectly stable for over a week on a 3B. It will usually not let the hardware or software watchdog kick it after the hard lock occurs. I've been seeing this behavior since I got the 3B+ on day one.

JamesH65, 2018-05-23:
@amovitz What version of kernel are you running? Does rpi-update fix the issue.

JamesH65, 2018-05-23:
If you drop the max CPU frequency to 1200, does that help?

balboah, 2018-05-23:
@amovitz I have that same problem on 3B+ as well without any camera attached, from start it has frozen within a day or maximum 2 days. The recent crash logs are from my Zero W though. I never managed to get a dump from the 3B+, so maybe or maybe not two different issues. And these two rPi's are my first experience,  not great :P

JamesH65, 2018-05-23:
Almost certainly different problems, the SoC on the Zero and 3B+, along with the different wireless chips means they are sufficiently dissimilar. Unless (!) there is something in your environment that is triggering the fault. We've sold a lot of Pi's which run for years without issue, so this is unusual. First check is power supply, is it sufficient?

amovitz, 2018-05-23:
The power supply I'm using is the branded one. we've tried many different power supplies and it will even hard lock when no applications are running. 

We don't use any Zeros, so I couldn't tell you if the same happened for some reason. We have extensively tested with the 3B, running for months at a time, and never had an issue until we put the SD card into a 3B+ (and upgraded the boot for it). Same image works perfectly well on a 3B, but the 3B+ will constantly lock up. 

So far, it's been a little over 6 hours, and it hasn't locked with the lowered CPU Freq, but we'll see within 24 hours if it's more stable.

From our testing, it _feels_ like it is an issue with either the upgraded SoC or another chip on the board which has changed that the Kernel is not handling properly or that the Broadcom firmware is not interfacing properly with the new chip.

balboah, 2018-05-24:
@JamesH65 yeah I would have expected more people to react if it was common, I must have hit the jackpot with 2 out of 2. My power is the raspberry branded ones which I'm assuming has the correct rating.

The thing both have in common is that the SDCard brand is Sandisk class 10 and that it is written to by `dd` on mac.

#### Zero W
The installed image is not customized in any way except for configuring WiFI, camera module via raspi-config and SSH access.

#### 3B+
This one runs Kubernetes master which averages on 20% cpu.
I also tried a more recent version with `rpi-update` without luck. Have not yet tried the frequencey limit. I no longer have a camera or camera module active on this one, which still freeze

amovitz, 2018-05-24:
About 1 hour and 15 minutes after my last post (Total Uptime: 07:35:46), it hard locked.

JamesH65, 2018-05-24:
Hmm, if it had kept running at 1200 I would have thought it was one of the test escapes (i.e. an SoC that passed test but shouldn't have). But I believe they work at 1200. So confused, and not sure what to suggest. Might be worth RMA'ing for a replacement.

balboah, 2018-05-24:
@amovitz and when it freeze it become burning hot right. sounds very much the same as my 3B+

JamesH65, 2018-05-24:
Burning hot is bad. That would indicate a broken SoC I suspect.

amovitz, 2018-05-25:
That's my issue, too. It may be a factory defect since we got the first batch on launch day. It still freezes even when it's clocked at 1200MHz

I would definitely agree that it is high above ambient temperature. We even have a custom heatsink on it with a fan and it's still very hot after it hard locks. If it will help, I can try to get external thermal readings and thermal images, but I'm pretty certain it's the SoC that's hot.

JamesH65, 2018-05-25:
@popcornmix @pelwell Any thoughts?

zoff99, 2018-05-25:
@amovitz i had the same thing with some PIs.
if you just overvoltage (not overclock) it got a bit better in my tests.

```
add to /boot/config.txt:
over_voltage=5
```

amovitz, 2018-05-29:
Overvoltage did not help, it still hard locks.

imbashamba, 2018-06-01:
setup: pi 3B+, Linux raspberrypi 4.14.44-v7+ #1117. Completely stuck after ~1 hour of playing video. gpu_mem=256, everything other is by default.
Nothin in syslog, but some strange warnings in dmesg at boot time:

```
[   41.957094] ------------[ cut here ]------------
[   41.957115] WARNING: CPU: 0 PID: 585 at drivers/staging/vc04_services/interface/vchiq_arm/vchiq_2835_arm.c:541 vchiq_prepare_bulk_data+0x4d0/0x6dc
[   41.957118] Modules linked in: cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic brcmfmac brcmutil cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd uio_pdrv_genirq fixed uio ip_tables x_tables ipv6
[   41.957173] CPU: 0 PID: 585 Comm: info-beamer Tainted: G         C      4.14.44-v7+ #1117
[   41.957176] Hardware name: BCM2835
[   41.957193] [<8010ffd8>] (unwind_backtrace) from [<8010c240>] (show_stack+0x20/0x24)
[   41.957201] [<8010c240>] (show_stack) from [<80785424>] (dump_stack+0xd4/0x118)
[   41.957210] [<80785424>] (dump_stack) from [<8011da4c>] (__warn+0xf8/0x110)
[   41.957217] [<8011da4c>] (__warn) from [<8011db34>] (warn_slowpath_null+0x30/0x38)
[   41.957224] [<8011db34>] (warn_slowpath_null) from [<80666194>] (vchiq_prepare_bulk_data+0x4d0/0x6dc)
[   41.957233] [<80666194>] (vchiq_prepare_bulk_data) from [<8065e078>] (vchiq_bulk_transfer+0x2e4/0x568)
[   41.957241] [<8065e078>] (vchiq_bulk_transfer) from [<80663aec>] (vchiq_ioctl+0x1344/0x1a14)
[   41.957250] [<80663aec>] (vchiq_ioctl) from [<8029de50>] (do_vfs_ioctl+0xac/0x7c4)
[   41.957257] [<8029de50>] (do_vfs_ioctl) from [<8029e5ac>] (SyS_ioctl+0x44/0x6c)
[   41.957266] [<8029e5ac>] (SyS_ioctl) from [<80108060>] (ret_fast_syscall+0x0/0x28)
[   41.957269] ---[ end trace 0dbf65047004fe73 ]---
```

balboah, 2018-06-02:
So I've gotten 2nd Zero W and it seems that at least for that model, it's a hardware issue.
Running the same camera module and sdcard installation works fine without any issues (for at least 30 minutes so far).

I noticed one difference on the hardware layout, the "H" marking on the chip closest to the camera module connector. Not sure if it helps with anything but here are photos of the two:

### Broken one
[image]

### Working one
[image]

amovitz, 2018-08-07:
@balboah Are you still having issues after the latest kernel update?

usedbytes, 2018-09-03:
I'm seeing this (or something similar) on the latest kernel (and whatever kernel I was running before I upgraded - still 4.14). It's a Pi Zero W.

`raspistill` works fine, and will happily capture JPEG and sit with the camera preview running for dozens of seconds. On the other hand `raspivid` (or RPi-Cam-Web-Interface) will immediately panic the kernel.

The reported process in the Oops and the backtrace appears to be random, though this one does show vc.ril.video_en. Maybe the video codec is trashing random bits of memory somehow?

I don't have a serial port hooked up, and it normally crashes before the Oops makes it onto the SSH session, so gathering useful crash dumps is a bit hard.

```
[  421.582317] Unable to handle kernel paging request at virtual address 00001038
[  421.589872] pgd = d3fc8000
[  421.592639] [00001038] *pgd=0d7c8831, *pte=00000000, *ppte=00000000
[  421.599216] Internal error: Oops: 17 [#1] ARM
[  421.603660] Modules linked in: fuse rfcomm cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic evdev spidev brcmfmac brcmutil cfg80211 rfkill snd_bcm2835(C) gpio_keys snd_pcm snd_timer snd spi_bcm2835 fixed uio_pdrv_genirq uio hid_sony ff_memless i2c_dev ip_tables x_tables ipv6
[  421.628996] CPU: 0 PID: 1000 Comm: vc.ril.video_en Tainted: G         C      4.14.67+ #1139
[  421.637475] Hardware name: BCM2835
[  421.640930] task: cd7f3780 task.stack: cd454000
[  421.645556] PC is at grab_cache_page_write_begin+0x20/0x3c
[  421.651146] LR is at ext4_da_write_begin+0xb8/0x438
[  421.656096] pc : [<c00f5020>]    lr : [<c02025b0>]    psr: 60000013
[  421.662453] sp : cd455d40  ip : cd455d58  fp : cd455d54
[  421.667751] r10: c01fbcc4  r9 : cd455df4  r8 : 00001000
[  421.673053] r7 : 00001000  r6 : c8903638  r5 : d6ec81e0  r4 : d7d697a8
[  421.679675] r3 : d7fc3320  r2 : 0000000e  r1 : 000000f2  r0 : 00001000
[  421.686298] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  421.693539] Control: 00c5387d  Table: 13fc8008  DAC: 00000055
[  421.699369] Process vc.ril.video_en (pid: 1000, stack limit = 0xcd454188)
[  421.706259] Stack: (0xcd455d40 to 0xcd456000)
[  421.710688] 5d40: d7fc3320 d7d697a8 cd455dc4 cd455d58 c02025b0 c00f500c 00000000 c8903638

Message from syslogd@minimouse at Sep  3 21:48:04 ...
 kernel:[  421.599216] Internal error: Oops: 17 [#1] ARM
[  421.718991] 5d60: 00001000 00001000 000f3000 00000000 00001000 00000000 000000f2 00001000
[  421.727296] 5d80: d7d69784 00000000 000f2000 00000000 d689fd20 00000000 c0204a60 00000000
[  421.735600] 5da0: 00001000 c8903714 cd455ef0 00001000 00001000 c065f388 cd455e24 cd455dc8
[  421.743903] 5dc0: c00f51d4 c0202504 00001000 00000000 cd455df0 cd455df4 c8903638 cd454000
[  421.752206] 5de0: 00008000 d689fd20 000f2000 00000000 d7d69784 00000000 5b8dac14 00000000
```

balboah, 2018-09-04:
@amovitz sorry for the late reply. After replacing the hardware I am no longer experiencing the problem (still the same sdcard) and I've stopped using the broken one. Maybe there are faulty hardware or some small differences which cause the bug to trigger

usedbytes, 2018-09-08:
I did some experiments. On a different Pi Zero, my card + camera + power supply worked fine. That seems to point to an issue with my actual hardware, supporting @balboah's theory.

Another example of an Oops when running the camera below:

```
[  153.531076] NOHZ: local_softirq_pending 40
[  278.981974] NOHZ: local_softirq_pending 40
[  282.294591] NOHZ: local_softirq_pending 40
[  282.716442] ------------[ cut here ]------------
[  282.721176] kernel BUG at Returning to usermode but unexpected PSR bits
set?:5!
[  282.728604] Internal error: Oops - BUG: 0 [#1] ARM
[  282.733469] Modules linked in: fuse rfcomm cmac bnep hci_uart btbcm serdev
bluetooth ecdh_generic evdev spidev brcmfmac brcmutil gpio_keys snd_bcm2835(C)
cfg80211 snd_pcm snd_timer rfkill snd i2c_bcm2835 spi_bcm2835 uio_pdrv_genirq
uio fixed hid_sony ff_memless i2c_dev ip_tables x_tables ipv6
[  282.759865] CPU: 0 PID: 596 Comm: apache2 Tainted: G         C
4.14.67+ #1139
[  282.767548] Hardware name: BCM2835
[  282.770999] task: cc858000 task.stack: cc89a000
[  282.775623] PC is at no_work_pending+0x30/0x34
[  282.780140] LR is at 0xbeb31b40
[  282.783329] pc : [<c000fe54>]    lr : [<beb31b40>]    psr: 20000013
[  282.789685] sp : cc89bfa8  ip : cc89bfa8  fp : 00000000
[  282.794984] r10: 00000000  r9 : cc89a000  r8 : c000ff64
[  282.800282] r7 : 0000008c  r6 : b5a621b4  r5 : b5a620a0  r4 : 00000000
[  282.806903] r3 : c093d414  r2 : cc89bfe4  r1 : b60bb98c  r0 : 00000000
[  282.813526] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment
user
[  282.820768] Control: 00c5387d  Table: 0c8b8008  DAC: 00000055
[  282.826598] Process apache2 (pid: 596, stack limit = 0xcc89a188)
[  282.832694] Stack: (0xcc89bfa8 to 0xcc89c000)
[  282.837119] bfa0:                   00000000 b5a620a0 0000000b 00000000
00002518 beb31b40
[  282.845421] bfc0: 00000000 b5a620a0 b5a621b4 0000008c b60bb954 b6f6bce8
00000000 beb3650c
[  282.853724] bfe0: b623e470 beb31b40 b60bb98c b6e16ba8 20000010 0000000b
00000000 00000000
[  282.862041] Code: e9527fff e1a00000 e28dd048 e1b0f00e (e7f001f2) 
``` 
Pretty annoyed that this seems to be a HW issue, as this zero is now deeply embedded in the project.

usedbytes, 2018-09-08:
With further fiddling, and following @zoff99 and this forum thread: https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=212777, I resolved my issue by over-volting.

Seems like Broadcom/RPi shipped a batch or two of marginal chips.

manoloromero, 2019-02-23:
The same sitution for me, crash and the board has the H and the dot.
Dod someone found a solution? Like a kernel patch, etc.

JamesH65, 2019-02-25:
Well, it;s not a particularly repeatable problem, as most people do not see an issue. I pressume you

1) Have a decent power supply
2) Are using up to date software?

manoloromero, 2019-03-01:
Hi

1.- The two power supply tested are also used with one RPI3, with 4 USB devices connected, without any problem. Anyway I will check again with a new power supply that I just received and provide more feedback.
2.- I have tested different OS. Included the last version of Raspbian.
Also different SD cards.

I have received a new RPI Zero 1.3. I will check this weekend. But with the RPI Zero W the situation is the same, some time after I used the camera I have the panic.

manoloromero, 2019-03-06:
Hi

I made some tests.

Three different power supply.
Different microSD cards.

The situation is:
I get a kernel panic when I access the stream from the PI NoIR camera.
For example. I start RTSP server ( https://github.com/mpromonet/v4l2rtspserver ), all is ok until I connect a plyer to the RTSP port, like VLC. I get a kernel panic.
This also happens using other ways to access the video device, like raspistill, ...

I remove bcm2835-v4l2 driver and use the unofficial driver from UV4L project. Also remove other modules related to bcm.
When I do this I have a segfault, but not a kernel panic. So I have the output.

`Mar  6 20:48:29 e2 kernel: [   66.890864] bcm2835-codec bcm2835-codec: Removing bcm2835-codec
Mar  6 20:48:29 e2 kernel: [   66.897596] bcm2835-codec bcm2835-codec: Removing bcm2835-codec
Mar  6 20:50:15 e2 kernel: [   66.898701] bcm2835-codec bcm2835-codec: Removing bcm2835-codec
Mar  6 20:50:15 e2 kernel: [  172.690461] Unable to handle kernel paging request at virtual address 00c07170
Mar  6 20:50:15 e2 kernel: [  172.696889] pgd = bc7448ad
Mar  6 20:50:15 e2 kernel: [  172.702382] [00c07170] *pgd=00000000
Mar  6 20:50:15 e2 kernel: [  172.708151] Internal error: Oops: 5 [#1] ARM
Mar  6 20:50:15 e2 kernel: [  172.712794] Modules linked in: bnep hci_uart btbcm serdev bluetooth ecdh_generic joydev evdev brcmfmac brcmutil sha256_generic raspberrypi_hwmon snd_bcm2835(C) hwmon snd_pcm snd_timer cfg80211 snd rfkill v4l2_mem2mem videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) uio_pdrv_genirq fixed uio cuse fuse ip_tables x_tables ipv6 [last unloaded: bcm2835_mmal_vchiq]
Mar  6 20:50:15 e2 kernel: [  172.731954] Unable to handle kernel paging request at virtual address dff58040
Mar  6 20:50:15 e2 kernel: [  172.731961] pgd = bc7448ad
Mar  6 20:50:15 e2 kernel: [  172.731967] [dff58040] *pgd=00000000
Mar  6 20:50:15 e2 kernel: [  172.731973] Internal error: Oops: 5 [#2] ARM
Mar  6 20:50:15 e2 kernel: [  172.731975] Modules linked in: bnep hci_uart btbcm serdev bluetooth ecdh_generic joydev evdev brcmfmac brcmutil sha256_generic raspberrypi_hwmon snd_bcm2835(C) hwmon snd_pcm snd_timer cfg80211 snd rfkill v4l2_mem2mem videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) uio_pdrv_genirq fixed uio cuse fuse ip_tables x_tables ipv6 [last unloaded: bcm2835_mmal_vchiq]
Mar  6 20:50:15 e2 kernel: [  172.732039] CPU: 0 PID: 637 Comm: v4l2rtspserver Tainted: G         C        4.19.25+ #1205
Mar  6 20:50:15 e2 kernel: [  172.732041] Hardware name: BCM2835
Mar  6 20:50:15 e2 kernel: [  172.732044] PC is at cfb_imageblit+0x42c/0x980
Mar  6 20:50:15 e2 kernel: [  172.732047] LR is at 0xda800000
Mar  6 20:50:15 e2 kernel: [  172.732051] pc : [<c03f3be8>]    lr : [<da800000>]    psr: 20000193
Mar  6 20:50:15 e2 kernel: [  172.732053] sp : d393b880  ip : 00000000  fp : d393b8d4
Mar  6 20:50:15 e2 kernel: [  172.732056] r10: 00000008  r9 : 000000ff  r8 : 00000000
Mar  6 20:50:15 e2 kernel: [  172.732060] r7 : 00000020  r6 : 00000020  r5 : ff000000  r4 : ffaaaaaa
Mar  6 20:50:15 e2 kernel: [  172.732063] r3 : dff58040  r2 : d703e100  r1 : 00001580  r0 : d7017800
Mar  6 20:50:15 e2 kernel: [  172.732068] Flags: nzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
Mar  6 20:50:15 e2 kernel: [  172.732071] Control: 00c5387d  Table: 13800008  DAC: 00000055
Mar  6 20:50:15 e2 kernel: [  172.732075] Process v4l2rtspserver (pid: 637, stack limit = 0xf78ab353)
Mar  6 20:50:15 e2 kernel: [  172.732078] Stack: (0xd393b880 to 0xd393c000)
Mar  6 20:50:15 e2 kernel: [  172.732082] b880: 00000004 00005600 00001580 dff58040 00000007 00000038 0000000f 00000000
Mar  6 20:50:15 e2 kernel: [  172.732086] b8a0: d703e100 dff58040 f0980540 00000000 d703e107 d72af704 00000000 00000001
Mar  6 20:50:15 e2 kernel: [  172.732092] b8c0: 000000ff d702a400 d393b8e4 d393b8d8 c03f4634 c03f37c8 d393b9b4 d393b8e8
Mar  6 20:50:15 e2 kernel: [  172.732096] b8e0: c03f2230 c03f4628 00000001 dabf3120 d703f963 d7210960 d393b98c d702a508
Mar  6 20:50:15 e2 kernel: [  172.732101] b900: ffffffff d70179a8 ffffffff 00000000 00000200 00000000 00000000 d7017800
Mar  6 20:50:15 e2 kernel: [  172.732105] b920: d72af6f6 00000007 00000007 c03f4628 00000010 00000001 00000000 00000001
Mar  6 20:50:15 e2 kernel: [  172.732110] b940: 00000000 00000010 d393b98c 00000208 00000280 00000038 00000010 00000007
Mar  6 20:50:15 e2 kernel: [  172.732115] b960: 00000000 d393b901 d703e100 c03f2400 00000000 00000000 dac06a80 00000002
Mar  6 20:50:15 e2 kernel: [  172.732119] b980: 00000720 9598c53f d393b9b4 d7017800 d702a400 00000028 d72af6f6 00000007
Mar  6 20:50:15 e2 kernel: [  172.732123] b9a0: c03f1fb4 00000007 d393b9f4 d393b9b8 c03eba48 c03f1fc0 00000028 00000041
Mar  6 20:50:15 e2 kernel: [  172.732128] b9c0: 00000007 00000000 00000049 d72af706 00000720 00000008 d72af5b0 00000700
Mar  6 20:50:15 e2 kernel: [  172.732133] b9e0: 00000041 d72af6f6 d393ba3c d393b9f8 c03edeb8 c03eb960 00000041 d393ba08
Mar  6 20:50:15 e2 kernel: [  172.732138] ba00: 00000007 d702a400 00000028 d72af7c8 d393ba54 d702a400 00000001 00000000
Mar  6 20:50:15 e2 kernel: [  172.732142] ba20: c0b0d1c0 c09d5ae8 0000002f 00000005 d393baa4 d393ba40 c03f0ab8 c03edd60
Mar  6 20:50:15 e2 kernel: [  172.732146] ba40: 000000aa d7017800 d72afefe 00000049 00000000 00000030 00000005 d7017800
Mar  6 20:50:15 e2 kernel: [  172.732152] ba60: 00000000 c0b0f1b8 d702a400 00000000 00000001 c0b0d208 000002f0 00000005
[... cut ...]
Mar  6 20:50:15 e2 kernel: [  172.732467] [<c058073c>] (sys_sendto) from [<c0009000>] (ret_fast_syscall+0x0/0x28)
Mar  6 20:50:15 e2 kernel: [  172.732470] Exception stack(0xd393bfa8 to 0xd393bff0)
Mar  6 20:50:15 e2 kernel: [  172.732474] bfa0:                   be845d64 00000010 00000010 007ff4d0 000005b0 00000000
Mar  6 20:50:15 e2 kernel: [  172.732478] bfc0: be845d64 00000010 bf08020a 00000122 00000000 00000000 b6fc1000 be845d7c
Mar  6 20:50:15 e2 kernel: [  172.732480] bfe0
Mar  6 20:50:15 e2 kernel: [  172.732489] Lost 2 message(s)!
`

manoloromero, 2019-03-18:
I test with a RPI Zero and with a new RPI Zero W and with both is is. So I think that is a hardware issue. Now the system is working perfectly.

vascojdb, 2019-04-07:
Hi guys, any update on this? I was trying to use raspivid on a zero W with a TCP stream and after some minutes my Wi-Fi is gone, I guess it's the same issue.

lategoodbye, 2019-04-07:
@vascojdb Do you see a kernel oops in dmesg?

vascojdb, 2019-04-07:
@lategoodbye i don't think I have seen oops, I saw something related to mail(something), sorry I'm out of home for a week so I can't exactly remember. My Wi-Fi drops completely and only a reboot solves it. I'm still trying to figure out if this is the same problem or the other opened issue (opened 3 years ago)  around here related to Wi-Fi dropping on Pi zero w while streaming video. That is exactly my case and when it happens

pabclsn, 2019-06-07:
Same here RPI Zero W brand new, brand new SD card and power supply, test with a power supply from my pi3 and an other SD card, Camera Module 2.1. I have the H dot near the camera connector too. I'm getting kernel panic within 5 minutes

pabclsn, 2019-06-07:
I downclock the CPU to 950 and no more problem.

IgnacioHR, 2019-09-17:
I'm having the same issue here. Also My Pi zero-W has the H* mark and the pi hangs after less than a minute of "video" reproduction. BUT. I realised the command that freezes the PI is

`raspivid -n -w 1280 -h 720 --bitrate 2000000 --framerate 30 -t 0 -ih --profile baseline --intra 30 --flush -o -| gst-launch-1.0 -v fdsrc do-timestamp=true ! h264parse ! rtph264pay config-interval=1 pt=96 ! udpsink host=127.0.0.1 port=8004`

but the left part of the pipe that reads the camera:
`raspivid -n -w 1280 -h 720 --bitrate 2000000 --framerate 30 -t 0 -ih --profile baseline --intra 30 --flush -o - | cat > /dev/null` 
seems to be working fine for longer. It is when I pipe to gst-launch-1.0 when the pi crashes after some time.

IgnacioHR, 2019-09-17:
After reading this thread I'm having good results after underclocking raspberry pi as follows:

/boot/config.txt
```
arm_freq=600
arm_freq_max=700
arm_freq_min=500

```

CPU is running at 500000 MHz (half the nominal speed) and I'm going to report back to the seller of the card because this is not working as indicated in the specifications

eldad-a, 2019-09-24:
> After reading this thread I'm having good results after underclocking raspberry pi as follows:
>
> /boot/config.txt
>
> ```
> arm_freq=600
> arm_freq_max=700
> arm_freq_min=500
> ```
>
> CPU is running at 500000 MHz [...]

This issue affected me as well, on 3 out of 6 Pi-Zeros W, running [RPi-Cam-Web-Interface](https://elinux.org/RPi-Cam-Web-Interface) (not sure whether this has anything to do with the issue).

Thanks @IgnacioHR , your suggestion seems to resolve the matter for me (based on 4 days run so far).

avanc, 2019-09-30:
I can confirm the very same problem with my Raspberry Pi Zero W (Revision with H*) an dsolved ith with `arm_freq=950`.

It would be interesting to get to know what the differences between the two HW revisions are!

avanc, 2019-10-01:
I was to early: 950MHz reduces the freezes, but they still occur.

@eldad-a Had all six Raspi Zeros the _H*_?

eldad-a, 2019-10-02:
> I was to early: 950MHz reduces the freezes, but they still occur.
> 
> @eldad-a Had all six Raspi Zeros the _H*_?

@avanc 
Sorry this is not working for you :-/
The machines are currently isolated in terms of light, so I cannot check the H*;
I should be able to so in the coming weeks.
Is there other information I can provide?
Anything that's accessible through SSH should work.

NB:  I have it set at `arm_freq=600`
(not `950`)

eldad-a, 2019-10-07:
> I was to early: 950MHz reduces the freezes, but they still occur.
> 
> @eldad-a Had all six Raspi Zeros the _H*_?

@avanc Hi again,
Just wanted to let you know that I have 5 Pi Zero W having the _H*_ mark; I have 4 more which are currently visually inaccessible; will check them in a couple of weeks when the experiment is completed. All 9 were purchased from the same supplier (Adafruit) in the past six months.

Hope this is of any help

avanc, 2019-10-08:
@eldad-a Thanks for the update.

I also got freezes with 600MHz. So it seems the device is really broken. As last try, I will install Rasbian and do some tests.

JamesH65, 2019-10-08:
Hi all, sorry about the delay in replying. Can anyone seeing this issue try a core overclock to see if this changes anything?

In config.txt, add the line  

over_voltage=1

If no help, slightly larger numbers might help.

avanc, 2019-11-03:
@JamesH65 No luck. I tried `over_voltage=1`, but my Raspberry Pi still hangs after some time (approx. 30 min).
So I guess its really a HW issue.

JamesH65, 2019-11-04:
Did you try larger numbers for over_voltage?

avanc, 2019-11-04:
After reading a little more about over_voltage, I'm confused:
According to https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md, the default value for the Raspberry Pi Zero W is already `over_voltage=6`.

Shall I set it to 7? In that case I have to set `force_turbo` and loose warranty.

popcornmix, 2019-11-06:
over_voltage is added to the default voltage (1.35V for turbo mode on Pi0).
You should be able to set it to 1 or 2 to get 1.375V or 1.4V.

maciek01, 2020-02-03:
the same behavior here on stretch RPIZW when streaming via gst-launch-1.0 - boh via wlan0 and ppp0 (modem)

cmptscpeacock, 2020-05-13:
also set my Ri0 W to over_voltage=1 after crashing after 15 mins and now it's been running over an hour no issue.

Using streameye

MertDalbudak, 2020-06-17:
Also having the h* sign near my camera port and also getting kernel panic. I am using a 1.3 Raspberry Pi Camera module. I get crashes after few minutes.

kbharadwaj93, 2020-06-21:
I have the RPi Zero W with the H*. I got it working well (so far) with below added to config file.
`arm_freq=950`
```
force_turbo=1
over_voltage=4
```

wollew, 2021-04-12:
Ok, I had a PI Zero W ("model H") that, after fresh install of Raspbian buster, consistently froze when trying to `apt-get update && apt-upgrade` via WiFi. Tried several power supplies, reflashing images, nothing worked. 

Setting 
```
arm_freq=950
force_turbo=1
over_voltage=4
```

in config.txt fixed the problem for me, too.

Krautmaster, 2021-04-25:
got a similar problem. I just setup my pi zero w with the latest raspian image. Now it gets stuck once streaming between minutes and hours.
`Apr 25 08:56:18 raspberrypi rngd[271]: stats: FIPS 140-2(2001-10-10) Runs: 0
Apr 25 08:56:18 raspberrypi rngd[271]: stats: FIPS 140-2(2001-10-10) Long run: 0
Apr 25 08:56:18 raspberrypi rngd[271]: stats: FIPS 140-2(2001-10-10) Continuous run: 0
Apr 25 08:56:18 raspberrypi rngd[271]: stats: HRNG source speed: (min=119.300; avg=410.195; max=980.238)Kibits/s
Apr 25 08:56:18 raspberrypi rngd[271]: stats: FIPS tests speed: (min=1.314; avg=4.723; max=8.871)Mibits/s
Apr 25 08:56:18 raspberrypi rngd[271]: stats: Lowest ready-buffers level: 2
Apr 25 08:56:18 raspberrypi rngd[271]: stats: Entropy starvations: 0
Apr 25 08:56:18 raspberrypi rngd[271]: stats: Time spent starving for entropy: (min=0; avg=0.000; max=0)us
Apr 25 09:17:01 raspberrypi CRON[1689]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Apr 25 09:40:44 raspberrypi kernel: [53089.270500] ------------[ cut here ]------------
Apr 25 09:40:44 raspberrypi kernel: [53089.270610] WARNING: CPU: 0 PID: 1699 at kernel/time/timer.c:909 add_timer+0x1b0/0x1f4
Apr 25 09:40:44 raspberrypi kernel: [53089.270618] Modules linked in: 8021q garp stp llc brcmfmac brcmutil sha256_generic libsha256 raspberrypi_hwmon snd_soc_rpi_simple_soundcard cfg80211 rfkill bcm2835_codec(C) v4l2_mem2mem bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops snd_soc_bcm2835_i2s videobuf2_v4l2 videobuf2_common regmap_mmio snd_soc_pcm5102a videodev snd_soc_core mc vc_sm_cma(C) snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd uio_pdrv_genirq uio fixed ip_tables x_tables ipv6
Apr 25 09:40:44 raspberrypi kernel: [53089.270831] CPU: 0 PID: 1699 Comm: kworker/u2:0 Tainted: G         C        5.10.17+ #1403
Apr 25 09:40:44 raspberrypi kernel: [53089.270838] Hardware name: BCM2835
Apr 25 09:40:44 raspberrypi kernel: [53089.271380] Workqueue: brcmf_wq/mmc1:0001:1 brcmf_sdio_dataworker [brcmfmac]
Apr 25 09:40:44 raspberrypi kernel: [53089.271400] Backtrace:
Apr 25 09:40:44 raspberrypi kernel: [53089.271443] [<c08693a8>] (dump_backtrace) from [<c0869728>] (show_stack+0x20/0x24)
Apr 25 09:40:44 raspberrypi kernel: [53089.271460]  r7:0000038d r6:00000009 r5:c0080e18 r4:c0a637c4
Apr 25 09:40:44 raspberrypi kernel: [53089.271488] [<c0869708>] (show_stack) from [<c086c940>] (dump_stack+0x28/0x30)
Apr 25 09:40:44 raspberrypi kernel: [53089.271521] [<c086c918>] (dump_stack) from [<c00230e8>] (__warn+0xe8/0x110)
Apr 25 09:40:44 raspberrypi kernel: [53089.271532]  r5:c0080e18 r4:c0a60400
Apr 25 09:40:44 raspberrypi kernel: [53089.271555] [<c0023000>] (__warn) from [<c0869d3c>] (warn_slowpath_fmt+0x68/0xd8)
Apr 25 09:40:44 raspberrypi kernel: [53089.271568]  r7:00000009 r6:c0080e18 r5:0000038d r4:c0a60400
Apr 25 09:40:44 raspberrypi kernel: [53089.271599] [<c0869cd8>] (warn_slowpath_fmt) from [<c0080e18>] (add_timer+0x1b0/0x1f4)
Apr 25 09:40:44 raspberrypi kernel: [53089.271616]  r9:00508ca2 r8:00000001 r7:bf5c57c0 r6:c0ba5028 r5:00000001 r4:c37d4de4
Apr 25 09:40:44 raspberrypi kernel: [53089.271877] [<c0080c68>] (add_timer) from [<bf5b06b4>] (brcmf_sdio_wd_timer+0x70/0x8c [brcmfmac])
Apr 25 09:40:44 raspberrypi kernel: [53089.271897]  r9:00000000 r8:00000001 r7:bf5c57c0 r6:c0ba5028 r5:00000001 r4:c37d4c00
Apr 25 09:40:44 raspberrypi kernel: [53089.272299] [<bf5b0644>] (brcmf_sdio_wd_timer [brcmfmac]) from [<bf5b07d4>] (brcmf_sdio_bus_sleep+0x104/0x1e0 [brcmfmac])
Apr 25 09:40:44 raspberrypi kernel: [53089.272320]  r5:00000000 r4:c37d4c00
Apr 25 09:40:44 raspberrypi kernel: [53089.272704] [<bf5b06d0>] (brcmf_sdio_bus_sleep [brcmfmac]) from [<bf5b1350>] (brcmf_sdio_dataworker+0x1b8/0x2480 [brcmfmac])
Apr 25 09:40:44 raspberrypi kernel: [53089.272725]  r8:c37d4e10 r7:c18eb900 r6:c37d4c00 r5:c3519400 r4:18002000
Apr 25 09:40:44 raspberrypi kernel: [53089.272952] [<bf5b1198>] (brcmf_sdio_dataworker [brcmfmac]) from [<c003dc0c>] (process_one_work+0x208/0x4e0)
Apr 25 09:40:44 raspberrypi kernel: [53089.272972]  r10:c37d4e14 r9:00000000 r8:00000000 r7:c18eb900 r6:00000040 r5:c41d2ae0
Apr 25 09:40:44 raspberrypi kernel: [53089.272980]  r4:c37d4e10
Apr 25 09:40:44 raspberrypi kernel: [53089.272996] [<c003da04>] (process_one_work) from [<c003df18>] (worker_thread+0x34/0x594)
Apr 25 09:40:44 raspberrypi kernel: [53089.273011]  r10:c1045200 r9:00000088 r8:c0c19e40 r7:c1045214 r6:c41d2af4 r5:c1045200
Apr 25 09:40:44 raspberrypi kernel: [53089.273019]  r4:c41d2ae0
Apr 25 09:40:44 raspberrypi kernel: [53089.273037] [<c003dee4>] (worker_thread) from [<c0043848>] (kthread+0x148/0x15c)
Apr 25 09:40:44 raspberrypi kernel: [53089.273053]  r10:cc121e88 r9:c41d2ae0 r8:c003dee4 r7:c5352000 r6:00000000 r5:c53ba7e0
Apr 25 09:40:44 raspberrypi kernel: [53089.273062]  r4:c51ef2c0 r3:00000000
Apr 25 09:40:44 raspberrypi kernel: [53089.273079] [<c0043700>] (kthread) from [<c00082ec>] (ret_from_fork+0x14/0x28)
Apr 25 09:40:44 raspberrypi kernel: [53089.273087] Exception stack(0xc5353fb0 to 0xc5353ff8)
Apr 25 09:40:44 raspberrypi kernel: [53089.273098] 3fa0:                                     00000000 00000000 00000000 00000000
Apr 25 09:40:44 raspberrypi kernel: [53089.273111] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Apr 25 09:40:44 raspberrypi kernel: [53089.273121] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[... cut ...]
Apr 25 09:40:44 raspberrypi kernel: [53089.603960] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Apr 25 09:40:44 raspberrypi kernel: [53089.611908] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Apr 25 09:40:44 raspberrypi kernel: [53089.616485]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0043700
Apr 25 09:40:44 raspberrypi kernel: [53089.621297]  r4:c53ba7e0
Apr 25 09:40:44 raspberrypi kernel: [53089.624324] Code: e1a039a3 e1a03983 e1833002 e5843010 (e5982000)
Apr 25 09:40:44 raspberrypi kernel: [53089.626815] ---[ end trace 43d52116de660cbe ]---`

Krautmaster, 2021-04-25:
after half a day with 
arm_freq=950
force_turbo=1
over_voltage=4

it seems to be more stable. So far no freeze.

julled, 2021-05-29:
i am also having this problem when using mavlink-router with a pi zero. With a CM4 based board the issue doesnt exist.

for me 
```
force_turbo=1
over_voltage=4
```
works to prevent it.

kernel panic:
```
[11648.304339] 8<--- cut here ---
[11648.310466] Unable to handle kernel paging request at virtual address 6000009b
[11648.320179] pgd = 6f2a4fb3
[11648.325266] [6000009b] *pgd=00000000
[11648.331185] Internal error: Oops: 5 [#1] ARM
[11648.337616] Modules linked in: aes_arm aes_generic cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc libaes 8021q garp stp llc brcmfmac brcmutil sha256_generic libsha256 cdc_ac
m cfg80211 rfkill raspberrypi_hwmon bcm2835_codec(C) snd_bcm2835(C) bcm2835_v4l2(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf
2_memops snd_pcm videobuf2_v4l2 videobuf2_common snd_timer snd videodev mc vc_sm_cma(C) fixed uio_pdrv_genirq uio ip_tables x_tables ipv6
[11648.401750] CPU: 0 PID: 4306 Comm: mavlink-routerd Tainted: G         C        5.10.17+ #1403
[11648.417653] Hardware name: BCM2835
[11648.425166] PC is at __xfrm_policy_check+0x18/0x6e4
[11648.432602] LR is at ip_protocol_deliver_rcu+0xe4/0x250
[11648.440328] pc : [<c0802920>]    lr : [<c078ab78>]    psr: 60000013
[11648.450498] sp : c29d98b8  ip : c29d9988  fp : c29d9984
[11648.459274] r10: 0000012c  r9 : c29d9adc  r8 : c0cbcae8
[11648.466835] r7 : 00000000  r6 : c0ba6a44  r5 : c29ed180  r4 : c0cc467c
[11648.475700] r3 : 60000093  r2 : 60000093  r1 : 00000000  r0 : 00000000
[11648.484716] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[11648.494220] Control: 00c5387d  Table: 05310008  DAC: 00000055
[11648.502283] Process mavlink-routerd (pid: 4306, stack limit = 0xb68bea8e)
[11648.511465] Stack: (0xc29d98b8 to 0xc29da000)
[11648.518114] 98a0:                                                       c4250424 00000011
[11648.530677] 98c0: c0cbcae8 00000001 c0ba7294 c4aa0c80 c29d98ec c29d98e0 c06f69f4 c01dc6bc
[11648.543533] 98e0: c29d990c c29d98f0 c0013674 c001442c 00000000 c0ba5028 c0ba6a44 c078ac50
[11648.560674] 9900: c29d9944 c29d9910 c00144a0 c0013658 c00144a0 00000000 c078ac50 c29d99b4
[11648.576166] 9920: c0c252d8 c0028028 c0ba6a44 c078ac50 c0cbcae8 c29d9adc c29d996c c29d9948
[11648.592663] 9940: c00f5f64 c00f72b8 000001ff ffffe000 c0ba6a44 00000000 c0cbcae8 c0cc467c
[11648.609758] 9960: c29ed180 c0ba6a44 00000000 c0cbcae8 c29d9adc 0000012c c29d99b4 c29d9988
[11648.628999] 9980: c078ab78 c0802914 00000a98 fa3f2200 c29d99bc c29ed180 c0ba5028 c0cbcae8
[11648.646278] 99a0: c29ed180 00000000 c29d99c4 c29d99b8 c078ad40 c078aaa0 c29d9a04 c29d99c8
[11648.665975] 99c0: c078adcc c078acf0 c005de3c c29d9a8c c0c252d8 c005ddb4 c29d9a30 c078a384
[11648.680787] 99e0: 00000001 fa3f2200 c29ed180 c104a000 c0cbcae8 c29ed180 c29d9a24 c29d9a08
[11648.700955] 9a00: c078a3a0 c078ad58 c0cbcae8 c0ba5028 c104a000 c29ed180 c29d9a64 c29d9a28
[11648.722792] 9a20: c078af14 c078a310 00000000 c0ba5028 c0cd6f68 fa3f2200 c29d9a7c c0ba5028
[11648.741906] 9a40: c104a000 fa3f2200 c0ba5028 c078ae58 c0cd6f68 00000040 c29d9a8c c29d9a68
[11648.762845] 9a60: c071293c c078ae64 00000000 c29ed180 c0ba7328 fa3f2200 c0cd6fb0 00000000
[11648.781717] 9a80: c29d9aa4 c29d9a90 c071299c c07128e4 c0cd6fb0 00000000 c29d9acc c29d9aa8
[11648.802564] 9aa0: c0712b18 c0712988 c0cd6fb0 00000001 00000000 c0cd6f60 00000040 c29d9adc
[11648.818221] 9ac0: c29d9b1c c29d9ad0 c07143e8 c0712ab8 c0c252d8 001150e8 00000000 c29d9adc
[11648.838507] 9ae0: c29d9adc c29d9ae4 c29d9ae4 fa3f2200 c00f5f64 c0cee760 00000004 00000003
[11648.858342] 9b00: ffffe000 0000003a 00000101 00000008 c29d9b6c c29d9b20 c00094b0 c0714294
[11648.876956] 9b20: c29d9b64 c29d9b30 c00144a0 00400100 00000000 001150e7 0000000a 0000000c
[11648.898024] 9b40: c29d9ba0 60000093 ffffe000 00000000 c341c878 0000003a 00000010 c0ba5028
[11648.918352] 9b60: c29d9b84 c29d9b70 c0027f70 c00093a4 000001ff ffffe000 c29d9b9c c29d9b88
[11648.938737] 9b80: c0028084 c0027f20 c29ed180 00000000 c29d9be4 c29d9ba0 c078e288 c0027fb4
[11648.955848] 9ba0: c29d9bcc c29d9bb0 c104a160 c104a000 0100007f fa3f2200 c0ba5028 c29ed180
[11648.974883] 9bc0: c4bd0600 c0cbcae8 c0cbcae8 0000ffff 00000000 c104a000 c29d9c24 c29d9be8
[11648.993073] 9be0: c078f3c8 c078e0ec c0ba5028 c0790ecc 0ca8a7a4 fa3f2200 c29d9c2c c4bd0600
[11649.011799] 9c00: c4bd0600 c29ed180 00000000 c0cbcae8 0ca8a7a4 00000000 c29d9c44 c29d9c28
[11649.032668] 9c20: c078f56c c078f2f8 c29ed180 c0cbcae8 c0ba5028 c4bd0600 c29d9c9c c29d9c48
[11649.053062] 9c40: c0790ef4 c078f544 c29d9c9c 00000000 c0790480 c078dacc c29ed180 c29d9d80
[11649.073067] 9c60: c29d9cac c29d9c70 c0791770 fa3f2200 c2867c00 c29ed180 c0cbcae8 c4bd0600
[11649.090988] 9c80: c4250424 c29d9db0 00000011 c4bd076c c29d9cbc c29d9ca0 c07905b8 c0790e44
[11649.109726] 9ca0: c29ed180 c0cbcae8 c4bd0600 c4250424 c29d9cd4 c29d9cc0 c0791870 c079057c
[11649.127194] 9cc0: c29ed180 00000018 c29d9d04 c29d9cd8 c07c2abc c0791858 fa3f2200 c4bd0600
[11649.145999] 9ce0: c29d9eb0 c29d9db0 00000018 0100007f c29d9d80 c0ba5028 c29d9e5c c29d9d08
[11649.166548] 9d00: c07c458c c07c2990 00000018 00000008 c29d9d58 c29d9d54 c29d9d80 00000040
[11649.187674] 9d20: c29d9d54 c29d9d30 c29d9d54 c0cbcae8 000036eb 00000000 c29d9d58 00000000
[11649.207299] 9d40: c078ddcc 00000011 00000010 00000000 c0013674 00000000 00000000 00000000
[... 62 lines cut ...]
[11650.225040] [<c079184c>] (ip_send_skb) from [<c07c2abc>] (udp_send_skb+0x138/0x368)
[11650.239692]  r5:00000018 r4:c29ed180
[... cut ...]
[11650.439575] 9fa0:                   011f3f24 00000010 00000004 011f9008 00000010 00000000
[11650.458050] 9fc0: 011f3f24 00000010 befc0b88 00000122 00000001 00000001 0004c6a0 00000001
[11650.475550] 9fe0: 00000000 befc0ad8 00019dd8 b6d8f764
[11650.486748] Code: e24cb004 e24dd0a4 e52de004 e8bd4000 (e592c008) 
[11650.498733] ---[ end trace 61df269a54441ea3 ]---
[11650.507285] Kernel panic - not syncing: Fatal exception in interrupt
[11650.519817] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---

```

Krautmaster, 2021-05-29:
Problem is that most including ne had it without OC at all. Might be a issue with the Turbo

idarek, 2021-12-29:
> After reading this thread I'm having good results after underclocking raspberry pi as follows:
>
> /boot/config.txt
>
> ```
> arm_freq=600
> arm_freq_max=700
> arm_freq_min=500
> ```
>
> CPU is running at 500000 MHz [...]

This solution solved my problem with crashing Pi Zero 2 W