raspberrypi/linux #2560: eth0: hw csum failure - with traffic control

Ogekuri, 2018-05-21:
Hi,
I'm working to implement an ingress traffic control trought the IFB interface.

When I'm trying to redirect the packets from eht0 to ibf0 I get an kernel error.

Here the steps to reproduce the issue:

// load module
`modprobe ifb`

// bring up the interface
`ip link set ifb0 up`

// add ingress queue
`tc qdisc add dev eth0 handle ffff: ingress`

// create a simple class on ifb interface
```
tc qdisc add dev ifb0 root handle 1: htb default 10 r2q 50
tc class add dev ifb0 parent 1: classid 1:1 htb rate 50mbit
tc class add dev ifb0 parent 1:1 classid 1:10 htb rate 50mbit
```

// then try to redirect the traffic (this command get the error)
`tc filter add dev eth0 parent ffff: protocol ip prio 10 u32 match u32 0 0 flowid 1:10 action mirred egress redirect dev ifb0`

Kernel error:
`kernel: Mirror/redirect action on
kernel: eth0: hw csum failure
kernel: CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.9.35-v7+ #1014
kernel: Hardware name: BCM2835

kernel: [<8010fb3c>] (unwind_backtrace) from [<8010c058>] (show_stack+0x20/0x24)
kernel: [<8010c058>] (show_stack) from [<80455880>] (dump_stack+0xd4/0x118)
kernel: [<80455880>] (dump_stack) from [<80627628>] (netdev_rx_csum_fault+0x44/0x48)
kernel: [<80627628>] (netdev_rx_csum_fault) from [<8061a2b8>] (__skb_checksum_complete+0xb4/0xb8)
kernel: [<8061a2b8>] (__skb_checksum_complete) from [<80690384>] (tcp_v4_rcv+0x4fc/0xc10)
kernel: [<80690384>] (tcp_v4_rcv) from [<80668b5c>] (ip_local_deliver_finish+0xe4/0x330)
kernel: [<80668b5c>] (ip_local_deliver_finish) from [<806693c0>] (ip_local_deliver+0x54/0xe4)
kernel: [<806693c0>] (ip_local_deliver) from [<80668fdc>] (ip_rcv_finish+0x234/0x48c)
kernel: [<80668fdc>] (ip_rcv_finish) from [<8066976c>] (ip_rcv+0x31c/0x55c)
kernel: [<8066976c>] (ip_rcv) from [<80625154>] (__netif_receive_skb_core+0x304/0xbc0)
kernel: [<80625154>] (__netif_receive_skb_core) from [<80627978>] (__netif_receive_skb+0x20/0x7c)
kernel: [<80627978>] (__netif_receive_skb) from [<80627a00>] (netif_receive_skb_internal+0x2c/0xa4)
kernel: [<80627a00>] (netif_receive_skb_internal) from [<80627a9c>] (netif_receive_skb+0x24/0x98)
kernel: [<80627a9c>] (netif_receive_skb) from [<7f0e6514>] (ifb_ri_tasklet+0xf4/0x29c [ifb])
kernel: [<7f0e6514>] (ifb_ri_tasklet [ifb]) from [<80123220>] (tasklet_action+0x74/0x10c)
kernel: [<80123220>] (tasklet_action) from [<8010169c>] (__do_softirq+0x18c/0x3cc)
kernel: [<8010169c>] (__do_softirq) from [<80122ca8>] (irq_exit+0x10c/0x168)
kernel: [<80122ca8>] (irq_exit) from [<801735f4>] (__handle_domain_irq+0x70/0xc4)
kernel: [<801735f4>] (__handle_domain_irq) from [<8010150c>] (bcm2836_arm_irqchip_handle_irq+0xa8/0xac)
kernel: [<8010150c>] (bcm2836_arm_irqchip_handle_irq) from [<8071967c>] (__irq_svc+0x5c/0x7c)
kernel: Exception stack(0x80c01f18 to 0x80c01f60)
kernel: 1f00:                                                       00000000 b9f65470
kernel: 1f20: 00000000 00000000 80c00000 80c0312c 00000001 80c03198 80c15f06 80c15f06
kernel: 1f40: ba7ffa80 80c01f74 80c040a4 80c01f68 80108a54 80108a58 60000013 ffffffff
kernel: [<8071967c>] (__irq_svc) from [<80108a58>] (arch_cpu_idle+0x34/0x4c)
kernel: [<80108a58>] (arch_cpu_idle) from [<80718dc4>] (default_idle_call+0x34/0x48)
kernel: [<80718dc4>] (default_idle_call) from [<80161d2c>] (cpu_startup_entry+0xe4/0x160)
kernel: [<80161d2c>] (cpu_startup_entry) from [<80713650>] (rest_init+0x84/0x88)
kernel: [<80713650>] (rest_init) from [<80b00d28>] (start_kernel+0x3b4/0x3c0)
`

pelwell, 2018-05-21:
1. Which hardware platform is this?

2. Which distribution and version is it running?

3. Have you tried any other platforms or OS versions and, if so, what were the results?

I would recommend running `sudo rpi-update` to get the latest 4.14 kernel because there have been a number of networking changes since the 4.9.35 kernel.

Ogekuri, 2018-05-21:
Hi pelwell,
I was on a Raspberry Pi 3 with Debian 8 and a Linux 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux.

As you have suggested I've upgraded to **Linux 4.14.41-v7+ #1113** SMP Thu May 17 16:29:48 BST 2018 armv7l GNU/Linux

Same error:
```
May 21 08:20:06 envision1 kernel: Mirror/redirect action on
May 21 08:20:07 envision1 kernel: eth0: hw csum failure
May 21 08:20:07 envision1 kernel: CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.14.41-v7+ #1113
May 21 08:20:07 envision1 kernel: Hardware name: BCM2835
...

```

JamesH65, 2018-06-26:
Confirmed that this also fails on 4.14.50 on a 3B.

JamesH65, 2018-06-26:
I think this might be this 

https://patchwork.ozlabs.org/patch/920116/

I'll try the proposed patch to see if it fixes anything.

JamesH65, 2018-06-26:
@pelwell @popcornmix OK, so pulling in that patch does make the error go away. Whether the system works as planned I have no idea, but I do think we should pull this patch in. It's already merged to mainline.

pelwell, 2018-06-26:
Cherry-picked and pushed to rpi-4.14.y.

JamesH65, 2018-06-26:
@Ogekuri Would it be possible for you to test this (you can get it using rpi-update, in a couple of days, although best not on a production system, or build your own kernel from top of tree 4.14.y!), and if you are happy things are fixed, close the issue?. Thanks.

Ogekuri, 2018-06-29:
Yes, I'll test it over the weekend with rpi-update.

Ogekuri, 2018-06-30:
I've upgrade all Pi3 to this version:
```
uname -a
**Linux *** 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l GNU/Linux**
/opt/vc/bin/vcgencmd version
**Jun 29 2018 13:12:24
Copyright (c) 2012 Broadcom
version 97765855d9ae607d0e678171cf9d83b218f237dd (clean) (release)**
```

This issue is resolved.
Thanks!
_-- 
Oge_

closed 2018-06-30 by Ogekuri