raspberrypi/linux #2561: Interrupt handling problem between rpi3 & wm8731

dinoHee, 2018-05-21:
Hi!! I am a student who is going to make wm8731 ~ rpi3 i2s device driver.

And there are some problems that I have now. I want to solve these...

1. How can I handle the interrupt??

I think the timing is when transmitter buffer is full. Am I wrong?

Plus, when i set the interrupt number. Can I set the number in ramdom??

My kernel version is 4.14.30-v7+

2. Do I have to have interrupt handling??

3. Without interrupt handling, I tried my driver can send the buffer to codec when read audio file.
And I check MOSI, MISO by checking LED, I can see that the information can go right way to codec.(By light in led.)
But the sound is not coming.
What is the problem??

Sorry for my english, and please let me know..:<

amovitz, 2018-05-30:
You will get a better solution from the Raspberry Pi forums if you're having trouble setting it up. [Issue 1302 may answer some of your questions](https://github.com/raspberrypi/linux/issues/1302), but this issue tracker is mainly for problems in the Linux Kernel itself. I hope this helps.

JamesH65, 2018-06-29:
Closing as this issue is not kernel related.

closed 2018-06-29 by JamesH65