raspberrypi/linux #2563: /dev/tty1 stdin is spammed with 'x' character when dtparam=audio=on and BCM pin 18 is in OUT mode

juj, 2018-05-22:
STR:

Connect a Pi 3B or a Pi Zero W to HDMI output, and have a USB keyboard attached, and `/boot/config.txt` with `dtparam=audio=on`, then after booting to terminal, type

`gpio -g mode 18 OUT`

Observed: the terminal stdin buffer starts to receive a stream of `x` characters. That is, not terminal stdout, but stdin, as if one had typed those input characters with keyboard, since pressing enter will then submit these `x` characters as an input command.

In Freenode/#raspbian IRC channel it was mentioned that the pin in question is controlled by audio driver for its PWM operation. That's fair, though I'm having trouble fathoming how an incorrect pin config would result in default `/dev/tty1` stdin starting to receive a stream of input characters? Pin 18 should not be related to UART serial input in any way?

(In this test, nothing was physically connected to pin 18. Tested on a Pi 3B, and a Zero W)

pelwell, 2018-05-22:
Which GPIO numbering scheme is this? You can confirm with `raspi-gpio get` which uses the (one true) Broadcom numbering.

juj, 2018-05-22:
Ops, very sorry, now I see this is caused by https://github.com/adafruit/Adafruit-Retrogame/blob/master/configs/retrogame.cfg#L20 . I thought I was testing on the stock Raspbian, but instead had one of the Adafruit based images on the card. I have too many SD cards lying on my table!

Apologies for the noise, closing this as invalid. I don't know why Adafruit's `retrogame` app behaves different compared to when `dtparam=audio=on` vs `off`, but doesn't matter any more - moved to a proper stock image now.

juj, 2018-05-22:
Thanks for the fast troubleshooting help though, as always :)

closed 2018-05-22 by juj