raspberrypi/linux #2564: Issue with connected HFS+ USB Drive

Eyeless77, 2018-05-23:
I've got a RPi running some docker containers and I've got an issue with one of them which serves Time Machine backups. There is an external Seagate HDD with external power supply which is formatted to HFS+ and connected via USB to the raspberry and then mounted to container. When I start a Time Machine backup from my mac there are lots of errors in console. I tried different images from [odarriba](https://github.com/odarriba/docker-timemachine) and [mbentley](https://github.com/mbentley/docker-timemachine/blob/master/Dockerfile) with different versions of netatalk (tested with 3.1.10 and 3.1.11). Also I tried to downgrade to 4.9 kernel, but errors are the same. After I get this error I cannot gracefully reboot raspberry, it freezes on reboot, I have to plug the power off manually or use sysrq magic commands. After that I plug HDD to my mac and fix the drive with fsck manually, because this issue damages the backup partition. Then I make sure that fsck.hfsplus on raspberry returns no errors and try to make one more backup but get the same error and that's annoying. Could anyone tell what is the problem?

`pi@pi-home:~ $ uname -a
Linux pi-home 4.14.42-v7+ #1114 SMP Mon May 21 16:39:21 BST 2018 armv7l GNU/Linux`

`pi@pi-home:~ $ /opt/vc/bin/vcgencmd version
May 18 2018 18:05:14
Copyright (c) 2012 Broadcom
version c75d403819718d12049aa2edd2009356a6cb592f (clean) (release)`

```
May 23 22:29:32 pi-home kernel: [  633.005077] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.005293] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.005494] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.005693] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.005891] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.006172] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.006370] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.006564] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.006758] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.006949] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.007148] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.007522] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.007713] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.007900] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.008087] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.008404] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.008591] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.008790] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.008973] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.009166] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.009345] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.009520] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.009694] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.009867] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.010038] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.010211] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.010381] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.010551] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.010719] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.010885] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.011051] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.011218] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.011382] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.011545] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.011706] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.011865] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.012022] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.012181] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.012339] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.012495] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.012649] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.012801] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.012954] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.013104] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.013253] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.013403] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.013559] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.013718] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.013865] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.014007] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.014149] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.014290] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.014430] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.014570] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.014708] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.014843] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.014977] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.015115] hfsplus: trying to free free bnode 0(1)
May 23 22:29:32 pi-home kernel: [  633.015249] hfsplus: trying to free free bnode 0(1)
[... 68 lines cut ...]
May 23 22:29:32 pi-home kernel: [  633.020089] dec0: 00000000 00000000 00000000 00000000 b97abce8 00000000 b97abd58 b6d65800
May 23 22:29:32 pi-home kernel: [  633.020107] dee0: 00000000 b97abce8 a2c7df14 a2c7def8 7f3c4288 7f3c6394 8bd9a840 b97abce8
May 23 22:29:32 pi-home kernel: [  633.020124] df00: a2fac810 940514c8 a2c7df54 a2c7df18 8028bc40 7f3c41f4 00000000 00000000
May 23 22:29:32 pi-home kernel: [  633.020140] df20: 8028bde4 8bd9a848 b9195028 00000000 b9195028 b9194b00 b9195044 80c97e04
May 23 22:29:32 pi-home kernel: [  633.020157] df40: 8bd9a840 00000080 a2c7df64 a2c7df58 8028bdfc 8028bbb0 a2c7df8c a2c7df68
May 23 22:29:32 pi-home kernel: [  633.020174] df60: 8013bbfc 8028bdf0 a2c7c000 a2c7c010 80108204 a2c7dfb0 80108204 a2c7c000
May 23 22:29:32 pi-home kernel: [  633.020191] df80: a2c7dfac a2c7df90 8010b810 8013bb4c 7eed24b0 00000001 00000000 00000006
May 23 22:29:32 pi-home kernel: [  633.020210] dfa0: 00000000 a2c7dfb0 80108094 8010b750 00000000 00000001 00000000 0000000b
May 23 22:29:32 pi-home kernel: [  633.020226] dfc0: 7eed24b0 00000001 00000000 00000006 00000001 00000000 00000001 7eed24b0
May 23 22:29:32 pi-home kernel: [  633.020242] dfe0: 00000000 7eed2434 76eb7ff8 768ecbdc 40000010 0000000b 00000000 00000000
May 23 22:29:32 pi-home kernel: [  633.020317] [<807834f0>] (memcpy) from [<7f3c8788>] (hfsplus_bnode_write+0xac/0x128 [hfsplus])
May 23 22:29:32 pi-home kernel: [  633.020429] [<7f3c8788>] (hfsplus_bnode_write [hfsplus]) from [<7f3c8838>] (hfsplus_bnode_write_u16+0x34/0x3c [hfsplus])
May 23 22:29:32 pi-home kernel: [  633.020515] [<7f3c8838>] (hfsplus_bnode_write_u16 [hfsplus]) from [<7f3cadf0>] (hfsplus_brec_remove+0x128/0x178 [hfsplus])
May 23 22:29:32 pi-home kernel: [  633.020599] [<7f3cadf0>] (hfsplus_brec_remove [hfsplus]) from [<7f3cd448>] (__hfsplus_delete_attr+0xa4/0xec [hfsplus])
[... cut ...]
May 23 22:29:32 pi-home kernel: [  633.020963] [<8028bdfc>] (____fput) from [<8013bbfc>] (task_work_run+0xbc/0xe0)
May 23 22:29:32 pi-home kernel: [  633.020988] [<8013bbfc>] (task_work_run) from [<8010b810>] (do_work_pending+0xcc/0xd0)
May 23 22:29:32 pi-home kernel: [  633.021012] [<8010b810>] (do_work_pending) from [<80108094>] (slow_work_pending+0xc/0x20)
May 23 22:29:32 pi-home kernel: [  633.021029] Code: 14d13001 24d14001 24d1c001 14c03001 (24c04001)
May 23 22:29:32 pi-home kernel: [  633.021046] ---[ end trace 0a803365bebd3f0f ]---
```

pelwell, 2018-05-23:
Through the advanced technique known as "googling the error message" it seems the problem appeared as early as the 4.8 kernel (https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1646565) and affects multiple/all platforms (https://lkml.org/lkml/2017/1/9/808, https://lime-technology.com/forums/topic/67183-help-unraid-frozen-after-a-rsync-that-segfaulted/).

There's no way we can spend the time (if we even have enough expertise) to debug a problem in a generic upstream module like hfsplus. I was hoping I could at least give you the email address of one of the developers responsible, but the Linux MAINTAINERS file has this to say:
```
HFSPLUS FILESYSTEM
L:	linux-fsdevel@vger.kernel.org
S:	Orphan
F:	Documentation/filesystems/hfsplus.txt
F:	fs/hfsplus/
```
which would explain the lack of progress.

JamesH65, 2018-06-25:
I'm inclined to close within 30 days unless further interactions are posted.  If you wish this issue to remain open, please add a comment, but be aware there's no way we will be doing any work on this. A closed issue may be reopened if requested.

closed 2018-08-31 by JamesH65