raspberrypi/linux #2567: no sdcard inserted causes 10% kworker cpu usage (usb boot)

fanoush, 2018-05-28:
with usb boot one can have no sd card inserted at all. It looks like this causes  kworker thread to eat %10 cpu 
more info here
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=211912#p1307781
workaround is to boot with blank sd card in slot

pelwell, 2018-05-28:
I'm curious to see what effect setting the `non-removable` flag on the sdhost interface would do with no card inserted. Assuming it had the desired effect, it could be enabled with an overlay, but the effect would persist until the reboot after the removal of the overlay. I'll report back.

pelwell, 2018-05-29:
`non-removable` works as expected - if there is a card present at boot time then it should be detected and used, otherwise the interface is effectively disabled.

I've pushed an update to the `sdtweak` overlay that adds two new parameters - `poll_once` and `enable`. Adding `dtoverlay=sdtweak,poll_once` to config.txt adds the `non-removable` property, which allows a card to be used if present from boot. `dtoverlay=sdtweak,enable=off` completely disables the SD card interface.

fanoush, 2018-05-29:
Thanks, sounds great, however do you know why the polling is so cpu intensive and runs continuously? Wouldn't e.g. check once per 500ms be enough to detect card insert? Why the kworker is not mostly sleeping even when doing the check?

pelwell, 2018-05-29:
Without reading the code, no, I don't know why it is so processor intensive, but apart from the interface driver this is all unmodified upstream code. It's possible that the combination of low interface speed (SD cards start up at around 100KHz) and small transfers means we're using PIO rather than DMA (i.e. the ARMs are polling the interface rather than letting the DMA controller do it). The PIO->DMA handover was tuned for performance at regular operating clock rates (50MHz+) so there may be room for improvement, but I'm not planning on optimising the absent card case.

fanoush, 2018-05-29:
Thanks. And BTW is it possible to call the scan on demand from user space when poll_once is set? some ioctl perhaps or triggering some sysfs file?  will check

fanoush, 2018-05-29:
And btw another idea, why not to set as non-removable by default even when booting from sd card (i.e the typical case)? One will hardly remove the card when system is booted from it, won't this also save some I/O and polling like this? Because if slot is removable now, it is probably polling also to detect card removal now (which is quite unlikely to happen)?

pelwell, 2018-05-29:
I don't know a way of triggering a scan - let me know if you find one.

I'd rather not change the current polling behaviour - it may be useful in the case that a card is slow to respond for some reason, and I can think of use cases where the ability to swap cards is an asset. Once the card is detected any subsequent polling will happen at the usual clock speed and thus have negligible overhead. Another problem with the non-removable property is that it isn't possible to unset it from an overlay because overlays are unable to delete properties.

fanoush, 2018-05-29:
So far  I found this may be indeed tricky but one way could be to unbind/bind driver https://unix.stackexchange.com/questions/33285/how-can-one-force-a-re-scan-of-an-sdio-bus-from-linux-user-space . Will  check how it works at home. Also will try to mess with custom kernel module and call mmc_detect_change or play with mmc host variables when such module is inserted.

Looks like specific device drivers has various ways to trigger it
https://elixir.bootlin.com/linux/v4.14/ident/mmc_detect_change
 e.g. omap driver has it triggered when cover switch changes
https://elixir.bootlin.com/linux/v4.14/source/drivers/mmc/host/omap_hsmmc.c#L1301
so in theory we may have some custom trigger to rescan card in pi driver too. Just ideas.

As for  negligible overhead, well it may needlessly occupy the card with constant flow of MMC commands keeping it busy and powered up when it could  be idle. Also I wonder how this polling behaves when the card is already busy doing heavy reading/writing, is it queued, skipped or intermixed with regular i/o slowing things down a bit. Anyway changing defaults can be risky like you say, but when this sdtweak you just added is available it may make sense to try to set in not just when booting from usb but also when booting from sd. Will definitely try this in future to see if it makes any difference.

pelwell, 2018-05-29:
> So far I found this may be indeed tricky but one way could be to unbind/bind driver 

Ah yes, I have come across the unbind mechanism before when trying to trigger some of the kernels built-in card testing features.

> btw do you have the arch/arm/boot/dts/overlays/sdtweak-overlay.dts compiled so it can be used with your changes right now?

Yes - here you go: https://drive.google.com/file/d/1JOwJpvWNYbQI4AwwzlQPCJMNehuCFQoy/view?usp=sharing

fanoush, 2018-05-29:
Thanks a lot. it works. First the overlay - copying your file to /boot/overlays and adding line "dtoverlay=sdtweak,poll_once" to /boot/config.txt solved the kworker issue with polling the card. So far so good :-)

However the expected result is also that there is no rescan or detect of changes. Which also means that if you boot with card and later remove it the mmcblk0 is still there including all partitions in /proc/partitions and /dev/mmcblk0pX and any access to it produces nasty mmc errors in kernel log. Solution to this is unbinding/rebining  the driver which works as follows
```
root@raspberrypi:~# ls /sys/bus/platform/drivers/sdhost-bcm2835/
3f202000.mmc  bind  uevent  unbind
root@raspberrypi:~# echo -n "3f202000.mmc" >/sys/bus/platform/drivers/sdhost-bcm2835/unbind
root@raspberrypi:~# ls /sys/bus/platform/drivers/sdhost-bcm2835/
bind  uevent  unbind
```
now I see mmc0: card 21cb removed in kernel log
rebinding detects the card again if it is present in the slot
```
root@raspberrypi:~# echo -n "3f202000.mmc" >/sys/bus/platform/drivers/sdhost-bcm2835/bind 
root@raspberrypi:~# ls /sys/bus/platform/drivers/sdhost-bcm2835/
3f202000.mmc  bind  uevent  unbind
```
now I see
```
[  423.404655] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
[  423.471878] mmc0: host does not support reading read-only switch, assuming write-enable
[  423.474553] mmc0: new high speed SDHC card at address 21cb
[  423.477247] mmcblk0: mmc0:21cb APPSD 3.75 GiB
```
However when tried multiple times I also started to see some issue with DMA, so maybe not everything is freed properly when unbinding the driver?
```
[ 1578.852936] mmc0: unable to initialise DMA channel. Falling back to PIO
[ 1578.930326] mmc0: sdhost-bcm2835 loaded - DMA disabled
[ 1578.997265] mmc0: host does not support reading read-only switch, assuming write-enable
[ 1578.999812] mmc0: new high speed SDHC card at address 21cb
[ 1579.004406] mmcblk0: mmc0:21cb APPSD 3.75 GiB
```
So to sum it up - to stop polling for card your tweak works and is set and forget solution. If one does not use sdcard at all and just boot from usb it is the easiest way.  Also without your tweak another solution for the kworker issue is unbinding the driver at runtime. This is also solution if one needs to rescan for changes later.

pelwell, 2018-06-14:
Can we close this issue now?

fanoush, 2018-06-15:
yes, thank you, closing

closed 2018-06-15 by fanoush

mkreisl, 2018-06-28:
root@kmxbilr2 /boot/overlays # dtparam sd_poll_once=on
* Unknown parameter 'sd_poll_once'

pelwell, 2018-06-28:
It seems you didn't read my reply carefully enough - try `dtoverlay=sdtweak,poll_once`.

pelwell, 2018-06-28:
[Response to now edited comment deleted]

The updated overlay has been in the rpi-update releases since the start of the month - are you up-to-date? But I'm not sure that applying it at run time would have the desired effect - the unbinding method is a better bet.

mkreisl, 2018-06-28:
I tried to  set `poll_once=on` after booting, because when booting sd-card less it seems not to work as expected via config.txt, see issue [RPi3, Rpi3B+ network boot: config.txt ignored](/https://github.com/raspberrypi/firmware/issues/1012)