raspberrypi/linux #2571: Updating on IPv6

pompel82, 2018-06-02:
I'm having problems updating as archive.raspberrypi.org don't respond over IPv6. I can do e.g. sudo apt-get -o Acquire::ForceIPv4=true update when updating manually but when the update is done in a package such as openhabian I can't control it and get stuck.

0% [Connecting to mirrordirector.raspbian.org (2a00:1098:0:80:1000:75:0:3)] [Connecting to archive.raspberrypi.org (2a00:1098:0:80:1000:13:0:5)]

The AAAA records in DNS should be removed if IPv6 is not supported.

smallcreek, 2018-06-02:
Seems to work for me:

    $ curl -sI [2a00:1098:0:80:1000:75:0:3] | head -1
    HTTP/1.1 200 OK
    $ curl -sI [2a00:1098:0:80:1000:13:0:5] | head -1
    HTTP/1.0 302 Found

You may want to check your IPv6 connectivity.

pelwell, 2018-06-02:
Thanks for assisting.

JamesH65, 2018-06-26:
Tested here as well, works fine.

Closing this issue as questions answered/issue resolved.

closed 2018-06-26 by JamesH65