raspberrypi/linux #2574: mouse and keyboard was not an MTP device

loreberti89, 2018-06-06:
Hello I have a problem with keyboard and mouse with my raspberrypi3.

kernel version: **Kernel 4.14.44-v7+**

When I put the usb In /var/log/syslog I receive this error:

`mtp-probe: bus: 1, device: 6 was not an MTP device`

In **lsusb** I see correctly the device usb.

The problem is after boot because before the see the desktop I can press shift for enter in recovery mode and here mouse and keyboard work perfectly.

XECDesign, 2018-06-06:
The MTP message is not an error. The full log would be helpful.

loreberti89, 2018-06-06:
Hello, thanks, this is the log if I connect usb mouse
 
Jun  6 23:44:41 RaspbertiPy kernel: [   73.786958] usb 1-1.2: new low-speed USB device number 5 using dwc_otg
Jun  6 23:44:41 RaspbertiPy kernel: [   73.921750] usb 1-1.2: New USB device found, idVendor=093a, idProduct=2510
Jun  6 23:44:41 RaspbertiPy kernel: [   73.921765] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jun  6 23:44:41 RaspbertiPy kernel: [   73.921773] usb 1-1.2: Product: USB OPTICAL MOUSE
Jun  6 23:44:41 RaspbertiPy kernel: [   73.921781] usb 1-1.2: Manufacturer: PIXART
Jun  6 23:44:41 RaspbertiPy kernel: [   73.928565] input: PIXART USB OPTICAL MOUSE as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:093A:2510.0002/input/input1
Jun  6 23:44:41 RaspbertiPy kernel: [   73.929267] hid-generic 0003:093A:2510.0002: input,hidraw0: USB HID v1.11 Mouse [PIXART USB OPTICAL MOUSE] on usb-3f980000.usb-1.2/input0
Jun  6 23:44:41 RaspbertiPy mtp-probe: checking bus 1, device 5: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2"
Jun  6 23:44:41 RaspbertiPy mtp-probe: bus: 1, device: 5 was not an MTP device

I have try to make **sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel**

And now I have 4.9.35-v7+ but he problem persist

JamesH65, 2018-06-07:
I've had a quick trawl around the internet for this particular error, and every page that mentions this MTP issue conclude the HW being plugged in is faulty. In your case, it does sound like the mouse is faulty. Can you try a different one?

closed 2018-06-29 by JamesH65