raspberrypi/linux #2579: /bin/sh: 1: scripts/basic/fixdep: Permission denied

dissipator, 2018-06-09:
HOSTCC  scripts/basic/fixdep
/bin/sh: 1: scripts/basic/fixdep: Permission denied
scripts/Makefile.host:102: recipe for target 'scripts/basic/fixdep' failed
make[1]: *** [scripts/basic/fixdep] Error 126
Makefile:463: recipe for target 'scripts_basic' failed
make: *** [scripts_basic] Error 2

pelwell, 2018-06-09:
How did you download the repo? git clone is reliable, but some people find that the zip archive can be corrupted.

dissipator, 2018-06-09:
all try

dissipator, 2018-06-09:
git clone --depth=1 https://github.com/raspberrypi/linux

dissipator, 2018-06-09:
root@raspberrypi:/usr/src/linux-headers-4.14.48-v7+# bash scripts/basic/fixdep
scripts/basic/fixdep: scripts/basic/fixdep: 无法执行二进制文件

dissipator, 2018-06-09:
root@raspberrypi:/usr/src/linux-headers-4.14.48-v7+# ls -l scripts/basic/

-rw-r--r-- 1 root root   743 6月   9 23:46 bin2c.c
-rwxr-xr-x 1 root root 13436 6月  10 00:06 fixdep
-rw-r--r-- 1 root root 11236 6月   9 23:46 fixdep.c
-rw-r--r-- 1 root root   715 6月   9 23:46 Makefile

dissipator, 2018-06-09:
on the centos x86 can make ok。but copy to raspi ， will be same wrong。

pelwell, 2018-06-09:
When you say "copy", did you clone on the Pi or clone on Centos x86?

dissipator, 2018-06-10:
clone on pi

JamesH65, 2018-06-28:
I notice that you are doing your clones as root. That may be causing permission problems, as everything in the clone will then belong to root. 

Do the clone as a standard user, see if that helps.

JamesH65, 2018-06-29:
I've done a quick test, cloned both as a normal user and root. Root simply doesn't work, although I don't get the same error message, but the Pi user works fine.

Please retry cloning as the Pi user, and report back.

Otherwise this issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

JamesH65, 2018-08-31:
Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2018-08-31 by JamesH65

Zacharyprime, 2022-08-01:
I am also having this issue while trying to build the kernel.

I did not run the clone as sudo, although the download was insanely slow even with depth=1

I was not getting this error in the past when I was doing this exact process, the only thing I'm doing differently is building on a flash drive (the SD card on my embedded controller does not have enough space). My coworker however, built using cross-compile and by just using a larger SD card and got the same result.

The most frustrating feeling when working with Linux is getting a permission denied that doesn't go away when you sudo.

cshunt, 2023-02-19:
This fixed the issue for me:
`chmod a+x scripts/basic/fixdep`
where ever the fixdep script is

sannithrao2468, 2023-07-06:
hey @Zacharyprime  , @JamesH65  & @cshunt , I'm facing he same issue while i build a new kernel for my rpi-4 . Can you suggest any solution.
These are the instructions I am following:
https://www.raspberrypi.com/documentation/computers/linux_kernel.html
I'm stuck at this error: 
/F473-1E84/linux$ make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2711_defconfig

/bin/sh: 1: ./scripts/pahole-flags.sh: Permission denied
  HOSTCC  scripts/basic/fixdep
/bin/sh: 1: scripts/basic/fixdep: Permission denied
make[1]: *** [scripts/Makefile.host:110: scripts/basic/fixdep] Error 126
make[1]: *** Deleting file 'scripts/basic/fixdep'
make: *** [Makefile:628: scripts_basic] Error 2

Please help , Thanks in advance

pelwell, 2023-07-06:
I think you are trying to build a kernel in a FAT partition on an external drive. FAT filing systems don't store "executable" permissions for files, leading to the kind of errors you are seeing.

Run this test in your linux tree to confirm:
```
$ echo "echo Success" > foo
$ chmod +x foo
$ ./foo
$ ls -l foo
```

sannithrao2468, 2023-07-11:
Hey @pelwell , Thankyou for helping!  Actually i'm doing all these to reduce the boot time of my rpi-4( to run a gui application). As a part of that, im building a new kernel. Am i in the correct path ? 
By following your solution , im getting this ..
' ' ' 
madhavanenisannith@nhhydl-00271:/media/madhavanenisannith/F473-1E84/linux$ echo "echo Success" > foo
madhavanenisannith@nhhydl-00271:/media/madhavanenisannith/F473-1E84/linux$ chmod +x foo
madhavanenisannith@nhhydl-00271:/media/madhavanenisannith/F473-1E84/linux$ ./foo
bash: ./foo: Permission denied
madhavanenisannith@nhhydl-00271:/media/madhavanenisannith/F473-1E84/linux$ ls -l foo
-rw-r--r-- 1 madhavanenisannith madhavanenisannith 13 Jul 11 13:15 foo
madhavanenisannith@nhhydl-00271:/media/madhavanenisannith/F473-1E84/linux$ make bcm2711_defconfig
  HOSTCC  scripts/basic/fixdep
/bin/sh: 1: scripts/basic/fixdep: Permission denied
make[1]: *** [scripts/Makefile.host:95: scripts/basic/fixdep] Error 126
make[1]: *** Deleting file 'scripts/basic/fixdep'
make: *** [Makefile:535: scripts_basic] Error 2
 ' ' '
Please guide me , and Thanks in advance

pelwell, 2023-07-11:
```
-rw-r--r-- 1 madhavanenisannith madhavanenisannith 13 Jul 11 13:15 foo
```
This shows that you are using an unsuitable filing system (probably FAT32 or exFAT). A good result would have been:
```
-rwxr-xr-x 1 madhavanenisannith madhavanenisannith 13 Jul 11 13:15 foo
```
showing that the file is executable (a permission that you tried to add with `chmod +x foo`).

You are going to need a standard Linux filing system (`ext4` is the obvious one) to build the kernel, either on a dedicated volume or as a partition within a larger device. Don't waste your time looking for a workaround.

sannithrao2468, 2023-07-13:
Thank you @pelwell ! Can you suggest me a way to reduce the boot time of rpi-4 to (10-15) sec . I have tried suspending many services and other ways , but nothing's working. 
I'm working on rpi-4(32 bit OS) for loading a Python-Qt application (on-boot). I have tried using build root but unable to move forward with it. Can you suggest reading any documentation or references? . Thanks in advance