raspberrypi/linux #2580: Kernel panic (vxlan)

brb, 2018-06-11:
The latest kernel https://github.com/raspberrypi/linux/tree/raspberrypi-kernel_1.20180417-1 used in Raspbian includes https://github.com/coreos/linux/commit/4699beb7712768c38d009301a2d20f1f1e567294#diff-4f541554c5f8f378effc907c8f0c9115 which is known to cause a kernel panic due to NULL pointer deref when the vxlan driver is used. The fix is available: https://github.com/torvalds/linux/commit/f15ca723c1ebe6c1a06bc95fda6b62cd87b44559#diff-4f541554c5f8f378effc907c8f0c9115.

More context: https://github.com/weaveworks/weave/issues/3314 and https://github.com/coreos/bugs/issues/2382.

alexellis, 2018-06-20:
Thanks for reporting.. vxlan is also important for Docker Swarm to operate.

JamesH65, 2018-06-25:
@pelwell @popcornmix Quick read through the context, seems OK to pull in.

pelwell, 2018-06-26:
As a bug fix, this commit was already included in Linux 4.14.40 so there's no need to cherry-pick.

closed 2018-06-26 by pelwell

alexellis, 2018-06-26:
Will this be in the next Raspbian release or do we need to run rpi-update?

pelwell, 2018-06-26:
It will be in the next Raspbian (the raspberrypi-kernel package may already have been updated - it's worth checking).

bernhara, 2018-06-28:
I checked right now, and the latest raspbian kernel version is 4.14.**34**-v7+.
I'm afraid that we have still to way for the next release.

JamesH65, 2018-06-28:
rpi-update will get you 4.14.50. I think that will be the next one when the repos are updated, next few days I guess.

KptnKMan, 2018-06-28:
Which version is the change included?
I just updated, and saw that `4.14.52-v7+` is pulled

```
root@raspberrypi:~# rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 13545  100 13545    0     0  68365      0 --:--:-- --:--:-- --:--:-- 68756
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Backing up firmware
 *** Backing up modules 4.14.34-v7+
#############################################################
This update bumps to rpi-4.14.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=197689
##############################################################
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    278      0 --:--:-- --:--:-- --:--:--   278
100 55.8M  100 55.8M    0     0   657k      0  0:01:27  0:01:27 --:--:-- 1901k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.14.52+
 *** depmod 4.14.52-v7+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 963a31330613a38e160dc98b708c662dd32631d5
 *** A reboot is needed to activate the new firmware
```

JamesH65, 2018-06-29:
Doing apt update/upgrade will get you the official 4.14.50 now, so that the best one to use. 52 is pretty bleeding edge.