raspberrypi/linux #2581: skb_dst_update_pmtu in ip_vti.c: Used but not implemented?

Alex332211, 2018-06-12:
Hello,

I have a RPi2 running 4.9.y (since that's the version available when I got the board). For using a VPN software, there's have to be ip_vti module loaded.  Hence I've got linux-rpi-4.9.y, installed the appropriate packages for includes to be available and finally, ip_vti.c compiled into ip_vti.ko without a hitch.

Now a new board came along (still RPi2), which has 4.14.34-v7+ installed, which seems to be the current stable version of raspbian. But now, I **unable to compile** ip_vti.c. Make throws error of skb_dst_update_pmtu being implicit declaration error.

To my surprise, I'm indeed unable to find its' declaration. Are we missing a "h" file to be included? Is there something else with current ip_vtc.c source code? How can we compile it successfully?

Thank you/

pelwell, 2018-06-12:
I think you have the wrong (or even a mixed) version of the source checked out. The 4.14.34 release was made from commit f70eae4. That predates commit 6a3c946b that introduce the skb_dst_update_pmtu  helper function.

Enabling CONFIG_NET_IPVTI=m builds the required module, and works at both commits listed above. You can set the current working directory to any specific commit using `git checkout <hash>`, e.g.:
```
$ git checkout f70eae4
```
Note that in order to wind backwards like that you mustn't use the `--depth 1` option to `git clone`.

That compiled module is only 16kB and it sounds like it might be generally useful. Do you want to turn you issue into a request to build it as part of the standard kernel releases?

pelwell, 2018-06-12:
I've gone ahead and added the setting anyway - the ip_vti module will be in future kernel builds.

Alex332211, 2018-06-14:
Hello pelwell,

IMHO that's the right thing to do. VTI module is needed for many real world VPN implementations, aside only really special compatibility issues (with particular opposite end gateways). That will make it much easier and achievable for many, since not everybody know how compile kernel modules.

As for version differences you've mentioned above - maybe it's my lack of knowledge, but the only way I'm familiar with to get linux-rpi sources, is to download them here. The only thing is, they're not available for 3.14.34 specifically (to my knowledge), but instead as 3.14.y.

I've already was able to find exactly that diff (though in other OS release, namely DD-WRT), changed the source file accordingly and it compiled and inserted successfully. 

Anyhow, providing the module available in the first place, is the right thing to do. Thank you!

closed 2018-06-14 by pelwell