raspberrypi/linux #2587: Various kernel oopses / audio dropping out, when running Qt EGLFS

Ealdwulf, 2018-06-20:
We are seeing kernel 'Oopses', while running the following combination:
   * A simple Qt app running using EGLFS (source [here](https://github.com/KanoComputing/qmlmatrix/tree/Jessie-v3.15.0))
   * Something that plays a long wav file.

Both of these elements seem to be required for the crash. 
This is reproducable on stock raspbian (2018-04-18).
We did our own build of Qt to use eglfs and libinput, but it's build from stock qt 5.10.0. We carry one small patch to qt, but have reproduced the crash without it.  

Here are the kernel error messages we have seen. (closest to stock raspbian/qt first):
* Captured over serial, vanilla qt and kernel: https://gist.github.com/Ealdwulf/1dcaa867199fa59c921673479ef30fe1
* Captured over serial, patched qt: https://gist.github.com/Ealdwulf/2bb0beb46085b8b61501864b470265de

* Captured over ssh, patched qt: https://gist.github.com/Ealdwulf/d26ae24f059e37acda58ef384bce9c27

* Captured over serial, patched qt, kernel 4.14.50 rebuilt with more debug options[1]: https://gist.github.com/wizofe/469d8c4f092e5ae7d9adc9147f83e167

The following are the minimum steps to reproduce, that we have so far:
```
# (connected to wifi and enabled ssh, and switched audio to Analog output). 

sudo apt-get update
sudo apt-get install  libts-0.0-0  tsconf

# This is build from Stock QT 5.10.0:
wget http://dev.kano.me/temp/kernel_crash_repr/libqt5all_5.10-1_armhf.deb
# This is a very simple Qt app:
wget http://dev.kano.me/temp/kernel_crash_repr/qmlmatrix_1.0-1.20170815_all.deb
# A long wav file:
wget http://dev.kano.me/temp/kernel_crash_repr/chippytoon.wav
# Despite the name, this just contains some symlinks:
wget http://dev.kano.me/temp/kernel_crash_repr/kano-graphics-libs_4.0.0-0.20180530_all.deb

sudo dpkg -i qmlmatrix_1.0-1.20170815_all.deb libqt5all_5.10-1_armhf.deb kano-graphics-libs_4.0.0-0.20180530_all.deb

# a script to run the qt app repeatedly
cat <<END >>qtest
while true; do
  qmlmatrix &
  QPID=$!
  sleep 240
  kill $QPID
done
END

while true; do aplay chippytoon.wav ; done &

. ./qtest
```

The result is one of the following:
 *  A kernel crash, with varying output (see above). 25..50% of the time
 * No kernel crash, but instead audio stops working (aplay sometimes starts to give errors, `aplay: pcm_write:2011: write error: Input/output error` , sometimes continues oblivious). After this it needs a reboot to start working again.
* (Seen once, not on raspbian) The display siezes and the firmware stops responding to vcgencmd, but ssh still works
* (Seen once, on raspbian)  The display siezes but firmware still responds to vcgencmd, and ssh still works.
This is using a 2.5A power supply, where the pi and a  speaker are the only things powered by it. We do not see low voltage warnings in this setup.

It is not clear to us whether the kernel or firmware is at fault. 

Please let us know if there are other logs or traces we could enable which would help diagnose this.

We have a [patch](https://github.com/KanoComputing/qmlmatrix/commit/049060bebcfc532bba513cd54cccff72a66edf9d) to our app that seems to prevent the problem, but we have a fair few qt apps, so we'd like to persue this, to be sure that the others aren't vulnerable at some lower probability.

[1]
CONFIG_DEBUG_INFO=y    
CONFIG_MAGIC_SYSRQ=y
CONFIG_FRAME_POINTER=y  
CONFIG_KGDB=y
CONFIG_KGDB_SERIAL_CONSOLE=y
CONFIG_KGDB_KDB=y
CONFIG_KDB_KEYBOARD=y

pelwell, 2018-06-20:
Which platform is this running on?

Have you tried with the latest rpi-update firmware?

Ealdwulf, 2018-06-20:
We are using pi 3B. I've seen it on a 2B as well.
I did reproduce with the lastest rpi-update as of a few days ago, but not on vanilla raspbian - I'll try that  today.

Ealdwulf, 2018-06-20:
Yes, the crash still occurs on the latest rpi-update firmware (firmware `fe525d2be041c1a9b924824e430b5d51214315c4`, kernel `4.14.50+`).

It seems to take a lot longer to trigger (the above test was running for half an hour)
Here are some more kernel messages:
```
﻿[ 1458.875616] Unhandled prefetch abort: unknown 1 (0x001) at 0xaaaa75a2
[ 1458.964953] systemd: 32 output lines suppressed due to ratelimiting
[ 1459.386477] EXT4-fs error (device mmcblk0p2): ext4_find_dest_de:1806: inode #2: block 8422: comm systemd: bad entry in directory: rec_len % 4 != 0 - offset=0(0), inode=2863326613, rec_len=43690, name_len=170
[ 1459.775672] Unable to handle kernel paging request at virtual address 00002b89
[ 1459.783024] pgd = 80004000
[ 1459.785787] [00002b89] *pgd=00000000
[ 1459.785835] Unable to handle kernel paging request at virtual address 00002c2d
[ 1459.785838] pgd = b75b4000
[ 1459.785840] [00002c2d] *pgd=37483835, *pte=00000000, *ppte=00000000
[ 1459.785854] Internal error: Oops: 17 [#1] SMP ARM
[ 1459.785859] Modules linked in: fuse rfcomm cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic brcmfmac brcmutil cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd uio_pdrv_genirq uio fixed evdev joydev hid_multitouch i2c_dev ip_tables x_tables ipv6
[ 1459.785922] CPU: 3 PID: 343 Comm: dhcpcd Tainted: G         C      4.14.50-v7+ #1122
[ 1459.785923] Hardware name: BCM2835
[ 1459.785927] task: b7795a00 task.stack: b92d6000
[ 1459.785939] PC is at sock_poll+0x24/0xa8
[ 1459.785948] LR is at ep_send_events_proc+0x90/0x144
[ 1459.785951] pc : [<8066c2d0>]    lr : [<802d6be4>]    psr: 80000013
[ 1459.785954] sp : b92d7e40  ip : b92d7e60  fp : b92d7e5c
[ 1459.785957] r10: 00000000  r9 : b92d7f18  r8 : b657af30
[ 1459.785960] r7 : b92d7e68  r6 : b09f9780  r5 : afaa8700  r4 : b970558c
[ 1459.785963] r3 : 8066c2ac  r2 : 00002b6d  r1 : b92d7e68  r0 : b09f9780
[ 1459.785967] Flags: Nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[ 1459.785970] Control: 10c5383d  Table: 375b406a  DAC: 00000055
[ 1459.785975] Process dhcpcd (pid: 343, stack limit = 0xb92d6210)
[ 1459.785979] Stack: (0xb92d7e40 to 0xb92d8000)
[ 1459.785985] 7e40: b970558c b92d7ea0 7ea68ab8 b92d7f18 b92d7e9c b92d7e60 802d6be4 8066c2b8
[ 1459.785991] 7e60: 60000013 b657af00 00000000 00000019 802d6b54 b657af00 b92d7ea0 00000000
[ 1459.785997] 7e80: 802d6b54 b657af30 b92d7f18 00000000 b92d7ecc b92d7ea0 802d74b4 802d6b60
[ 1459.786004] 7ea0: b92d7ea0 b92d7ea0 b97efb40 00000001 7ea68ab8 b97efb40 b657af30 00000000
[ 1459.786009] 7ec0: b92d7f6c b92d7ed0 802d8864 802d7424 00000000 00000504 0000031c 00000000
[ 1459.786015] 7ee0: 2bbff380 b92d7f40 b968da40 b657af30 05f5e100 00000000 b92d7f00 b657af00
[ 1459.786021] 7f00: 56df3a87 0000020d 000008d0 00000000 12cb1a87 b92d7f20 00000001 7ea68ab8
[ 1459.786026] 7f20: 22a5f107 00000504 00000001 b7795a00 80148e4c 00000100 00000200 801edbe0
[ 1459.786032] 7f40: 00000000 000c283e 00000001 7ea68ab8 00000003 b92d7f78 b92d6000 00000000
[ 1459.786038] 7f60: b92d7fa4 b92d7f70 802d8bc0 802d8670 00000000 00000000 00007a03 00000000
[ 1459.786044] 7f80: 7ea68a0c 7ea68bb0 00000008 0020c49b 0000015a 80108204 00000000 b92d7fa8
[ 1459.786050] 7fa0: 80108060 802d8b10 7ea68bb0 00000008 00000003 7ea68ab8 00000001 000c283e
[ 1459.786056] 7fc0: 7ea68bb0 00000008 0020c49b 0000015a ffffffff 431bde83 000f423f 00ac8c50
[ 1459.786061] 7fe0: 00000000 7ea68a90 00018494 76f1ddb8 60000010 00000003 00000000 00000000
[ 1459.786080] [<8066c2d0>] (sock_poll) from [<802d6be4>] (ep_send_events_proc+0x90/0x144)
[ 1459.786089] [<802d6be4>] (ep_send_events_proc) from [<802d74b4>] (ep_scan_ready_list+0x9c/0x1cc)
[ 1459.786097] [<802d74b4>] (ep_scan_ready_list) from [<802d8864>] (SyS_epoll_wait+0x200/0x4a0)
[ 1459.786104] [<802d8864>] (SyS_epoll_wait) from [<802d8bc0>] (SyS_epoll_pwait+0xbc/0x17c)
[ 1459.786116] [<802d8bc0>] (SyS_epoll_pwait) from [<80108060>] (ret_fast_syscall+0x0/0x28)
[ 1459.786126] Code: e5905090 e1a06000 e1a07001 e5952014 (e59230c0) 
[ 1459.786137] ---[ end trace bfc137ac1a74431c ]---
[ 1459.786688] Unable to handle kernel paging request at virtual address f7dd00e8
[ 1459.786690] pgd = b75b4000
[ 1459.786693] [f7dd00e8] *pgd=00000000
[ 1459.786699] Internal error: Oops: 5 [#2] SMP ARM
[ 1459.786703] Modules linked in: fuse rfcomm cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic brcmfmac brcmutil cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd uio_pdrv_genirq uio fixed evdev joydev hid_multitouch i2c_dev ip_tables x_tables ipv6
[ 1459.786752] CPU: 3 PID: 343 Comm: dhcpcd Tainted: G      D  C      4.14.50-v7+ #1122
[ 1459.786754] Hardware name: BCM2835
[ 1459.786756] task: b7795a00 task.stack: b92d6000
[ 1459.786766] PC is at locks_remove_posix+0x24/0x150
[ 1459.786774] LR is at filp_close+0x68/0x8c
[ 1459.786776] pc : [<802e0d44>]    lr : [<80286e40>]    psr: 20000113
[ 1459.786779] sp : b92d7b48  ip : b92d7bf0  fp : b92d7bec
[ 1459.786781] r10: b7795a00  r9 : b96e4818  r8 : 00000000
[ 1459.786784] r7 : b96e4800  r6 : b96e4800  r5 : f7dd0000  r4 : 00000000
[ 1459.786787] r3 : afa25a18  r2 : b09f9300  r1 : b96e4800  r0 : b09f9300
[ 1459.786791] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[ 1459.786794] Control: 10c5383d  Table: 375b406a  DAC: 00000055
[ 1459.786798] Process dhcpcd (pid: 343, stack limit = 0xb92d6210)
[ 1459.786802] Stack: (0xb92d7b48 to 0xb92d8000)
[... 105 lines cut ...]
[ 1460.462146] Unable to handle kernel paging request at virtual address 00002b69
[... cut ...]
[ 1460.462174] [00002b69] *pgd=00000000
[ 1460.462178] Internal error: Oops: 5 [#14] SMP ARM
[ 1460.462179] Modules linked in: (FEK)
[ 1460.462183] Unable to handle kernel paging request at virtual address 00002b69
[ 1460.462185] pgd = b9690000
[ 1460.462187] [00002b69] *pgd=00000000
[ 1460.462190] Internal error: Oops: 5 [#15] SMP ARM
[ 1460.462191] Modules linked in: (FEK)
[ 1460.462196] Unable to handle kernel paging request at virtual address 00002b69
[ 1460.462198] pgd = b9690000
[ 1460.462200] [00002b69] *pgd=00000000

```

pelwell, 2018-06-20:
Have you run `vcgencmd get_throttled` to confirm that you are not getting an under-voltage condition? Semi-random crashes under load in different places are usually voltage-related.

timg236, 2018-06-21:
I was able to reproduce the issue on a Raspberry 3 model B+ with an official Raspberry Pi power supply. The issue only seems to occur with analog audio. 

`vcgencmd get_throttled` always returned "throttled=0x0"

@pelwell  suggested disabling high quality audio by adding audio_pwm_mode=1 to config.txt. I tried this and the crash no longer occurs.

It may be that qmlmatrix is causing a high system/GL load which slows down the VC audio component or ALSA driver. If the ALSA VC audio driver fails to handle a buffer underflow correctly and tramples memory then it would explain the random oops.

I was eventually able to trigger the same issue by running glmark2 plus multiple instances of memtester. I'm not sure why qmlmatrix was worse, perhaps large texture uploads ?

Ealdwulf, 2018-06-21:
Thanks @timg236! 

I'm glad someone else managed to reproduce it, because after moving offices I'm having difficulty doing so (at least on vanilla raspbian - we reproduced it many times on Kano OS, which is closely related). 

In the version of qmlmatrix above, we missed setting the qml attribgute `renderTarget: Canvas.FramebufferObject` which we believe meant that there was an additional copy-to-gpu option each frame. (Hence this [fix](https://github.com/KanoComputing/qmlmatrix/commit/049060bebcfc532bba513cd54cccff72a66edf9d)). So that would be consistent with memory subsystem load. 

We will try out `audio_pwm_mode=1`. 

We are pretty sure that it's not a voltage issue; we have seen it on both kits where don't get low voltage warnings, and those where we do. Thats from the kernel rather than vcgencmd; I'll check vcgencmd as well just to be sure.

Ealdwulf, 2018-06-25:
We have tested `audio_pwm_mode=1` and it seems to prevent the problem. 

By the way, the above is the simplest reproducer, not the fastest. We can reproduce it in about 5 minutes. When we are running another app at the same time  as qmlmatrix (a love2d app, which runs on its own dispmanx layer) the bug seems to happen more quickly. We can give you a binary of that, or an image containing it, if that would help you debug the problem.

timg236, 2018-06-29:
I think the existing tests are good enough. The problem seems to be a bug triggered by underflow in the audio_pwm_mode=2 code which corrupts VC memory and then causes invalid DMA writes which in turn cause random kernel oops.

I have a vc firmware patch which resolves the crash but needs a bit more testing.

N.B This would only fix the crash and not the audio glitches. Ensuring that high quality analog audio gets enough resources when graphics/VCHIQ is heavily loaded might not be practical to fix.

Ealdwulf, 2018-07-02:
That's really good news; we'll test it when it comes out.

niwa3836, 2018-07-03:
Hello (and thanks to jdb for pointing me to this problem page). Here is another view / test which I believe quickly reveals the issue. rPI 3B+ with latest code, non desktop using mpg123 to play audio, using 2 x SSH windows to start and stop audio.

Test 1:
dtparam=audio=on
audio_pwm_mode=1
window 1: start audio (good)
window 2: start audio (good and mixing with first)
window 1, stop and restart same audio (good)
Test 2:
dtparam=audio=on
audio_pwm_mode=2
window 1: start audio (good)
window 2: start audio (good and mixing with first)
window 1, stop and restart same audio (BAD).

Added here in case it helps anyone. Really looking forward to a fix for this, fingers crossed!

popcornmix, 2018-07-03:
Latest rpi-update firmware has the fix that has been discussed here. Please update and test.

niwa3836, 2018-07-04:
Hello all, and thanks for those involved in looking at this. The test case I described above (window 1: start audio (good), window 2: start audio (good and mixing with first), window 1, stop and restart same audio (BAD).) no longer fails. I've tested back and forth between windows and with the new rpi-update i can no longer break it. Thanks for the fix!

popcornmix, 2018-07-04:
@Ealdwulf can you confirm the fix is good?

Ealdwulf, 2018-07-04:
We are running a soak test, will let you know later today @popcornmix .

Ealdwulf, 2018-07-04:
Looks pretty good - it's been running for 4 hours, restarting qmlmatrix every 450 seconds, with no crashes. We're going to leave the soak running overnight , but just out of an abundance of caution.

Ealdwulf, 2018-07-05:
It's still going this morning, so it looks like it's fixed 👍 .

closed 2018-07-05 by Ealdwulf