raspberrypi/linux #2588: Raspberry Pi 3 B and B+ Kernel panic: Unable to handle kernel paging request at virtual address 72656d41

gourry, 2018-06-22:
Hello,
I'm frequently getting this errors and I don't know how to solve them. Everything started after updating the kernel to the latest version (via apt-get) to resolve random lockups #2576 that could be due to low power.
Now the measured voltage on the Pi GPIO connector is exactly 5V and there are no messages telling about low voltages.

I tried replacing the Raspberry Pi 3 B+ with the older Raspberry Pi 3 B using the same SD card but it appears that the error is affecting both.

I installed all the updates available via apt-get and performed a rpi-update. 
I tried changing size of the swap memory (since the error was mentioning "paging") reducing or increasing it but apparently it didn't provide any change.

For reference I'm attaching some examples of the error. 

Any idea on how to solve it?

Thanks in advance.

Raspberry Pi 3 B+
===========
[ 2094.868335] Unable to handle kernel paging request at virtual address 72656d41
[ 2094.868426] pgd = 80004000
[ 2094.868455] [72656d41] *pgd=00000000
[ 2094.868496] Internal error: Oops: 5 [#1] SMP ARM
[ 2094.868537] Modules linked in: fuse appletalk psnap llc ax25 cmac sha256_generic arc4 ecb md4 md5 hmac nls_utf8 cifs ccm brcmfmac brcmutil cfg80211 rfkill i2c_bcm2835 fixed uio_pdrv_genirq uio i2c_dev i2c_bcm2708 snd_bcm2835(C) snd_pcm snd_timer snd ip_tables x_tables ipv6
[ 2094.868798] CPU: 1 PID: 915 Comm: cifsd Tainted: G         C      4.14.50-v7+ #1122
[ 2094.868861] Hardware name: BCM2835
[ 2094.868893] task: b553bc00 task.stack: b544c000
[ 2094.868942] PC is at prefetch_freepointer+0x20/0x2c
[ 2094.868988] LR is at kmem_cache_alloc+0xd8/0x230
[ 2094.869028] pc : [<80272870>]    lr : [<802772b8>]    psr: 20000013
[ 2094.869082] sp : b544de80  ip : b544de90  fp : b544de8c
[ 2094.869126] r10: 72656d41  r9 : bceab9c0  r8 : 802211c4
[ 2094.869172] r7 : 0001703e  r6 : 01011200  r5 : bceab9c0  r4 : b88fc700
[ 2094.869227] r3 : 00000000  r2 : b544de90  r1 : 72656d41  r0 : bceab9c0
[ 2094.869281] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[ 2094.869340] Control: 10c5383d  Table: 37d7006a  DAC: 00000055
[ 2094.869389] Process cifsd (pid: 915, stack limit = 0xb544c210)
[ 2094.869439] Stack: (0xb544de80 to 0xb544e000)
[ 2094.871379] de80: b544decc b544de90 802772b8 8027285c 80955fb0 00000002 b544dec4 72656d41
[ 2094.875396] dea0: 00000001 01011200 01411240 b55fc8c0 00400000 801608fc b544c000 00000049
[ 2094.879478] dec0: b544dedc b544ded0 802211c4 802771ec b544df24 b544dee0 802212ac 802211ac
[ 2094.883847] dee0: 0000000a b544c000 b88fd543 bc77e800 7f56c130 bc77e93c 80b6cc80 bc77e800
[ 2094.888407] df00: 7f56c130 bc77e93c b544c000 80cbdbd4 b88fd500 00000049 b544df34 b544df28
[ 2094.893124] df20: 7f52c100 80221254 b544df7c b544df38 7f51b3fc 7f52c0e0 b544df5c b88fd500
[ 2094.897917] df40: 00000000 80b6cc80 bc77e800 7f51af24 b89f919c b89f9180 00000000 b7d5aa40
[ 2094.902969] df60: bc77e800 7f51af24 b89f919c b5521db8 b544dfac b544df80 8013d8f8 7f51af30
[ 2094.908340] df80: 80102d94 b7d5aa40 8013d7bc 00000000 00000000 00000000 00000000 00000000
[ 2094.913897] dfa0: 00000000 b544dfb0 8010810c 8013d7c8 00000000 00000000 00000000 00000000
[ 2094.919712] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 2094.925741] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[ 2094.931875] [<80272870>] (prefetch_freepointer) from [<802772b8>] (kmem_cache_alloc+0xd8/0x230)
[ 2094.938270] [<802772b8>] (kmem_cache_alloc) from [<802211c4>] (mempool_alloc_slab+0x24/0x28)
[ 2094.944656] [<802211c4>] (mempool_alloc_slab) from [<802212ac>] (mempool_alloc+0x64/0x154)
[ 2094.951004] [<802212ac>] (mempool_alloc) from [<7f52c100>] (cifs_small_buf_get+0x2c/0x50 [cifs])
[ 2094.957414] [<7f52c100>] (cifs_small_buf_get [cifs]) from [<7f51b3fc>] (cifs_demultiplex_thread+0x4d8/0x7cc [cifs])
[ 2094.963640] [<7f51b3fc>] (cifs_demultiplex_thread [cifs]) from [<8013d8f8>] (kthread+0x13c/0x16c)
[ 2094.969691] [<8013d8f8>] (kthread) from [<8010810c>] (ret_from_fork+0x14/0x28)
[ 2094.972722] Code: e8bd4000 e3510000 089da800 e5903014 (e7913003) 
[ 2094.975764] ---[ end trace e2e618b6dc8074cf ]---
[ 2104.641124] Unable to handle kernel paging request at virtual address 72656d41
[ 2104.644944] pgd = 80004000
[ 2104.648633] [72656d41] *pgd=00000000
[ 2104.652276] Internal error: Oops: 5 [#2] SMP ARM
[ 2104.655888] Modules linked in: fuse appletalk psnap llc ax25 cmac sha256_generic arc4 ecb md4 md5 hmac nls_utf8 cifs ccm brcmfmac brcmutil cfg80211 rfkill i2c_bcm2835 fixed uio_pdrv_genirq uio i2c_dev i2c_bcm2708 snd_bcm2835(C) snd_pcm snd_timer snd ip_tables x_tables ipv6
[ 2104.669632] CPU: 1 PID: 2007 Comm: kworker/u8:1 Tainted: G      D  C      4.14.50-v7+ #1122
[ 2104.676124] Hardware name: BCM2835
[ 2104.679309] Workqueue: writeback wb_workfn (flush-cifs-1)
[ 2104.682474] task: b8b88f00 task.stack: 937c0000
[ 2104.685558] PC is at kmem_cache_alloc+0x80/0x230
[ 2104.688560] LR is at mempool_alloc_slab+0x24/0x28
[ 2104.691419] pc : [<80277260>]    lr : [<802211c4>]    psr: 20000013
[ 2104.694269] sp : 937c1b20  ip : 937c1b60  fp : 937c1b5c
[... 530 lines cut ...]
Jun 18 19:17:13 raspberrypi kernel: [330930.270470] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Jun 18 19:17:13 raspberrypi kernel: [330930.272272] Control: 10c5383d  Table: 00bf406a  DAC: 00000055
Jun 18 19:17:13 raspberrypi kernel: [330930.274101] Process cifsd (pid: 7068, stack limit = 0xbc0e4210)
Jun 18 19:17:13 raspberrypi kernel: [330930.275916] Stack: (0xbc0e5e80 to 0xbc0e6000)
Jun 18 19:17:13 raspberrypi kernel: [330930.277711] 5e80: bc0e5ecc bc0e5e90 80276eb8 8027245c 80955800 00000002 bc0e5ec4 20a8c320
Jun 18 19:17:13 raspberrypi kernel: [330930.281512] 5ea0: 00000001 01011200 01411240 b2aaff00 00400000 80160860 bc0e4000 00000064
Jun 18 19:17:13 raspberrypi kernel: [330930.285426] 5ec0: bc0e5edc bc0e5ed0 80220eec 80276dec bc0e5f24 bc0e5ee0 80220fd4 80220ed4
Jun 18 19:17:13 raspberrypi kernel: [330930.289635] 5ee0: 00000025 b241dfc0 a7c778c3 b2e0e000 7f465130 b2e0e13c bc2b3a00 b2e0e000
Jun 18 19:17:13 raspberrypi kernel: [330930.294075] 5f00: 7f465130 b2e0e13c bc0e4000 80cbdc14 a7c77880 00000064 bc0e5f34 bc0e5f28
Jun 18 19:17:13 raspberrypi kernel: [330930.298762] 5f20: 7f4250a0 80220f7c bc0e5f7c bc0e5f38 7f4143e0 7f425080 bc0e5f5c a7c77880
Jun 18 19:17:13 raspberrypi kernel: [330930.303550] 5f40: 00000000 bc2b3a00 b2e0e000 7f413f08 b16ebe5c b16ebe40 00000000 b241dfc0
Jun 18 19:17:13 raspberrypi kernel: [330930.308581] 5f60: b2e0e000 7f413f08 b16ebe5c 811a1db8 bc0e5fac bc0e5f80 8013d860 7f413f14
Jun 18 19:17:13 raspberrypi kernel: [330930.313912] 5f80: 80102d94 b241dfc0 8013d724 00000000 00000000 00000000 00000000 00000000
Jun 18 19:17:13 raspberrypi kernel: [330930.319420] 5fa0: 00000000 bc0e5fb0 8010810c 8013d730 00000000 00000000 00000000 00000000
Jun 18 19:17:13 raspberrypi kernel: [330930.325223] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Jun 18 19:17:13 raspberrypi kernel: [330930.331218] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
Jun 18 19:17:13 raspberrypi kernel: [330930.337291] [<80272470>] (prefetch_freepointer) from [<80276eb8>] (kmem_cache_alloc+0xd8/0x230)
Jun 18 19:17:13 raspberrypi kernel: [330930.343553] [<80276eb8>] (kmem_cache_alloc) from [<80220eec>] (mempool_alloc_slab+0x24/0x28)
Jun 18 19:17:13 raspberrypi kernel: [330930.349738] [<80220eec>] (mempool_alloc_slab) from [<80220fd4>] (mempool_alloc+0x64/0x154)
Jun 18 19:17:13 raspberrypi kernel: [330930.355982] [<80220fd4>] (mempool_alloc) from [<7f4250a0>] (cifs_small_buf_get+0x2c/0x50 [cifs])
Jun 18 19:17:13 raspberrypi kernel: [330930.362263] [<7f4250a0>] (cifs_small_buf_get [cifs]) from [<7f4143e0>] (cifs_demultiplex_thread+0x4d8/0x7cc [cifs])
Jun 18 19:17:13 raspberrypi kernel: [330930.368367] [<7f4143e0>] (cifs_demultiplex_thread [cifs]) from [<8013d860>] (kthread+0x13c/0x16c)
Jun 18 19:17:13 raspberrypi kernel: [330930.374295] [<8013d860>] (kthread) from [<8010810c>] (ret_from_fork+0x14/0x28)
Jun 18 19:17:13 raspberrypi kernel: [330930.377234] Code: e8bd4000 e3510000 089da800 e5903014 (e7913003) 
Jun 18 19:17:13 raspberrypi kernel: [330930.380193] ---[ end trace 123b2f9d1340b7e8 ]---

pelwell, 2018-06-22:
A quick search makes me think this problem is not exclusive to Raspberry Pis:
* https://bugzilla.redhat.com/show_bug.cgi?id=1582698
* https://www.mail-archive.com/kernel-packages@lists.launchpad.net/msg285947.html
* https://bugs.archlinux.org/task/57474

The last of these includes the suggestion to include "vers=1.0" in /etc/fstab (or the mount command), e.g.:
```
//FS1/data /home/username/data cifs vers=1.0,credentials=/home/username/smbcredentials,uid=1000
```

gourry, 2018-06-23:
Thanks. I tried reducing the samba version to 1.0 ( I was using 3.0 before) to see if the problem is resolved. This is not ideal because I know that samba 1.0 is not secure, but I will give it a try.

By the way, how could you relate those topics to the error I have? Before opening this issue I searched multiple times on google using the entire stacktrace or reducing it to only the "Unable to handle kernel paging request at virtual address 72656d41" string with or without the address and nothing relevant come up. What's the receipt to search for a similar issue? 
Thanks!

pelwell, 2018-06-23:
It varies from issue to issue. In this case I could see that the "address" 72656d41 was more likely to be a misinterpretation of some character data (`Amer` - does that have any obvious relevance to your samba share?), so although it was consistent for you it could easily be different for others. Full stack traces will be full of addresses which can change from build to build and architecture to architecture, so I reduced my search to just `cifs` and `prefetch_freepointer` (the function where the fault occurs).

pelwell, 2018-06-23:
It looks like a memory corruption issue - you may get somewhere with a kernel built with `CONFIG_SLUB_DEBUG=y`, and `slub_debug=FZ` on the kernel command line to enable a restricted set of checks to reduce the performance impact.

JamesH65, 2018-06-28:
It does look like this is a kernel bug, there are a number of additional, but all recent, posts with very similar back traces, but involving network traffic. Odd that things work OK with some network adaptors but not others. I guess there is a particular networking rune that some cards are not invoking, leading to a memory corruption. 
Your post doesn't say whether you are accessing the CIFS attached drive via networking, or whether this is a self contained system, any more information there that would help in replication?

gourry, 2018-06-30:
Hello,
I did an extensive test using samba version 1.0 and I can confirm that the problem is resolved. The reason why I started using a higher version of samba is that version 1.0 is no longer the default since it is not, as far as I understand, considered secure.
I am using samba as client to access a share on my nas. Details follows:
* Tried both a Raspberry Pi 3 B+ and Raspberry Pi 3 B. I got the same corruption.
* Used CIFS vers=3.0 mounted through autofs. Autofs conf:
`share -fstype=cifs,rw,soft,intr,nosuid,nodev,guest,file_mode=0777,dir_mode=0777,uid=1000,gid=1000,vers=3.0 ://10.0.0.1/Share`
* NAS is a Synology DS918+ with samba v3 enabled. Storage is a SHR volume with 3 4Tb WD RED drives and one SSD Samsung 840 evo for cache
* Both the Raspberry Pi and the Synology NAS are connected to a TP-Link gigabit switch TL-SG1005D
* Raspberry Pi are connected using the internal ethernet card. I know that Raspberry Pi 3 B+ and Raspberry Pi 3 B have a different network adapter, this seems to exclude the driver (however I don't know much about Linux drivers to tell)
* The problem does not seem related to the volume of traffic. It won't happen if there is no access to the NAS (probably due to autofs not mounting the device) but I saw it happening both with low traffic and high one. It seems more a matter of time.

JamesH65, 2018-07-11:
@gourry There is a new bug fix for the ethernet driver which seems to have fixed a CIFS related issues along the way. It should be available in rpi-update tomorrow, or you can build you own kernel now if you wish.

https://github.com/raspberrypi/linux/pull/2614

I suspect this may fix the issue you are seeing.

popcornmix, 2018-07-11:
The potential fix is now in rpi-update kernel.

gourry, 2018-07-12:
Thank you for the update! This week I am out from home, as soon as I get back I will update the kernel and try again samba v3. Than I will report the results. Thanks for now!

JamesH65, 2018-08-31:
@gourry Did you have any luck with the latest kernel?

gourry, 2018-09-02:
Yes the issue appears to be solved, thank you! I didn't reply before because now I have another kernel panic (Same reported here: #2307 ) that I initially thought to be the same issue, but I think it's unrelated. I guess that I am very unlucky with the Raspberry Pi...
Thanks for the help!

closed 2018-09-02 by gourry