raspberrypi/linux #2592: IPv6 MASQUERADE broken with kernel 4.14

gcoretech, 2018-06-26:
Example:

root@client:/root# ping6 -c 5 google.de
client (fd00:1023:cafe:1002::2) -> raspberry pi (fd00:1023:cafe:1002::1) -> MASQUERADE -> internet uplink
(The internet uplink's global IPv6 is assigned via dhcpv6 and is reachable)

root@router:/home/pi# uname -a
Linux router 4.14.50+ #1121 Mon Jun 18 16:19:29 BST 2018 armv6l GNU/Linux

root@router:/home/pi# tcpdump -nn -i eth0.1 icmp6
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth0.1, link-type EN10MB (Ethernet), capture size 262144 bytes

11:25:22.067803 IP6 fd00:1023:cafe:1002::2 > 2a00:1450:4001:815::2003: ICMP6, echo request, seq 0, length 16
11:25:23.069350 IP6 fd00:1023:cafe:1002::2 > 2a00:1450:4001:815::2003: ICMP6, echo request, seq 1, length 16
11:25:24.068949 IP6 fd00:1023:cafe:1002::2 > 2a00:1450:4001:815::2003: ICMP6, echo request, seq 2, length 16
11:25:25.067708 IP6 fd00:1023:cafe:1002::2 > 2a00:1450:4001:815::2003: ICMP6, echo request, seq 3, length 16
11:25:26.068858 IP6 fd00:1023:cafe:1002::2 > 2a00:1450:4001:815::2003: ICMP6, echo request, seq 4, length 16

root@router:/home/pi# ip6tables -t nat -L POSTROUTING -v -n
Chain POSTROUTING (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 MASQUERADE  all      *      eth0.1  fd00::/8             ::/0       

No packets are matched in POSTROUTING chain with kernel 4.14.50 (git commit ced2c6ff6ed484112c265d44b23efffb9b67c02b) or the current raspbian stable kernel. SNAT and NETMAP work as expected.

    
With kernel 4.9.80 (rpi-update 5c80565c5c0c7f820258c792a98b56f22db2dd03) IPv6 MASQUERADE works as expected:

root@router:/home/pi# ip6tables -t nat -L POSTROUTING -v -n
Chain POSTROUTING (policy ACCEPT 5 packets, 352 bytes)
 pkts bytes target     prot opt in     out     source               destination         
   29  1732 MASQUERADE  all      *      eth0.1  fd00::/8             ::/0
   
kernel 4.4.50 52241088c1da59a359110d39c1875cda56496764 works as well with the exact same network/iptables configuration.

Does anybody know if this is a known issue?

Best regards
Gerald

pelwell, 2018-06-26:
I found a few hits on the Arch Linux pages suggesting this is a known problem. I'd like to know if this is fixed in the latest 4.18 build, so can you either:
a) build your own kernel from the rpi-4.18.y branch, or
b) post full instructions to reproduce on 4.14?

pelwell, 2018-06-26:
If somebody can identify the fixing commit(s) then we'll happily cherry-pick (with a small amount of patching within reason). Are you up for trying an rpi-4.17.y build to narrow it down a bit further? I think that's possibly the next standard kernel version.

pelwell, 2018-06-26:
Can you guess where this is going? If it's not too much trouble, try 4.16 and 4.15, stopping at the first failure.

gcoretech, 2018-06-26:
Now I'm glad I kept my old cross compile environment ...

It's fixed in 4.16 and 4.15:

root@router:/home/pi# uname -a
Linux router 4.16.16 #1 Tue Jun 26 15:39:03 CEST 2018 armv6l GNU/Linux

root@router:/home/pi# ip6tables -t nat -L POSTROUTING -v -n
Chain POSTROUTING (policy ACCEPT 5 packets, 352 bytes)
 pkts bytes target     prot opt in     out     source               destination         
    1    56 MASQUERADE  all      *      eth0.1  fd00::/8             ::/0                

root@router:/home/pi# uname -a
Linux router 4.15.18 #1 Tue Jun 26 15:54:35 CEST 2018 armv6l GNU/Linux

root@router:/home/pi# ip6tables -t nat -L POSTROUTING -v -n
Chain POSTROUTING (policy ACCEPT 5 packets, 352 bytes)
 pkts bytes target     prot opt in     out     source               destination         
    1    56 MASQUERADE  all      *      eth0.1  fd00::/8             ::/0                

Best regards
Gerald

6by9, 2018-06-26:
https://github.com/raspberrypi/linux/commit/23715275e4fb6f64358a499d20928a9e93819f2f#diff-9b9f755e4675a7eb03e63cb37dc9935b ?

6by9, 2018-06-26:
It was the only thing I could see that mentioned masquerading at all in the 4.14 to 4.15 time frame. Both the commits referenced  in the commit msg are in 4.14.

pelwell, 2018-06-26:
@gcoretech Can you start with rpi-4.14.y, `git cherry-pick 2371527` and build that for testing?

gcoretech, 2018-06-26:
First I compiled 4.14.50 without the patch to verify it does not work:

root@router:/home/pi# uname -a
Linux router 4.14.50 #1 Tue Jun 26 17:50:49 CEST 2018 armv6l GNU/Linux

root@router:/home/pi# ip6tables -t nat -L POSTROUTING -v -n
Chain POSTROUTING (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 MASQUERADE  all      *      eth0.1  fd00::/8             ::/0                

WITH patch applied: https://github.com/raspberrypi/linux/commit/23715275e4fb6f64358a499d20928a9e93819f2f#diff-9b9f755e4675a7eb03e63cb37dc9935b
(make clean / recompiled all modules)

root@router:/home/pi# uname -a
Linux router 4.14.50 #2 Tue Jun 26 18:14:33 CEST 2018 armv6l GNU/Linux

root@router:/home/pi# ip6tables -t nat -L POSTROUTING -v -n
Chain POSTROUTING (policy ACCEPT 5 packets, 352 bytes)
 pkts bytes target     prot opt in     out     source               destination         
    1    56 MASQUERADE  all      *      eth0.1  fd00::/8             ::/0                

Problem solved :-)

Best regards
Gerald

pelwell, 2018-06-26:
Cherry-picked to rpi-4.14.y - thanks all!

JamesH65, 2018-06-27:
@gcoretech Happy to close this?

gcoretech, 2018-06-27:
Yes, of course. Thanks for reminding!

closed 2018-06-27 by gcoretech