raspberrypi/linux #2593: There is a potential Null Pointer Dereference error in linux/fs/afs/server.c

laoyaolandq, 2018-06-27:
In function _afs_find_server_ , 
```
196-217,
	p = afs_servers.rb_node;
	while (p) {
		server = rb_entry(p, struct afs_server, master_rb);

		_debug("- consider %p", p);

		if (addr.s_addr < server->addr.s_addr) {
			p = p->rb_left;
		} else if (addr.s_addr > server->addr.s_addr) {
			p = p->rb_right;
		} else {
			afs_get_server(server);
			goto found;
		}
	}

	server = NULL;
found:
	read_unlock(&afs_servers_lock);
	ASSERTIFCMP(server, server->addr.s_addr, ==, addr.s_addr);
	_leave(" = %p", server);
	return server;
```
The statement  `server = NULL;` will be executed if the loop is skipped or the loop is ended without executing `goto found`.Then statement `ASSERTIFCMP(server, **server->addr.s_addr**, ==, addr.s_addr);` dereferences _server_, whose value is NULL.

pelwell, 2018-06-27:
This is untouched upstream Linux code and any issues should be reported to the maintainers, but to save you wasting your time...

This is the definition of ASSERTIFCMP:
```
#define ASSERTIFCMP(C, X, OP, Y)					\
do {									\
	if (unlikely((C) && !((X) OP (Y)))) {				\
		printk(KERN_ERR "\n");					\
		printk(KERN_ERR "AFS: Assertion failed\n");		\
		printk(KERN_ERR "%lu " #OP " %lu is false\n",		\
		       (unsigned long)(X), (unsigned long)(Y));		\
		printk(KERN_ERR "0x%lx " #OP " 0x%lx is false\n",	\
		       (unsigned long)(X), (unsigned long)(Y));		\
		BUG();							\
	}								\
} while(0)
```
The rules of C's short-circuit evaluation of boolean operations (`&&` and `||`) force a left-to-right evaluation of terms in a boolean expression, with evaluation halting as soon as the overall result is known - i.e. stopping at the first false with `&&` and the first true with `||`. The above macro can therefore be rewritten like this without _any_ change of functionality:
```
#define ASSERTIFCMP(C, X, OP, Y)					\
do {									\
	if (unlikely((C)) {						\
		if (!((X) OP (Y)))) {					\
			...						\
			BUG();						\
		}							\
	}								\
} while(0)

```

JamesH65, 2018-06-27:
Closing this issue as questions answered/issue resolved.

closed 2018-06-27 by JamesH65