raspberrypi/linux #2599: Raspberry Pi 3B+ IPv6 cause alignment issues

lategoodbye, 2018-06-28:
Hi,
if i connect my Raspberry Pi 3B+ (running Raspbian with 4.14.52-v7+) to my home network, i see a slowly increasing number of alignment issues:

```
# cat /proc/cpu/alignment 
User:		0
System:		172 (__csum_ipv6_magic+0x1c/0xc8)
Skipped:	0
Half:		0
Word:		0
DWord:		0
Multi:		172
User faults:	2 (fixup)
```

As far as i remember this issue is new and didn't exist in the last Raspbian release from April.

wizofe, 2018-06-29:
Similarly here: https://gist.github.com/wizofe/469d8c4f092e5ae7d9adc9147f83e167 from the #2587

cyph84, 2018-06-30:
Same thing seems to happen on Rpi 3B on kernel 4.14.50-2-ARCH

lategoodbye, 2018-06-30:
@wizofe 
Please provide some context to trigger the issue.

@cyph84 
Do you refer to wizofe's issue or mine?

cyph84, 2018-06-30:
@lategoodbye I mean the issue by @wizofe

wizofe, 2018-06-30:
@lategoodbye There's a detailed description on how to replicate if you click on the issue but apparently, @timg236 already will submit a firmware patch :)

wizofe, 2018-06-30:
@cyph84 How do you are triggering the alignment mismatch? Any logs?

timg236, 2018-06-30:
@wizofe The root cause for bug#2587 is a memory trample from the high quality analog audio processing and would only occur if audio was playing. The kernel oops will be pretty random depending on what bit of memory was trampled i.e. I don't think it will fix this issue.

lategoodbye, 2018-07-03:
Here is a tcpdump when the alignment issue occur. At least one of those 3 ICMPv6 messages seems to trigger this issue. A simple ping6 doesn't trigger it.

[alignment_issue.zip](https://github.com/raspberrypi/linux/files/2160238/alignment_issue.zip)

JamesH65, 2018-07-03:
No expert, but reading this https://www.kernel.org/doc/Documentation/arm/mem_alignment, can you 
`echo 1 > /proc/cpu/alignment` to turn on a bit more tracking which should narrow down what is causing the traps?

lategoodbye, 2018-07-03:
This won't work, because a unaligned access would crash the kernel. Here the output of dmesg:
`alignment: Ignoring faults is unsafe on this CPU. Defaulting to fixup mode.`

lategoodbye, 2018-07-03:
Sorry, i didn't get any dump with last command.

JamesH65, 2018-07-03:
OK, no worries. But it kept going and continued to report alignment issues?

That kernel.org doc is about 17 years old, so perhaps not the most reliable source.

JamesH65, 2018-07-03:
I suspect it's here (__csum_ipv6_magic+0x1c/0xc8) anyway, as reported in the dump above. Perhaps it repots the last known location? Anyway, that's in arch/arm/lib/csumipv6.S, which is prototyped  as `__csum_ipv6_magic(const struct in6_addr *saddr, const struct in6_addr *daddr, __be32 len,
		__be32 proto, __wsum sum);`

So presumably something calling that is sending in badly aligned buffers.

lategoodbye, 2018-07-03:
Yes, the `System` counter increase by 4 every ~ 2 minutes. I retest it with a Raspberry Pi 3 B and it shows the same issue.

JamesH65, 2018-07-03:
Might be worth putting in a conditional dump_stack in those functions if one of the addresses is not aligned correctly (32bit I presume?)

pelwell, 2018-07-03:
a.k.a. `WARN_ON(<not aligned>)`

JamesH65, 2018-07-04:
Just did cat /proc/cpu/alignment on the Pi3B+ on my desk,clearly something is up. I'll put in the warn suggested and see if I can get things to backtrace.

```
User:           0
System:         1307359 (ip6_datagram_recv_common_ctl+0xc8/0xd4 [ipv6])
Skipped:        0
Half:           0
Word:           0
DWord:          0
Multi:          1307344
User faults:    2 (fixup)
```

6by9, 2018-07-04:
http://antirez.com/news/111 was written 495 days ago and is referencing alignment exceptions being logged in ip6_datagram_recv_common_ctl, so this doesn't appear to be a new thing.

lategoodbye, 2018-07-05:
Since __csum_ipv6_magic is in Assembler, i placed the warnings into the wrapper function csum_ipv6_magic.

Here is the dmesg:
[alignment-warns.log](https://github.com/raspberrypi/linux/files/2167672/alignment-warns.log)

JamesH65, 2018-07-05:
Nothing specific to Pi in that backtrace AFAICT, but I'll take a look tomorrow to see if there is anything obvious.

lategoodbye, 2018-07-06:
From the fact that smsc95xx and lan78xx are affected we cannot conclude that's a problem of the main network stack. AFAIK it's the drivers job to provide skbs in a properly aligned state. At least the stacktrace give us the information that the issue happens on the receiving side.

So the question is, why is only ICMPv6 affected?

JamesH65, 2018-07-06:
@lategoodbye Which particular structure/pointer did your warning fire on? Need that to try and track down which buffer might have been allocated with he wrong alignment, which is what I think might be the problem.

lategoodbye, 2018-07-06:
I will provide my patch when i'm at home (requires a nasty cast in order to avoid compile error).

Please look at the second line (line below WARNING: ...)
saddr unaligned -> const struct in6_addr *saddr is unaligned
daddr unaligned -> const struct in6_addr *daddr is unaligned

I didn't reference a member from these structures.

JamesH65, 2018-07-06:
Ah, sorry, hadn't seen that.  Thanks.

JamesH65, 2018-07-06:
Just a note as to progress. I suspect that in the skb that causes the problem, the network header offset is not divisible by 4, this is the offset from the head, of a structure that is dereferenced., hence the alignment error.  Now I nee to figure out how that header (which is a struct ipv6hdr) has been dropped in to the skb incorrectly aligned. This might be from the firmware, as I have currently no idea how it gets in there!

lategoodbye, 2018-07-06:
Here is my patch:

```
diff --git a/arch/arm/include/asm/checksum.h b/arch/arm/include/asm/checksum.h
index 20043e0..b344266 100644
--- a/arch/arm/include/asm/checksum.h
+++ b/arch/arm/include/asm/checksum.h
@@ -148,6 +148,9 @@ static inline __sum16
 csum_ipv6_magic(const struct in6_addr *saddr, const struct in6_addr *daddr,
                __u32 len, __u8 proto, __wsum sum)
 {
+       WARN_ONCE((__force u32)saddr & 0x3, "saddr unaligned");
+       WARN_ONCE((__force u32)daddr & 0x3, "daddr unaligned");
+
        return csum_fold(__csum_ipv6_magic(saddr, daddr, htonl(len),
                                           htonl(proto), sum));
 }
```

I also have a different ARMv7 board with a "real" Ethernet interface at home (NXP i.MX6ULL EVK 14x14), which has an old Kernel 4.1 in its BSP. This system shows the same alignment issues.

pelwell, 2018-07-06:
If you add an alignment check to `hdr` in `ipv6_rcv` you'll have access to `skb->head` and `skb->network_header`, and could even dump the entire packet on a failure.

JamesH65, 2018-07-06:
I did that this afternoon, head was a sane number but network_header was 0x5a, so head+network_header was not aligned to 4 bytes, and that is where the structure is extracted from. Didn't find how that was set up though, I suspect it might be coming from the firmware.

pelwell, 2018-07-06:
DId the data at those locations look plausible?

JamesH65, 2018-07-06:
Didn't get to that stage, sounds like Monday work to me. Spent quite some time simply reading through the code trying to figure out how it all bolts together - none of the stack traces go near the driver, so wanting to determine how the data was setting in to the skb in the first place. 

The two drivers, contrary to what I wrote above, are fairly dissimilar, so to  have the same 'fault' may simply be down to the data coming over the link not aligning.

pelwell, 2018-07-06:
softirqs and tasklets decouple the caller from the callee, making it harder to track the data flow, but look for calls to `netif_rx`.

lategoodbye, 2018-07-15:
Today i prepared the skb chain patch for upstream using current net-tree (4.18rc3) which still shows the issue, so i don't believe this is fixed in 4.14.54.

In order to reproduce this issue please use the command line tool efsu from the [open-plc-utils](https://github.com/qca/open-plc-utils/) on a PC which is connected to the Raspberry Pi:

Example to send a Multicast listener query (adjust network interface to your setup):
sudo ./efsu -i eth0 -h MLDv2.hex

Content of MLDv2.hex (taken from wireshark trace above):
```
33 33 00 00 00 01 70 4f 57 ed 6c 00 86 dd 60 00
00 00 00 24 00 01 fe 80 00 00 00 00 00 00 72 4f
57 ff fe ed 6c 00 ff 02 00 00 00 00 00 00 00 00
00 00 00 00 00 01 3a 00 05 02 00 00 01 00 82 00
21 5a 27 10 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 02 7d 00 00
```

JamesH65, 2018-07-16:
OK. Odd that it went away for me, I wonderif something in our network was changed and the offending packets are no longer present.

tilosp, 2018-07-16:
I can confirm that it is not fixed in `4.14.54`.

My RPi3B+ still has a increasing number of alignment issues:
```bash
$ cat /proc/cpu/alignment 
User:		0
System:		2689233 (ip6_datagram_recv_common_ctl+0xc8/0xd4 [ipv6])
Skipped:	0
Half:		0
Word:		0
DWord:		0
Multi:		2689221
User faults:	2 (fixup)
```

lategoodbye, 2018-07-16:
Can anyone confirm that my test scenario above (efsu) also cause the alignment issue?

JamesH65, 2018-07-16:
Anyway of easily installing that without checking out and building all that Atheros stuff?

Note, I am now seeing the issue again, even with the latest skb chain fix, so not sure what happened before that meant I got no alignment failures. Anyway, will start digging today since I can replicate.

lategoodbye, 2018-07-16:
I don't know a tool to generate MLDv2 packets, so i suggested open-plc-utils because they are small, fast to compile. Maybe there is a way to generate this with scapy, but i'm not that Python fan.

Those messages are usually generated by IPv6 routers.

pelwell, 2018-07-16:
The `efsu` command with the supplied .hex causes 4 additional system faults for each invocation, the last of which is at __csum_ipv6_magic+0x1c, so this is a useful test.

JamesH65, 2018-07-16:
Hmm, been looking at the source of the lan78xx driver. It appears to be very similar to that of the smsc75xx driver, with one major difference. smsc uses as its base driver the usbnet implementation, whilst lan78xx copies a lot of structures out of usbnet and uses them, which is a bit nasty. I have no idea what the reasoning is for that, perhaps usbnet was added after lan78xx to provide an abstration for usb network devices and lan78xx was never updated to use it, but it means that a couple of the structures and some bitfield definitons are actually different between usbnet and the lan78xx implementation. I have no idea whether that is relevent.  I'll have a bit of a git blame dig.

6by9, 2018-07-16:
I'd looked at that too. I think usbnet is missing a load of the PHY setup stuff that is required for gigabit.

JamesH65, 2019-01-21:
Nice work, and thanks for throwing it upstream. Hopefully there will be some response!

lategoodbye, 2019-01-21:
@yumkam Thanks for this, but please fix the following in your upstream patch:
- always send it with git-send. MIME mails with attachments will be ignored
- add your Signed-off-by
- add maintainer to CC (scripts/get_maintainer.pl your.patch)
- make sure the patch apply clean with current net tree

JamesH65, 2019-07-31:
Fixed by #2953

closed 2019-07-31 by JamesH65