raspberrypi/linux #2602: TCP slowdowns on rate-limited flows over to a Pi3B+

cyph84, 2018-07-01:
Yes, I know the title might be a bit duh-inducing, but please read on.

While upgrading a video-over-HTTP system from Pi3B to Pi3B+, I came to the bizarre conclusion that incoming TCP connections running at about 300 kilobytes/s will randomly but definitely stall. When this happens other TCP connections are perfectly fine.

Currently running a 4.14.50-2-ARCH kernel (equivalent to 3b01f059d2ef9)

I am hoping that someone can reproduce this.

1. Host a large ( > 20 minutes) video on any webserver reachable over ethernet.
2. From a PiB+ do:

`ffmpeg -re -i 'http://videoserver.local/test.mp4' -codec copy -y  -f mp4 /dev/null`

or just to be sure its not an ffmpeg bug:

`curl -v --limit-rate 100k   'http://videoserver.local/test.mp4' > /dev/null `

The ffmpeg -re in the above command will rate limit the reading of the stream to the video's bitrate.

What you should see if you can reproduce this behavior is the transfer running at a typical video rate of 200-300 kilobytes/s, then after a few minutes abruptly falling to < 4 kilobytes/s.

Stuff I've tried to no avail:
- Disable tcp SACK on the server side
- Disable all ethtool -K features
- Use iptables to clamp TCPMSS to 800bytes

Things that still work:
- running the webserver on localhost
- running the ffmpeg test on a file over NFS
- curl'ing the file without rate limit 
- running the above ffmpeg command without the `-re` flag
- running the same setup on a Pi3B

cyph84, 2018-07-03:
I "fixed" it by doing this:

`echo -n '4096 100000 100000' >  /proc/sys/net/ipv4/tcp_rmem`

But I have no idea why. And why this problem only happens on a Pi3B+

JamesH65, 2019-07-31:
"The first value tells the kernel the minimum receive buffer for each TCP connection, and this buffer is always allocated to a TCP socket, even under high pressure on the system. ... The second value specified tells the kernel the default receive buffer allocated for each TCP socket. This value overrides the /proc/sys/net/core/rmem_default value used by other protocols. ... The third and last value specified in this variable specifies the maximum receive buffer that can be allocated for a TCP socket."

closed 2019-07-31 by JamesH65