raspberrypi/linux #2603: Many items on applications menu do not open when running xrdp

MrYsLab, 2018-07-01:
Running the 2018-06-27 version of Raspbian and using xrdp, launching many of the applications from
the applications menu fails silently. For example, if I launch the SD Card Copier utility, no dialog boxes appear and nothing happens.

The Fix: If I change the menu entry from:
```
env SUDO_ASKPASS=/usr/lib/pwdpic.sh sudo -A dbus-lauinch piclone
```

To:
```
env SUDO_ASKPASS=/usr/lib/pwdpic.sh sudo -AE dbus-lauinch piclone
```
Then it launches properly. I needed to do this for several other items in the applications menu as well.

Perhaps all entries should be changed from sudo -A to sudo-AE.

JamesH65, 2019-07-31:
@spl237 @XECDesign Is this still an issue?

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

spl237, 2019-07-31:
We'll add the extra flag to the relevant applications in due course.

MrYsLab, 2019-07-31:
For Buster, the only item that I believe is non-functional without the change mentioned above is the sd card copier utility.  I have not tested extensively, so please test all the other items on the menu when you get to repairing this item. Thanks.