raspberrypi/linux #2610: fatal error: linux/compiler-gcc6.h: No such file or directory  #include gcc_header(__GNUC__)

allen807, 2018-07-08:
I cloned the latest master code and tried to compile it on my raspberry Pi 3B+. And got the following error message:

include/linux/compiler-gcc.h:94:30: fatal error: linux/compiler-gcc6.h: No such file or directory
 #include gcc_header(__GNUC__)

Here it is my linux version: 4.14.50-v7+
Here it is my gcc version: gcc (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516

pelwell, 2018-07-08:
What commands did you run to build from a clean git clone?

allen807, 2018-07-08:
I ran the following commands and then got the error message.
make defconfig
make V=1 all

allen807, 2018-07-08:
I followed this instruction:

cd linux
KERNEL=kernel7
make bcm2709_defconfig

and got another issue:
***
*** Can't find default configuration "arch/arm/configs/bcm2709_defconfig"!
***

I didn't find file"bcm2709_defconfig" from the source I cloned. The Raspberry pi I have is 3B+. I'm not sure if the instruction can be applied to this new version.

pelwell, 2018-07-08:
You've either cloned the wrong kernel or are in the wrong branch. What do the following commands return?
```
$ git remote -v
$ git branch
```

allen807, 2018-07-08:
see this:

pi@raspberrypi:~/Git/linux $ git remote -v
origin	git@github.com:raspberrypi/linux.git (fetch)
origin	git@github.com:raspberrypi/linux.git (push)
pi@raspberrypi:~/Git/linux $ git branch
* master

pelwell, 2018-07-08:
Run `git checkout rpi-4.14.y`, then start the build again.

allen807, 2018-07-08:
I get an extremely slow internet connection in China. This may take a whole night to download. I'll get you back tomorrow.

pelwell, 2018-07-08:
`bcm2709_defconfig` is still the right choice on a 3B+. We should mention it in the kernel building guide to avoid confusion.

allen807, 2018-07-09:
Everything is smooth now after using branch rpi-4.14.y and following the guide to building kernel. I think this issue can be closed now

closed 2018-07-09 by allen807