raspberrypi/linux #2611: add a configurable i2c address to an RTC?

shawaj, 2018-07-09:
On PiJuice (https://github.com/PiSupply/PiJuice) we are using an emulated ds1339 RTC running on the STM32 MCU.

However, to avoid hardware clashes we are offering the possibility for users to change the i2c address of the RTC.

We would like to add support for changing of the address using the overlay. Would this be acceptable if we submitted a PR?

See this issue for further discussion also - https://github.com/PiSupply/PiJuice/issues/166#issuecomment-403537810

Thanks

pelwell, 2018-07-09:
> Would this be acceptable if we submitted a PR?

Go for it.

shawaj, 2018-07-09:
Cool...will sort something and then post back here

pelwell, 2018-07-09:
A PR would be better.

tvoverbeek, 2018-07-09:
See PiSupply/PiJuice#166 comment by @hemingchen for a work-around using `new_device` in `/sys/bus/i2c/devices/i2c-1`.
Suppose you have the emulated ds1339 RTC at 0x65.
Then `echo ds1339 0x65 | sudo tee /sys/bus/i2c/devices/i2c-1/new_device` will load the rtc_ds1307 module and e.g. `sudo hwclock -r` works.
But something like the i2c-rtc overlay with an added address parameter override is preferable.

pelwell, 2018-07-09:
See 1fa2579.

tvoverbeek, 2018-07-10:
@pelwell Thanks for the quick implementaton.
How long does it take before this change shows up in rpi-update?
Days, Weeks? Thanks in advance.

pelwell, 2018-07-10:
It varies from hours to days, but is rarely longer than a week. In the meantime you can build your own:
1. Download the source file: https://raw.githubusercontent.com/raspberrypi/linux/1fa25792a2e83012e91de333034b9c9033adc1c3/arch/arm/boot/dts/overlays/i2c-rtc-overlay.dts
2. Compile it:
    ```
    dtc -@ -I dts -O dtb -o i2c-rtc.dtbo i2c-rtc-overlay.dts
    ```
    Don't worry about any warnings - the dtc utility isn't very overlay-aware.
3. Install it:
    ```
    sudo cp i2c-rtc.dtbo /boot/overlays
    ```

tvoverbeek, 2018-07-10:
@pelwell Thanks again. Just tried the build and it works as expected.
Will keep an eye on raspberrypi/linux/firmware to see when it shows up.

pelwell, 2018-07-12:
Last night's firmware release includes the patched i2c-rtc (and i2c-rtc-gpio) overlay(s).

tvoverbeek, 2018-07-12:
@pelwell Thanks Phil for the heads-up

shawaj, 2018-07-12:
Thanks @pelwell @tvoverbeek

pelwell, 2018-07-18:
@shawaj Please close if you feel the issue has been resolved.

closed 2018-07-18 by shawaj