raspberrypi/linux #2612: hcd_init_fiq: Possible sleeping in invalid context

lategoodbye, 2018-07-10:
Hi,
i enabled CONFIG_DEBUG_ATOMIC_SLEEP in the kernel config and then recompiled the current Raspberry Pi kernel 4.14.52. During the boot i get the following warning:

```
[    1.685905] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
[    1.916045] Core Release: 2.80a
[    1.916340] Setting default values for core params
[    1.916545] Finished setting default values for core params
[    2.118625] Using Buffer DMA mode
[    2.118826] Periodic Transfer Interrupt Enhancement - disabled
[    2.119042] Multiprocessor Interrupt Enhancement - disabled
[    2.119249] OTG VER PARAM: 0, OTG VER FLAG: 0
[    2.119499] Dedicated Tx FIFOs mode
[    2.121507] WARN::dwc_otg_hcd_init:1046: FIQ DMA bounce buffers: virt = 0xbad14000 dma = 0xfad14000 len=9024
[    2.121878] FIQ FSM acceleration enabled for :
[    2.121878] Non-periodic Split Transactions
[    2.121878] Periodic Split Transactions
[    2.121878] High-Speed Isochronous Endpoints
[    2.121878] Interrupt/Control Split Transaction hack enabled
[    2.123019] WARN::hcd_init_fiq:459: FIQ on core 1 at 0x80640d7c
[    2.123254] WARN::hcd_init_fiq:460: FIQ ASM at 0x806410dc length 36
[    2.123498] WARN::hcd_init_fiq:486: MPHI regs_base at 0xbb87e000
[    2.123760] BUG: sleeping function called from invalid context at kernel/locking/mutex.c:747
[    2.124061] in_atomic(): 1, irqs_disabled(): 128, pid: 0, name: swapper/1
[    2.124308] no locks held by swapper/1/0.
[    2.124470] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 4.14.52-v7+ #15
[    2.124704] Hardware name: BCM2835
[    2.124863] [<80110968>] (unwind_backtrace) from [<8010ca40>] (show_stack+0x20/0x24)
[    2.125153] [<8010ca40>] (show_stack) from [<807f8de8>] (dump_stack+0xcc/0x110)
[    2.125434] [<807f8de8>] (dump_stack) from [<8014b16c>] (___might_sleep+0x170/0x1a4)
[    2.125725] [<8014b16c>] (___might_sleep) from [<8014b214>] (__might_sleep+0x74/0xac)
[    2.126016] [<8014b214>] (__might_sleep) from [<80810530>] (__mutex_lock+0x40/0xa70)
[    2.126304] [<80810530>] (__mutex_lock) from [<80810f8c>] (mutex_lock_nested+0x2c/0x34)
[    2.134479] [<80810f8c>] (mutex_lock_nested) from [<8018b470>] (irq_find_matching_fwspec+0x38/0x124)
[    2.142750] [<8018b470>] (irq_find_matching_fwspec) from [<806b2080>] (of_irq_get+0x58/0x78)
[    2.151010] [<806b2080>] (of_irq_get) from [<8059a2f4>] (platform_get_irq+0x2c/0xf8)
[    2.159309] [<8059a2f4>] (platform_get_irq) from [<8063805c>] (hcd_init_fiq+0x20c/0x290)
[    2.167610] [<8063805c>] (hcd_init_fiq) from [<801b5444>] (flush_smp_call_function_queue+0x94/0x1ac)
[    2.176022] [<801b5444>] (flush_smp_call_function_queue) from [<801b6350>] (generic_smp_call_function_single_interrupt+0x1c/0x20)
[    2.184603] [<801b6350>] (generic_smp_call_function_single_interrupt) from [<8010ed8c>] (handle_IPI+0x27c/0x398)
[    2.193241] [<8010ed8c>] (handle_IPI) from [<80101520>] (bcm2836_arm_irqchip_handle_irq+0x80/0xb0)
[    2.201913] [<80101520>] (bcm2836_arm_irqchip_handle_irq) from [<808163bc>] (__irq_svc+0x5c/0x7c)
[    2.210685] Exception stack(0xb9daff30 to 0xb9daff78)
[    2.219456] ff20:                                     00000000 000003f4 396be000 00000000
[    2.228423] ff40: ffffe000 80d075cc 80d07568 80d987b9 80a628b4 00000001 00000000 b9daff8c
[    2.237370] ff60: b9daff80 b9daff80 80108c2c 80108c30 60000013 ffffffff
[    2.246285] [<808163bc>] (__irq_svc) from [<80108c30>] (arch_cpu_idle+0x34/0x4c)
[    2.255185] [<80108c30>] (arch_cpu_idle) from [<80815564>] (default_idle_call+0x34/0x48)
[    2.264163] [<80815564>] (default_idle_call) from [<8016aea4>] (do_idle+0xf0/0x158)
[    2.273155] [<8016aea4>] (do_idle) from [<8016b1c0>] (cpu_startup_entry+0x28/0x2c)
[    2.282171] [<8016b1c0>] (cpu_startup_entry) from [<8010e894>] (secondary_start_kernel+0x178/0x184)
[    2.291379] [<8010e894>] (secondary_start_kernel) from [<00101aac>] (0x101aac)
[    2.300580] dwc_otg 3f980000.usb: DWC OTG Controller
[    2.310256] dwc_otg 3f980000.usb: new USB bus registered, assigned bus number 1
[    2.319621] dwc_otg 3f980000.usb: irq 62, io mem 0x00000000
```

I suggest to call platform_get_irq in the hcd probe and store the irq in a common structure.

pelwell, 2019-05-24:
This should be fixed by https://github.com/raspberrypi/linux/commit/d903e18191d372cf917b0db788b5cd43bcd16025.

JamesH65, 2019-07-31:
Closing this issue as questions answered/issue resolved.

closed 2019-07-31 by JamesH65