raspberrypi/linux #2613: RFID-RC522 module usage instructions

pelwell, 2018-07-10:
A few years ago I wrote some tried and tested instructions for wiring up the RFID-RC522 module. installing the SPI-Py library and the MFRC-522 python library and test application. The thread where it appeared (https://github.com/raspberrypi/linux/issues/1547) is now so long that the relevant comment is hidden by default, and the whole thread takes ages to load.

This issue a repost of that comment.

========

Starting from a clean current Raspbian install:

1) Use the following wiring:

| RFID-RC522 Pin | Raspberry PI Pin | Raspberry PI Pin name | SPI1 Pin | SPI1 Pin name |
| --- | --- | --- | --- | --- |
| SDA/NSS | 24 | GPIO8 | 12 | GPIO18 |
| SCK | 23 | GPIO11 | 40 | GPIO21 |
| MOSI | 19 | GPIO10 | 38 | GPIO20 |
| MISO | 21 | GPIO9 | 35 | GPIO19 |
| IRQ | None | None | None | None |
| GND | 20 | GND | 34 | GND |
| RST | 22 | GPIO25 | 22 | GPIO25 |
| 3.3V/VCC | 17 | 3V3 | 17 | 3V3 |

Note that there are multiple versions of the module with different pin-outs - I've shown the pin names from two that I know.

2) Run `sudo raspi-config`, choose Advanced Options -> SPI -> Yes . You can reboot, but it isn't necessary (`dtoverlay -l` will show that there is a run-time overlay that enables the SPI interface).

3) Install git and python-dev:

```
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install python-dev git
```

4) Install the SPI-Py library:

```
git clone https://github.com/lthiery/SPI-Py.git
cd SPI-Py
sudo python setup.py install
cd ..
```

5) Install MFCR522-python:

```
cd ~
git clone https://github.com/mxgxw/MFRC522-python.git
cd MFRC522-python/
```

6) Run the script:

```
sudo python Read.py
```

If it still doesn't work, triple-check your wiring then upload a photo of it. Also make sure your power-supply is up to the task - 2.5A for a Pi3.

pelwell, 2018-07-10:
Closing to escape the attentions of the @JamesH65 bot.

closed 2018-07-10 by pelwell

JamesH65, 2018-07-10:
Too late...

Is this something that out to be in docs somewhere, or perhaps the projects section (not my domain)?

pelwell, 2018-07-11:
Is there any equivalent, application-specific documentation? It does seem like more of a project to me, but just having this short (but growing...) thread with the important information at the top should be sufficient.