raspberrypi/linux #2615: Panics in functions related to CPU accounting

maxnet, 2018-07-10:
I occasionally see kernel panics -hanging the system- in functions that seem to modify CPU accounting / load average data.

E.g. yesterday I had:

```
[  594.356235] Unable to handle kernel paging request at virtual address 03056564
[  594.363590] pgd = 80004000
[  594.366332] [03056564] *pgd=00000000
[  594.369966] Internal error: Oops: 5 [#1] SMP ARM
[  594.374650] Modules linked in: rfcomm cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic fuse sg i2c_dev ip_tables x_tables ipv6 brcmfmac brcmutil cfg80211 snd_bcm2835(C) joydev snd_pcm rfkill snd_timer snd uio_pdrv_genirq uio fixed
[  594.395973] CPU: 2 PID: 84 Comm: usb-storage Tainted: G         C      4.14.52v7-aufs #1
[  594.404180] Hardware name: BCM2835
[  594.407627] task: aeeb0f00 task.stack: b3044000
[  594.412233] PC is at dequeue_task_fair+0x14c/0xc44
[  594.417095] LR is at __update_load_avg_cfs_rq+0x16c/0x258
[  594.422567] pc : [<80152818>]    lr : [<8014e408>]    psr: 20000093
[  594.428924] sp : b3045c10  ip : 0305651c  fp : b1045c64
[  594.434220] r10: 00000001  r9 : 00000000  r8 : 00000009
[  594.439515] r7 : b5031d78  r6 : aeeb0f80  r5 : 0000008a  r4 : 00000001
[  594.446134] r3 : 00000001  r2 : 00000000  r1 : 0000b770  r0 : 00000001
[  594.452754] Flags: nzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  594.460076] Control: 10c5383d  Table: 13c1406a  DAC: 00000055
[  594.465902] Process usb-storage (pid: 84, stack limit = 0xb3044210)
[  594.472259] Stack: (0xb3045c10 to 0xb3046000)
[  594.476676] 5c00:                                     aeeb0f80 b1846400 b4acd000 b4acba00
[  594.484972] 5c20: b3045c64 b5031d78 00000009 80f94a3c b5031d40 80e92400 b3045c54 264aa53b
[  594.493264] 5c40: 00000003 264aa53b 00000003 aeeb0f00 b5031d40 00000009 b3045c94 b3045c68
[  594.501558] 5c60: 80148004 801526d8 00000000 00000000 80e9ad40 80f044c0 aeeb1364 b5031d40
[  594.509852] 5c80: aeeb0f00 80e9ad40 b3045cf4 b3045c98 80909c2c 80147f34 b22c4f88 aeee9430
[  594.518145] 5ca0: b3045d74 b3045cb0 806f97c8 80727284 b3045d1c 8090a198 34197000 aeeb1360
[  594.526439] 5cc0: aee2b108 00000000 b3045d9c b3044000 7fffffff 7fffffff 00000002 00000000
[  594.534735] 5ce0: b3044000 00000000 b3045d0c b3045cf8 8090a198 809097ec 00000000 aeee94dc
[  594.543031] 5d00: b3045d64 b3045d10 8090dcd4 8090a154 b3045d34 b3045d20 8090a198 809097ec
[  594.551327] 5d20: 00000000 b3045de8 aeee94e0 7fffffff aeee94e0 00000002 aeee94c4 aeee94dc
[  594.559619] 5d40: 7fffffff aeee94e0 00000002 00000000 b3044000 00000000 b3045dac b3045d68
[  594.567915] 5d60: 8090ae10 8090db14 b3045dbc 00000001 aeeb0f00 801493b0 aeee94e4 aeee94e4
[  594.576213] 5d80: aeee94c4 aeee94bc aeee94dc 00000000 00000013 aeee94c4 0000001f aeee9430
[  594.584508] 5da0: b3045dbc b3045db0 8090aed4 8090ad38 b3045dec b3045dc0 806fc978 8090aec0
[  594.592803] 5dc0: 0000001f aeee9430 aeee94bc aeee9430 aeee94bc 0001e000 c0010600 aeee9454
[  594.601096] 5de0: b3045e24 b3045df0 80734308 806fc850 9f95d200 00000013 00000000 01400000
[  594.609389] 5e00: c0010600 9f9586d0 aeee9430 9f9586d0 b591824f 0001e000 b3045e4c b3045e28
[  594.617683] 5e20: 807343c4 8073428c 0001e000 b3045e34 aeee9430 9f9586d0 b591824f b5918240
[  594.625975] 5e40: b3045e84 b3045e50 807344f0 80734374 00000000 b3045e60 8090a198 809097ec
[  594.634270] 5e60: 9f9586d0 00000000 aeee94f8 80f02d00 00000000 00000000 b3045f14 b3045e88
[  594.642565] 5e80: 80734dc0 807343d8 801a25f4 8010e8d4 aeee94fc 7fffffff aeee94fc 00000001
[  594.650860] 5ea0: 00000001 b3044000 b3045ecc aeee94f8 7fffffff aeee94fc 00000001 00000001
[  594.659155] 5ec0: b3045f14 b3045ed0 8090ae84 801edfa0 00000001 00000001 aeeb0f00 801493b0
[  594.667449] 5ee0: 00000100 00000200 aeee9018 aeee9430 00000000 aeee94f8 80f02d00 00000000
[  594.675744] 5f00: 00000000 9f9586d0 b3045f24 b3045f18 80733aac 80734d9c b3045f7c b3045f28
[  594.684038] 5f20: 80736324 80733aa0 34197000 aeeb1360 b336a09c 00000000 b3045f5c b3044000
[  594.692335] 5f40: 00000000 b336a040 b336a09c aeee9430 8073618c b336a080 00000000 b336a040
[  594.700627] 5f60: b336a09c aeee9430 8073618c b4a81a50 b3045fac b3045f80 8013e0e0 80736198
[  594.708923] 5f80: ffffffff b336a040 8013dfa4 00000000 00000000 00000000 00000000 00000000
[  594.717219] 5fa0: 00000000 b3045fb0 8010894c 8013dfb0 00000000 00000000 00000000 00000000
[  594.725513] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  594.733809] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[  594.742109] Code: e1a04000 f57ff05a e51bc040 e19aa000 (e1cc24d8) 
[  594.748297] ---[ end trace fa5488ba26256917 ]---
[  600.089709] lan78xx 1-1.1.1:1.0 eth0: Failed to read stat ret = 0xffffff92
```

Today I have:

```
pi@raspberrypi:~$ time sync
[11704.656912] Unable to handle kernel paging request at virtual address 343ac98e
[11704.656915] Unable to handle kernel paging request at virtual address 3437998e
[11704.656918] pgd = b02fc000
[11704.656925] [3437998e] *pgd=00000000
[11704.656935] Internal error: Oops: 1 [#1] SMP ARM
[... 115 lines cut ...]
[11705.550193] [<8014e9d0>] (update_curr) from [<801529d0>] (task_tick_fair+0x3e0/0x538)
[11705.558135] [<801529d0>] (task_tick_fair) from [<801497d8>] (scheduler_tick+0x74/0xe8)
[11705.566165] [<801497d8>] (scheduler_tick) from [<8018ad30>] (update_process_times+0x60/0x6c)
[11705.574724] [<8018ad30>] (update_process_times) from [<8019c698>] (tick_sched_handle+0x64/0x70)
[11705.583546] [<8019c698>] (tick_sched_handle) from [<8019c8cc>] (tick_sched_timer+0x50/0xa0)
[11705.592016] [<8019c8cc>] (tick_sched_timer) from [<8018b69c>] (__hrtimer_run_queues+0x160/0x30c)
[11705.600926] [<8018b69c>] (__hrtimer_run_queues) from [<8018c058>] (hrtimer_interrupt+0xc8/0x218)
[11705.609836] [<8018c058>] (hrtimer_interrupt) from [<80777ef0>] (arch_timer_handler_phys+0x38/0x40)
[11705.618924] [<80777ef0>] (arch_timer_handler_phys) from [<80179e08>] (handle_percpu_devid_irq+0x88/0x228)
[11705.628627] [<80179e08>] (handle_percpu_devid_irq) from [<801747e8>] (generic_handle_irq+0x34/0x44)
[11705.637803] [<801747e8>] (generic_handle_irq) from [<80174e34>] (__handle_domain_irq+0x6c/0xc0)
[11705.646624] [<80174e34>] (__handle_domain_irq) from [<80101504>] (bcm2836_arm_irqchip_handle_irq+0xa8/0xac)
[11705.656504] [<80101504>] (bcm2836_arm_irqchip_handle_irq) from [<808fb20c>] (__irq_usr+0x4c/0x60)
[11705.665499] Exception stack(0x95621fb0 to 0x95621ff8)
[11705.670617] 1fa0:                                     00000000 76882804 feffffff feffffff
[11705.678910] 1fc0: 76508338 76a07b3c 000614cf 00185338 0013fe01 00023260 00023248 00023258
[11705.687201] 1fe0: 00000000 7ef8fe50 0001222c 000113e4 20000010 ffffffff
[11705.693907] Code: e1a05185 e59c1068 ee1d4f90 e0844001 (e18400d5) 
[11705.700082] ---[ end trace 58efcdea07c985b1 ]---
[11705.719655] SMP: failed to stop secondary CPUs
[11705.724164] ---[ end Kernel panic - not syncing: Fatal exception in interrupt
```

Interesting about the last one is that cpuacct_charge() seems to be running simultaneously on 2 cores at the same time.
Exact line: https://github.com/raspberrypi/linux/blob/rpi-4.14.y/kernel/sched/cpuacct.c#L361

lategoodbye, 2018-07-11:
Please describe your hardware setup? I assume it's a Raspberry Pi 3B+.
Is it only reproducible there?
Do you know a kernel version which wasn't affected?

maxnet, 2018-07-11:
It's a 3B+, official 2.5A power supply, and the panics mentioned were with a WD Pidrive USB HDD attached, but do not know if that matters.

Problem is that it is not easy to reproduce.
And while I have seen hangs in previous 4.14.x kernel versions of as well, it's hard to tell if they were related to this, or say the lan78xx bug. I do not always have serial console attached, and you do not get to see the panic message without it, system just hangs.

lategoodbye, 2018-07-16:
Since i never seen this issue, it would be nice if you could narrow down this:
Is only Raspberry Pi 3B+ affected?
Does the issue "require" a network connection?
Does the issue "require" the connection to your WD Pidriver?
Is there a specific task which could trigger this issue?

maxnet, 2018-07-16:
Problem is that it is not so reproducible.
Haven't seen it again since last post.

Do need to seem to do something that occupies CPU to trigger it, be it compiling something large or running stress programs like memtester.
Never seen it when system was idle.
Have the feeling hangs occurred more often in 4.14.x than it did with 4.9.x, but have no objective data to back that up (did not count occurrences)
And have not used anything but a 3B+ since 4.14.x
Hard to tell if USB drive makes a difference either.

JamesH65, 2019-07-31:
@maxnet Has this happened recently on recent kernels?

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

maxnet, 2019-07-31:
Have not seen it recently. Will reopen if it returns.

closed 2019-07-31 by maxnet