raspberrypi/linux #2616: Init_resize.sh Not tainted 4.4.38+ #938. Associated with kernel panic

mharris3960, 2018-07-11:
(Title), Idk what is happening. I reinstalled debian on my sdcard and then it booted up like this. (Image link coming soon)

mharris3960, 2018-07-11:
The kernel panic is here: Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000000

mharris3960, 2018-07-11:
[image]

mharris3960, 2018-07-11:
Ok, I have flashed it, I will insert it and boot it up

mharris3960, 2018-07-11:
It still gives the same error. :(

pelwell, 2018-07-12:
You may also want to verify that the SD card has the capacity it claims to have. See this article for details of some tools for detecting fake cards: https://www.raymond.cc/blog/test-and-detect-fake-or-counterfeit-usb-flash-drives-bought-from-ebay-with-h2testw/

JamesH65, 2018-08-31:
@mharris3960 Did you manage to try a recent Raspbian install?

mharris3960, 2018-09-03:
Yeah, I think it might be the pi, so i'm getting a new one now, thanks
anyway!

JamesH65, 2019-07-31:
@mharris3960 Did the new Pi fix the issue?

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

closed 2019-09-18 by JamesH65