raspberrypi/linux #2619: 802.11ac 5Ghz wifi AP in RPI 3 B+

hamedafra, 2018-07-14:
Hi,
Does rpi 3 b+ supports 802.11ac 5Ghz in access point mode?
This is my hostapd.conf
```
ssid=ap
wpa_passphrase=password

country_code=GB
# ieee80211d=1
# ieee80211h=1

interface=wlan0
driver=nl80211

wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP

logger_syslog=0
logger_syslog_level=4
logger_stdout=-1
logger_stdout_level=0

hw_mode=a
ieee80211n=1
require_ht=1
ieee80211ac=1
require_vht=1

vht_oper_chwidth=1
channel=36
vht_oper_centr_freq_seg0_idx=42
```
And this the hostapd -dd result:

```
hostapd -dd /etc/hostapd/hostapd.conf
random: Trying to read entropy from /dev/random
Configuration file: /etc/hostapd/hostapd.conf
nl80211: TDLS supported
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:6
nl80211: Using driver-based off-channel TX
nl80211: Supported vendor command: vendor_id=0x1018 subcmd=1
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: Enable multi-channel concurrent (driver advertised support)
nl80211: use P2P_DEVICE support
nl80211: interface wlan0 in phy phy0
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Setup AP(wlan0) - device_ap_sme=0 use_monitor=1
nl80211: Create interface iftype 6 (MONITOR)
Failed to create interface mon.wlan0: -95 (Operation not supported)
nl80211: Driver does not support monitor interface type - try to run without it
nl80211: Enable Probe Request reporting nl_preq=0xaa5310
nl80211: Register frame type=0x40 (WLAN_FC_STYPE_PROBE_REQ) nl_handle=0xaa5310 match=
nl80211: Register frame command failed (type=64): ret=-22 (Invalid argument)
nl80211: Register frame match - hexdump(len=0): [NULL]
nl80211: Failed to enable Probe Request frame reporting in AP mode
rfkill: initial event: idx=1 type=1 op=0 soft=0 hard=0
nl80211: Add own interface ifindex 3 (ifidx_reason -1)
nl80211: if_indices[16]: 3(-1)
phy: phy0
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Previous country code GB, new country code GB 
nl80211: Regulatory information - country=GB (DFS-ETSI)
nl80211: 2402-2482 @ 40 MHz 20 mBm
nl80211: 5170-5250 @ 80 MHz 20 mBm
nl80211: 5250-5330 @ 80 MHz 20 mBm (DFS)
nl80211: 5490-5710 @ 160 MHz 27 mBm (DFS)
nl80211: 57000-66000 @ 2160 MHz 40 mBm
nl80211: Added 802.11b mode based on 802.11g information
Allowed channel: mode=1 chan=1 freq=2412 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=2 freq=2417 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=3 freq=2422 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=4 freq=2427 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=5 freq=2432 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=6 freq=2437 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=7 freq=2442 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=8 freq=2447 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=9 freq=2452 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=10 freq=2457 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=11 freq=2462 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=12 freq=2467 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=13 freq=2472 MHz max_tx_power=20 dBm
Allowed channel: mode=2 chan=36 freq=5180 MHz max_tx_power=20 dBm
Allowed channel: mode=2 chan=40 freq=5200 MHz max_tx_power=20 dBm
Allowed channel: mode=2 chan=44 freq=5220 MHz max_tx_power=20 dBm
Allowed channel: mode=2 chan=48 freq=5240 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=1 freq=2412 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=2 freq=2417 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=3 freq=2422 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=4 freq=2427 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=5 freq=2432 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=6 freq=2437 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=7 freq=2442 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=8 freq=2447 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=9 freq=2452 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=10 freq=2457 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=11 freq=2462 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=12 freq=2467 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=13 freq=2472 MHz max_tx_power=20 dBm
hw vht capab: 0x1020, conf vht capab: 0x0
Completing interface initialization
Mode: IEEE 802.11a  Channel: 36  Frequency: 5180 MHz
DFS 0 channels required radar detection
Could not set channel for kernel driver
Interface initialization failed
wlan0: interface state COUNTRY_UPDATE->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
hostapd_interface_deinit_free(0xaa2a58)
hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
hostapd_interface_deinit(0xaa2a58)
wlan0: interface state DISABLED->DISABLED
hostapd_bss_deinit: deinit bss wlan0
wlan0: Deauthenticate all stations
nl80211: sta_remove -> DEL_STATION wlan0 ff:ff:ff:ff:ff:ff --> 0 (Success)
wlan0: AP-DISABLED 
hostapd_cleanup(hapd=0xaa3710 (wlan0))
hostapd_free_hapd_data: Interface wlan0 wasn't started
hostapd_interface_deinit_free: driver=0x51a640 drv_priv=0xaa4b80 -> hapd_deinit
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
nl80211: Remove beacon (ifindex=3)
netlink: Operstate: ifindex=3 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 3 iftype 2 (STATION)
nl80211: Teardown AP(wlan0) - device_ap_sme=1 use_monitor=1
hostapd_interface_free(0xaa2a58)
hostapd_interface_free: free hapd 0xaa3710
hostapd_cleanup_iface(0xaa2a58)
hostapd_cleanup_iface_partial(0xaa2a58)
hostapd_cleanup_iface: free iface=0xaa2a58
```

asavah, 2018-08-06:
@urrohitsw if you run `iw list` on raspbian on rpi 3b**+** you will see that it does support 5GHz and AC.
Hence it's useful to read the topic title before giving a wrong answer.

@hamedafra try changing your country to US, it should work with your config, here is my config that does work:
```
ssid=PI3AC
wpa_passphrase=test12344321tset

country_code=US

interface=wlan0
driver=nl80211

wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP

macaddr_acl=0

logger_syslog=0
logger_syslog_level=4
logger_stdout=-1
logger_stdout_level=0

hw_mode=a
wmm_enabled=1

# N
ieee80211n=1
require_ht=1
ht_capab=[MAX-AMSDU-3839][HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]

# AC
ieee80211ac=1
require_vht=1
ieee80211d=0
ieee80211h=0
vht_capab=[MAX-AMSDU-3839][SHORT-GI-80]
vht_oper_chwidth=1
channel=36
vht_oper_centr_freq_seg0_idx=42
```

@JamesH65 can we have this looked at?
Also it would be nice to have some documentation and answers to the following questions:
Why AC AP does not work if country is set to any european country (GB,DE,FI do not work, tested)
```
nl80211: Failed to set channel (freq=5180): -22 (Invalid argument)
```
while it does work with US countrycode? Is this a bug? The frequencies (channels) used in the config I posted are allowed in ETSI zone and do not overlap with DFS channels.

Where does regulatory info for brcmfmac comes from? It certainly is not using crda, is it embedded in the brcmfmac firmware blob?

JamesH65, 2018-08-06:
We are all a bit busy on other stuff at the moment, with numerous people on holiday, so no promises on timescales. 
I have very little idea on country code stuff, but I do seem to remember that its built into the blob for the 5G stuff. The whole subject is a bit of a nightmare for us TBH, the regulations are horrible when you ship one product with 5G stuff worldwide. AIUI, most people have region specific builds.

deblauwetom, 2018-10-25:
I had an issue with AP too in the RPI3 B+ with the latest tagged kernel 4.14.71. And setting country_code=US explicitly seems to be better now.

See also here for more info what I saw in dmesg: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=225414&p=1384182#p1384182

puccaso, 2018-10-29:
im having the same issue here pi 3b+ bought in the UK, 
get no 5ghz AP's on my list.. 

i cant even get it to work as a client :/

jezcaudle, 2018-11-12:
I have the same problem. Kernel 4.14.71-v7+
Linux raspi-ap 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l GNU/Linux

I can get "n" working in 5Ghz with the country_code=GB but not "ac".

```
sudo iw reg set US
iw phy phy0 channels
```

and in the 5Ghz range I can see:

> Channel widths: 20MHz HT40+ VHT80

```
sudo iw reg set GB
iw phy phy0 channels
```

> Channel widths: 20MHz HT40-

Which is why I can get "n" but not "ac".

I ran 

`modinfo brcmfmac`

and saw:

> parm:           roamoff:Do not use internal roaming engine (int)

`sudo systool -v -m brcmfmac`

gives:

> roamoff             = "1"

I've tried setting this to 0:

```
more /etc/modprobe.d/brcmfmac.conf
options brcmfmac roamoff=0
```
It didn't make any difference. Then I realised the double negative of roamoff and it should be switched off so I removed the line and restored the default. Would really like to be able the AP as "ac" with country set to GB - want to use the AP's at a school, putting one in each classroom - and possibly add cameras, temperature sensors etc etc.

Thanks :-)

dvdesolve, 2019-02-10:
Still can't get 802.11ac mode working

`/etc/hostapd/hostapd.conf`
```
interface=wlan0
driver=nl80211
ssid=****

# 2.4 GHz settings
#hw_mode=g
#channel=1
#country_code=RU

# 5.0 GHz settings
hw_mode=a
channel=36
country_code=US

# N
ieee80211n=1
require_ht=1
ht_capab=[HT40+][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-3839][DSSS_CCK-40]

# AC
ieee80211ac=1
require_vht=1
ieee80211d=0
ieee80211h=0
vht_capab=[MAX-MPDU-3895][SHORT-GI-80][SU-BEAMFORMEE]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42

#wme_enabled=1
wmm_enabled=1
max_num_sta=2

wpa=2
auth_algs=1

rsn_pairwise=CCMP
wpa_key_mgmt=WPA-PSK
wpa_passphrase=****

logger_stdout=-1
logger_stdout_level=4
```

`$ iw list`
```
Wiphy phy0
	max # scan SSIDs: 10
	max scan IEs length: 2048 bytes
	max # sched scan SSIDs: 16
	max # match sets: 16
	max # scan plans: 1
	max scan plan interval: 508
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CMAC (00-0f-ac:6)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
[... 152 lines cut ...]
	software interface modes (can always be added):
	valid interface combinations:
		 * #{ managed } <= 1, #{ P2P-device } <= 1, #{ P2P-client, P2P-GO } <= 1,
		   total <= 3, #channels <= 2
		 * #{ managed } <= 1, #{ AP } <= 1, #{ P2P-client } <= 1, #{ P2P-device } <= 1,
		   total <= 4, #channels <= 1
	Device supports scan flush.
	Device supports randomizing MAC-addr in sched scans.
	Supported extended features:
		* [ 4WAY_HANDSHAKE_STA_PSK ]: 4-way handshake with PSK in station mode
		* [ 4WAY_HANDSHAKE_STA_1X ]: 4-way handshake with 802.1X in station mode
```

Log of hostapd start
```
Feb 11 01:20:03 desolve-rpi3bplus hostapd[783]: wlan0: AP-ENABLED
Feb 11 01:20:03 desolve-rpi3bplus hostapd[783]: wlan0: interface state HT_SCAN->ENABLED
Feb 11 01:20:03 desolve-rpi3bplus hostapd[783]: Using interface wlan0 with hwaddr b8:27:eb:c5:31:79 and ssid "eridanus"
Feb 11 01:20:03 desolve-rpi3bplus hostapd[783]: wlan0: Could not connect to kernel driver
Feb 11 01:20:03 desolve-rpi3bplus hostapd[783]: wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Feb 11 01:20:03 desolve-rpi3bplus hostapd[783]: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Feb 11 01:20:03 desolve-rpi3bplus hostapd[783]: Failed to create interface mon.wlan0: -95 (Operation not supported)
Feb 11 01:20:03 desolve-rpi3bplus hostapd[783]: Configuration file: /etc/hostapd/hostapd.conf
```
Xiaomi Redmi 5 Plus and nettop with Intel Wireless-AC 7265 tells me that connection speed is only 150 Mbps instead of 300. Any ideas?

pelwell, 2019-02-11:
The RPi 5G WiFi configuration does not permit AP mode. As I understand it, shipping an "open" (i.e. not a locked-down appliance) AP-capable device is a compliance nightmare that would require RPis to be region-locked.

asavah, 2019-02-11:
@pelwell wrong, it does work.
See my config which works in AC AP mode like a charm 
https://github.com/raspberrypi/linux/issues/2619#issuecomment-410703338

dvdesolve, 2019-02-11:
How can I check which mode is used by hostapd?

emiliogq, 2019-02-19:
Hi guys, I can confirm that @asavah was right, I copied (Ctrl+C and Ctrl+V) his/her config and it works absolutely. I had the problem that the bandwith was too slow but when I had tried to set up 5G Wi-Fi it didn't appear at all.

So, once I rebooted my Pi 3 B+ I did a speed test and now it reaches 60Mb and before only 10Mb
https://www.speedtest.net/result/8053964661

So many thanks @asavah

JamesH65, 2019-07-31:
Closing this issue as questions answered/issue resolved.

closed 2019-07-31 by JamesH65

theschles, 2019-08-01:
Hi all,

I am able to get my hostapd on my RPi 3 B+ to work with 802.11n.

However, I am running into trouble trying to get 802.11ac working on it.  

I ran:
```
sudo iw reg set US
iw phy phy0 channels
```
And got the expected list of the channels from 1-11, 36-48, and 149-165 (additional channels were listed as "disabled")

I then copied @asavah 's hostapd.conf file as listed in https://github.com/raspberrypi/linux/issues/2619#issuecomment-410703338 and ran `hostapd -dd hostapd.conf`

Result:

```
random: getrandom() support available
Configuration file: hostapd-ac.conf
nl80211: Using driver-based roaming
nl80211: TDLS supported
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:6
nl80211: Using driver-based off-channel TX
nl80211: Supported vendor command: vendor_id=0x1018 subcmd=1
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: Enable multi-channel concurrent (driver advertised support)
nl80211: use P2P_DEVICE support
nl80211: interface wlan0 in phy phy0
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Failed to set interface 3 to mode 3: -1 (Operation not permitted)
nl80211: Try mode change after setting interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
nl80211: Interface mode change to 3 from 0 failed
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
netlink: Operstate: ifindex=3 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 3 iftype 2 (STATION)
nl80211: Failed to set interface 3 to mode 2: -1 (Operation not permitted)
nl80211: Interface mode change to 3 from 0 failed
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
netlink: Operstate: ifindex=3 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 3 iftype 2 (STATION)
nl80211: Failed to set interface 3 to mode 2: -1 (Operation not permitted)
nl80211 driver initialization failed.
hostapd_interface_deinit_free(0xdefc20)
hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
hostapd_interface_deinit(0xdefc20)
wlan0: interface state UNINITIALIZED->DISABLED
hostapd_bss_deinit: deinit bss wlan0
wlan0: AP-DISABLED
hostapd_cleanup(hapd=0xdf0980 (wlan0))
wlan0: CTRL-EVENT-TERMINATING
hostapd_free_hapd_data: Interface wlan0 wasn't started
hostapd_interface_deinit_free: driver=(nil) drv_priv=(nil) -> hapd_deinit
hostapd_interface_free(0xdefc20)
hostapd_interface_free: free hapd 0xdf0980
hostapd_cleanup_iface(0xdefc20)
hostapd_cleanup_iface_partial(0xdefc20)
hostapd_cleanup_iface: free iface=0xdefc20
```

I'm running Raspbian Buster:
```
$ uname -a
Linux raspberrypi 4.19.58-v7+ #1245 SMP Fri Jul 12 17:25:51 BST 2019 armv7l GNU/Linux
$ dmesg | grep Rasp
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
```

Help?

focher, 2019-08-10:
You need to run with sudo, so try:

`sudo hostapd -d hostapd.conf`

djeedai, 2020-02-10:
@JamesH65 I don't believe this is fixed. Reading the comments above, @asavah proposed a working solution for US users only. However European users still have an issue, since like @jezcaudle mentioned setting any EU country code disables VHT80, which prevents 802.11ac from working. I don't know know if the issue lies with this repository or somewhere else, but I feel very uncomfortable running a device from inside EU with a country code set to US, for obvious legal reasons. And VHT80 was approved by all EU countries AFAIK so there is no reason to limit it.

```sh
# iw reg set US
# iw phy phy0 channels
[...]
Band 2:
	* 5170 MHz [34] (disabled)
	* 5180 MHz [36] 
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40+ VHT80
	* 5190 MHz [38] (disabled)
	* 5200 MHz [40] 
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40- VHT80
	* 5210 MHz [42] (disabled)
	* 5220 MHz [44] 
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40+ VHT80
	* 5230 MHz [46] (disabled)
	* 5240 MHz [48] 
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40- VHT80
[...]
# iw reg set IE
# iw phy phy0 channels
[...]
Band 2:
	* 5170 MHz [34] (disabled)
	* 5180 MHz [36] 
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40+
	* 5190 MHz [38] (disabled)
	* 5200 MHz [40] 
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40-
	* 5210 MHz [42] (disabled)
	* 5220 MHz [44] 
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40+
	* 5230 MHz [46] (disabled)
	* 5240 MHz [48] 
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40-
[...]
```

Note how VHT80 is present for the 36/40/44/48 channels with the US country code but disappear with the IE one. Do you know where this difference comes from?

Thanks

djeedai, 2020-02-10:
Additional note : the largest Irish broadband provider supplies a modem with 802.11ac / VHT80 active by default, which adds to the argument that there is no legal basis for VHT80 not to be present in the list of channel widths with a country code of IE.

AleXSR700, 2020-08-03:
> @urrohitsw if you run `iw list` on raspbian on rpi 3b**+** you will see that it does support 5GHz and AC.
> Hence it's useful to read the topic title before giving a wrong answer.
>
> @hamedafra try [...]

Hi @asavah ,
I am trying to get my AP up and running on my Pi 4 Model B with 8 GB of RAM and have been trying all kinds of stuff.
But no matter what I do, as soon as I turn on VHT the AP is no longer started by the system. So I can only run it with 20 MHz.

Do you have any idea what could be the issue here? Latest Raspberry OS, newest Pi. So it should all work, I would think.
Here is my  /etc/hostapd/hostapd.conf
`
country_code=US
interface=wlan0
bridge=br0
ssid=TestAP
wpa_passphrase=TestPass
wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP

hw_mode=a
macaddr_acl=0
ignore_broadcast_ssid=0
wmm_enabled=1

# AC
ieee80211ac=1
require_vht=1
ieee80211d=0
ieee80211h=0
#vht_capab=[MAX-MPDU-3895][SHORT-GI-80][SU-BEAMFORMEE]
#vht_oper_chwidth=1
channel=36
#vht_oper_centr_freq_seg0_idx=42
`
If I enable any of the three commented-out VHT features the AP will not start.

tomty89, 2020-08-14:
@asavah Does enabling ac/vht/80mhz actually make it work faster for you though? Coz that doesn't happen to me. Same case when I use my Pi 4's chip as a client, `country=US` make it work with ac-only AP (my router), but it's not faster either. (I *was* talking about my 3B+ for the AP case though.)

asavah, 2020-08-14:
@tomty89 my pi4 works as ac-only ap to serve only my phone because my main router is far away and with wall in between.
I don't have another ac capable device handy to do some proper testing.
My performance figures might be irrelevant.
Pi wifi chips are 1T1R toys with halfassed cypress firmware anyway.

preciouswater, 2020-08-31:
@AleXSR700 
I have been trying different combinations today too, it seems to be that the maximum channel width I can set is 20MHz as well, I am not sure whether that is a hardware limit or not.
Looking at 'iw list' on my RPI4B, I don't know how to translate / interpret

```
Supported Channel Width: neither 160 nor 80+80 
```

For your config, if you set vht_oper_chwidth=0, you can uncomment vht_capab, and vht_oper_centr_freq_se.... and it should work. Good luck

PierreGCD, 2020-09-14:
@asavah Thank you for your hostapd.conf it is very helpfull.
I am using RaspAP on my pi 3b+ and I got the 5Ghz working thanks to you. However, I wonder if it is possible to get the 802.11ac 5Ghz working in bridge mode (instead of AP)? I tried couple of time and it was not working. Is there a specific parameter I have to enter in the .conf file ? 
Thanks !

rand0mbyte, 2022-04-21:
> The RPi 5G WiFi configuration does not permit AP mode. As I understand it, shipping an "open" (i.e. not a locked-down appliance) AP-capable device is a compliance nightmare that would require RPis to be region-locked.

That's not true, I use my RPi 5G WiFi as an AP without any problem.

Here my hostapd for your reference (use channel 159 as in my config and will work):

`ssid=TPLINK
wpa_passphrase=@rRAQQ@77RAMba19@#GHiSM0##733$$

country_code=BZ

interface=wlan0
driver=nl80211

wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP

macaddr_acl=0

logger_syslog=0
logger_syslog_level=4
logger_stdout=-1
logger_stdout_level=0

hw_mode=a
wmm_enabled=1

# N
ieee80211n=1
require_ht=1
ht_capab=[MAX-AMSDU-3839][HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]

# AC
ieee80211ac=1
require_vht=1
ieee80211d=0
ieee80211h=0
vht_capab=[MAX-AMSDU-3839][SHORT-GI-80]
vht_oper_chwidth=1
channel=153
vht_oper_centr_freq_seg0_idx=42`

Now you have your 5G AP on your Rpi 4. Enjoy!

mmhorda, 2022-07-26:
> Now you have your 5G AP on your Rpi 4. Enjoy!

unfortunately your config is not quite AC just yet. I am struggling myself to make it work. If you check connection type and check the speeds - it is quite slow AC then. I know that connection can be done on 433mbuts and do 100mbits upload and download. Your config gives just half of that speed. :(