raspberrypi/linux #2623: Goodie 911 driver - high cpu.

AndrewCapon, 2018-07-23:
Hi Guys,

I have tested this on a Pi Zero and Model B with a HyperPixel 4.0 screen which uses a goodix 911 touchscreen.

Here is the goodix driver:

pi@PiDev:~ $ modinfo goodix
filename:       /lib/modules/4.14.52-v7+/kernel/drivers/input/touchscreen/goodix.ko
license:        GPL v2
description:    Goodix touchscreen driver
author:         Bastien Nocera <hadess@hadess.net>
author:         Benjamin Tissoires <benjamin.tissoires@gmail.com>
srcversion:     35F207673154C81B3E8EBA5
alias:          i2c:GDIX1001:00
alias:          of:N*T*Cgoodix,gt967C*
alias:          of:N*T*Cgoodix,gt967
alias:          of:N*T*Cgoodix,gt928C*
alias:          of:N*T*Cgoodix,gt928
alias:          of:N*T*Cgoodix,gt9271C*
alias:          of:N*T*Cgoodix,gt9271
alias:          of:N*T*Cgoodix,gt927C*
alias:          of:N*T*Cgoodix,gt927
alias:          of:N*T*Cgoodix,gt912C*
alias:          of:N*T*Cgoodix,gt912
alias:          of:N*T*Cgoodix,gt9110C*
alias:          of:N*T*Cgoodix,gt9110
alias:          of:N*T*Cgoodix,gt911C*
alias:          of:N*T*Cgoodix,gt911
depends:        
intree:         Y
name:           goodix
vermagic:       4.14.52-v7+ SMP mod_unload modversions ARMv7 p2v8 

When using the touchscreen this is causing high cpu load (between 20% and 70%) in its ISR (irq/169-gt911).

It is fairly easy to replicate with the HyperPixel 4.0, run top in a terminal and start moving your finger around on the screen, to see the cpu load get higher move your finger around quicker, now use 2 fingers, then 3 fingers.

Anyone any idea why this might be?

lategoodbye, 2018-07-23:
Did you tried to contact the authors of the driver or linux-input?

AndrewCapon, 2018-07-23:
Thanks for the reply.

No I haven't, I will though...

closed 2019-07-31 by JamesH65