raspberrypi/linux #2625: 1600x1200 not working with VC4 Full KMS

scpcom, 2018-07-25:
My Raspberry Pi 3B is connected to a hp 2035 LCD monitor which has a native resolution of 1600x1200 60 Hz (pixel clock 162000).
In legacy mode it works, in VC4 Full KMS mode the resolution goes down to 1280x960.

This was caused by this commit:
https://github.com/raspberrypi/linux/commit/48466df61a9c226b1b51e361aedf977dea6baf5c
If I compile a kernel with (1000 * 101 / 100) instead of (1000 * 108 / 100) it works perfectly.
es2gears and glxgears runs with 60 fps.
Where does the definition "HSM clock must be 108% of the pixel clock" come from?

popcornmix, 2018-07-27:
Can you report this issue here: https://github.com/anholt/linux/issues/
That is the source of the VC4 KMS code and we get updates from there.

closed 2018-07-27 by popcornmix

scpcom, 2018-07-30:
Thank you, I opened it here:
https://github.com/anholt/linux/issues/137