raspberrypi/linux #2628: 4.14.52-v7+ - IQaudio DAC and DMA transfer could not be terminated

whatdoineed2do, 2018-07-28:
@iqaudio, kernel devs

I have noticed an intermittent issue with my `RPi 3` and an `IQaudioDAC`.  Using forked-daapd playing music to the `hw:` device, I have noticed that after several hours (>3hrs) the sound drops out and I am unable to re-active the sound through the DAC unless I reboot the device.

In this state, plugging/unplugging the RCA from the DAC does NOT generate any feedback through the amp/speakers as it would normally - the amp is fine; plugging the RCA cables to my CD player gives feedback and sound.

`dmesg` shows:
```
[682720.987241] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated
[682725.413620] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated
[682747.015325] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated
[682771.303690] snd-rpi-iqaudio-dac soc:sound: snd_rpi_iqaudio_gpio_mute: muting amp using GPIO22
[682809.663438] pcm512x 1-004c: No SCLK, using BCLK: -2
[682809.666025] snd-rpi-iqaudio-dac soc:sound: snd_rpi_iqaudio_gpio_unmute: un-muting amp using GPIO22
[682925.576948] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated
[682933.384567] snd-rpi-iqaudio-dac soc:sound: snd_rpi_iqaudio_gpio_mute: muting amp using GPIO22
[682991.373137] pcm512x 1-004c: No SCLK, using BCLK: -2
[682991.377571] snd-rpi-iqaudio-dac soc:sound: snd_rpi_iqaudio_gpio_unmute: un-muting amp using GPIO22
[683017.569642] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated
[683022.585002] snd-rpi-iqaudio-dac soc:sound: snd_rpi_iqaudio_gpio_mute: muting amp using GPIO22
[683076.692295] pcm512x 1-004c: No SCLK, using BCLK: -2
[683076.697384] snd-rpi-iqaudio-dac soc:sound: snd_rpi_iqaudio_gpio_unmute: un-muting amp using GPIO22
```
Stopping foked-daapd and trying to play files directly result in the tools (sox/play/aplay) hanging.

Removing and reloading the `snd_soc_iqaudio_dac` driver makes no difference - only able to get sound after reboot.

Kernel: 4.14.52-v7+ armv7l

whatdoineed2do, 2018-08-19:
Reference to similar problem https://github.com/raspberrypi/linux/issues/1949 it suggests this error is linked to the slave clock on the DAC

I swapped out the IQaudio dac with an Allo Boss card (in master mode, with its own clock) on `4.14.52-v7+` and I have encountered the same `DMA could not be terminated`.  Unable to get sound output until a reboot

It suggests a bigger hides issue

any comments/ suggestions to track down issue?

whatdoineed2do, 2018-08-25:
Had different error msgs in log:
```
[633324.129919] bcm2835-i2s 3f203000.i2s: I2S SYNC error!
[633348.995784] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated
[633349.186290] bcm2835-i2s 3f203000.i2s: I2S SYNC error!
```
This comment: https://github.com/raspberrypi/linux/issues/1321#issuecomment-194570945 suggests that there could be a problem with BT (need to add: `dtoverlay=pi3-disable-bt`).

Does the following have any influence?
`dtparam=i2c_arm=on`

whatdoineed2do, 2018-08-25:
I only seem to notice this when the music is stopped and then restart (manually or on music player progress to next track).  Using `forked-daapd`for playback which is using 44100 exclusively

JamesH65, 2019-07-31:
Not sure us at RPI can do much about this, seems a third party problem.  Only thing to suggest is trying the latest kernel, and contacting the @iqaudio devs.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

whatdoineed2do, 2019-07-31:
Since this was reported this is RPi was upgraded to Stretch and still get this problem intermittently.  Is this the kernel driver?

I've seen this with an allo boss dac that is also based off the same kernel driver

closed 2020-02-13 by JamesH65