raspberrypi/linux #2629: IBSS (AdHoc) does not work on RPI 3B+ (and possibly others)

darkdrgn2k, 2018-07-28:
Can't seem to configure pi into IBSS (AdHoc) mode 

```
$ sudo  iwconfig wlan0 mode ad-hoc essid my-mesh ap 02:12:34:56:78:9A channel 1
Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlan0 ; Device or resource busy.
```

```
$ dmesg
brcmf_cfg80211_change_iface: iface validation failed: err=-16
```

Running Laters Raspbian and also rpi-update `4.14.56-v7+`

popcornmix, 2018-07-31:
Duplicate of https://github.com/raspberrypi/firmware/issues/990 ?

darkdrgn2k, 2018-07-31:
Could also be a re-surfacing of #1384

Not sure if its a Firmware issue or a Kernel Issue.

popcornmix, 2018-07-31:
It's not a (gpu) firmware issue (that has no knowledge of wifi).
I don't know enough to say if it's a kernel issue (possibly the wifi firmware) or a linux configuration issue.

closed 2019-09-18 by JamesH65