raspberrypi/linux #2630: Solution for uvccapture error failed to set probe control

Vlad-Romanov, 2018-07-29:
#623 #2082 #618

Hello, I had problem with uvccapture error, google led me to here, I have found solution so I will share.
https://github.com/Motion-Project/motion/releases
Download & install latest version of Motion - problem only happened with Motion in repo, doesn't happen with latest version. Hope this helps. If this happens when not using motion than I am sorry, but this what solved problem for me.

popcornmix, 2018-07-31:
Thanks for info, but if this is a distribution issue then it may be better to report here:
https://github.com/RPi-Distro/repo

where a decision can be made if moving to a newer version of the package is best.
If you have links to any other users reporting updating motion fixed their issue then including that.

JamesH65, 2019-07-31:
Closing as this issue is not kernel related.

closed 2019-07-31 by JamesH65